UniversalRobots/Universal_Robots_ROS_Driver #404: ROS Melodic build error due to ur_msgs mismatch

gokhansolak, 2021-06-15:
# Summary

I cannot `catkin_make` as the _ur_msgs_ package does not have the variable expected by the hardware interface.

```
Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:396:39: error: ‘ur_msgs::SetPayload::Request {aka struct ur_msgs::SetPayloadRequest_<std::allocator<void> >}’ has no member named ‘mass’
             << " set_payload(" << req.mass << ", [" << req.center_of_gravity.x << ", " << req.center_of_gravity.y
```

Similar to #235, but the solution there does not work (anymore). I have the Fmauch's fork on `calibration_devel` branch, that meta-package does not contain a `ur_msgs` package anymore. Rosdep installs a `ur_msgs` to the system, but that version does not build with this driver.

# Versions
 - ROS Driver version: Latest commit on `master`, ROS Melodic

# Details 

The system-wide `ur_msgs` package, installed by rosdep is:
`ros-melodic-ur-msgs/bionic,now 1.3.4-1bionic.20210525.170053 amd64 [installed]`

## Steps to Reproduce

Install according to the [instructions](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building).

fmauch, 2021-06-15:
Did you make sure to pull the `universal_robots` repository from my fork to the most recent version? There's been a major change including handling this properly. The driver's master branch now builds corrently with the `ur_msgs` package installed by rosdep. Which is why it is not anymore included [in my fork](https://github.com/fmauch/universal_robot/tree/calibration_devel).

gokhansolak, 2021-06-15:
The fork and the commit were correct. Strangely, deleting ROS build solved the issue (`rm -rf build/ devel/`). I used to have a build with older versions of these packages... Thank you!

closed 2021-06-15 by gokhansolak

fmauch, 2021-06-15:
right, as it has been built previously, it still hangs around in the devel / install space... thanks for following this up.

ljden, 2021-07-06:
I was having the same issue when updating to the newest setup. Removing the build did not fix it. Instead, I had to re-source the `/opt/ros/melodic/setup.bash` after installing ur_msgs so that ROS had the correct view of packages etc. After doing this, it built correctly.

ilariagi, 2022-11-15:
ROS Melodic, not working for me either

fmauch, 2022-11-16:
@ilariagi Please don't dig out old issues. Especially, if they don't connect to the repository. Neither `ur_modern_driver`, nor `universal_robot` nor `ur_msgs` are part of this repo.