UniversalRobots/Universal_Robots_ROS_Driver #405: How to set a cartesian pose and have the robot move to it

adhitir-magna, 2021-06-15:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.
I'm working on a UR10e hardware and I'm trying to do something quite simple. I really just want the robot to move from waypoint to waypoint. I'm able to send individual joint commands easily through the python API by setting a joint goal as shown in the tutorials on[ MoveIt!](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html). However, I'm not able to easily do the same by setting Cartesian coordiantes using move_group.get_current_pose().pose.

I understand that there are inverse kinematics solvers that I may have to use to get joint goals and set those, but I cannot find any resources on what those solvers are how to use them with the UR10 robot. Any help in that direction would be appreciated. 

# Versions
 - ROS Driver version: Using this driver with ROS Melodic and a UR10e

## Use Case and Setup
Looking to perform pick and place operations on hardware. 

## Expected Behavior
Move from one point to another

## Actual Behavior
When I use move_group.get_current_pose().pose, the robot is moving randomly. I tried to get it to move in the Z direction by 5 mm but it was moving somewhere else entirely.

fmauch, 2021-06-15:
Is your question about using moveit or is some interface from this driver not working correctly for you? This seems like it should be asked on ROS answers, instead.

adhitir-magna, 2021-06-16:
It's not specifically about using MoveIt! I want to use Python to control the UR10e hardware. In the MoveIt tutorials, this is being done by sending commands to geometry_msgs.msg.Pose(), which does not work with this robot and driver. I am able to read the end-effector pose with move_group.get_current_pose().pose but I'm not able to send commands to it. There isn't any documentation or test script that I can find that uses Python to send Cartesian commands that I can see. It doesn't have to be through MoveIt. I thought I could send joint position after performing inverse kinematics but I have no idea how to use the KDL or the ur_kinematics package to do this. I was hoping for some information or resources.

stefanscherzinger, 2021-06-16:
> I thought I could send joint position after performing inverse kinematics but I have no idea how to use the KDL or the ur_kinematics package to do this.

@adhitir-magna that seems like an ideal application for the new [cartesian_trajectory_controller](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/tree/main/cartesian_trajectory_controller#cartesian-trajectory-controller). The `position_controllers/CartesianTrajectoryController` should work with the driver's *master* branch. You could use it to send one-shot trajectory goals. The controller will take care of Inverse Kinematics and interpolation. That is actually a good timing of asking this. Many of the [Cartesian features](https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/tree/master#cartesian-ros-control) are about to be released with this driver (as in few days).

If you are in a hurry and are confident with Python (and some ROS-controller configuration), try [this integration test](url) as a first example of how to send Cartesian goals to the robot. If you manage to spawn the `jnt_cartesian_traj_controller` from [here](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/tree/main/cartesian_trajectory_controller#controller-yaml) as an additional ROS-controller for the driver, you should be good to go. Remember to adapt [this line](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/blob/main/cartesian_trajectory_controller/test/test_controller.py#L38) of the client for your application.
Probably to
```python
self.cart_client = actionlib.SimpleActionClient(
'jnt_cartesian_traj_controller/follow_cartesian_trajectory', FollowCartesianTrajectoryAction)
```

adhitir-magna, 2021-06-16:
This looks interesting and I am trying it out. Are there any installation instructions I'm missing? I was getting a CMake error:
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "speed_scaling_interface" with any of the following names:

    speed_scaling_interfaceConfig.cmake
    speed_scaling_interface-config.cmake

adhitir-magna, 2021-06-16:
I included the content of the contollers.yaml file in the UR10e_controllers.yaml file within the ROS driver. I added the `jnt_cartesian_traj_controller` to the list of controllers to spawn in the `ur10e_bringup` launch file as well, but that threw errors on execution. The specific error it threw didn;t make sense -- it said that it couldn't bring up the scaled_pos_joint_traj_controller. Once I removed it, I was able to connect to the UR10e hardware again. I can see the `jnt_cartesian_traj_controller` controller in the parameter server but I'm not sure how to spawn it with the controller_manager. More detailed instructions might be helpful.

On a sidenote, I couldn't even get the integration_test in the master branch of this driver to run just with the joint controllers. 
```
[ROSUNIT] Outputting test results to /home/adhitir/.ros/test_results/ur_robot_driver/rosunit-integration_test.xml
[Testcase: test_set_io] ... ok
[Testcase: test_trajectories] ... FAILURE!
FAILURE: False is not true
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "integration_test.py", line 82, in test_trajectories
    self.assertTrue(self.set_robot_to_mode(RobotMode.POWER_OFF))
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------
-------------------------------------------------------------
SUMMARY:
 * RESULT: FAIL
 * TESTS: 2
 * ERRORS: 0 []
 * FAILURES: 1 [test_trajectories]
 ```

fmauch, 2021-06-16:
The speed scaling interface is living [here](https://github.com/UniversalRobots/Universal_Robots_ROS_scaled_controllers) now. As this is in the process of being released (waiting for a binary sync) it will soon be available using the default `rosdep install` command.

stefanscherzinger, 2021-06-16:
> The specific error it threw didn;t make sense -- it said that it couldn't bring up the scaled_pos_joint_traj_controller. Once I removed it, I was able to connect to the UR10e hardware again. 

This is in fact expected behavior. Both controllers claim the same resources (joints). Only one of them will be allowed to run at a time. 

> but I'm not sure how to spawn it with the controller_manager

I like using the rqt GUI for this. You might need to add it with
```bash
sudo apt-get install ros-<your-distro>-rqt-controller-manager
```
In rqt, choose `Plugins` -> `Robot Tools` -> `Controller Manager`.
Try stopping the `scaled_pos_joint_traj_controller` and then starting the `jnt_cartesian_traj_controller`. To get a feeling for the Cartesian trajectory that your script will command, you might want to use [this variant for visualization](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/tree/main/cartesian_trajectory_controller#visualizing-cartesian-trajectories). That controller won't have resource conflicts.

PR #437 (2021-09-01): test_move scripts
This MR aims at providing test script to get familiar with moving the robot around using this driver without setting up moveit.

This is currently draft, as I'd like to add more documentation and warnings, that **this will move your robot** when a real robot is connected.

If you use the script(s) from this MR, have a look at the source code first and make sure that your robot is clear of any obstacles. Basically, these scripts assume that the robot is mounted on a table-like object and the upper hemisphere (seen from the robot's base) is clear of any obstacles. **Always occu [... cut]