UniversalRobots/Universal_Robots_ROS_Driver #407: got TF_OLD_DATA warning from time to time

lianghongzhuo, 2021-06-18:
# Summary

*Introduction to the issue*
I got TF_OLD_DATA from time to time, only restart the robot launch file can solve it.
error msg is like:
```
[ WARN] [1624001595.379673101]: TF_OLD_DATA ignoring data from the past for frame arm_upper_arm_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.379702295]: TF_OLD_DATA ignoring data from the past for frame arm_shoulder_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.379725272]: TF_OLD_DATA ignoring data from the past for frame arm_wrist_1_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.379750800]: TF_OLD_DATA ignoring data from the past for frame arm_wrist_2_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.379775379]: TF_OLD_DATA ignoring data from the past for frame arm_wrist_3_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380370301]: TF_OLD_DATA ignoring data from the past for frame arm_forearm_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380401027]: TF_OLD_DATA ignoring data from the past for frame arm_upper_arm_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380426203]: TF_OLD_DATA ignoring data from the past for frame arm_shoulder_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380452244]: TF_OLD_DATA ignoring data from the past for frame arm_wrist_1_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380478272]: TF_OLD_DATA ignoring data from the past for frame arm_wrist_2_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380502892]: TF_OLD_DATA ignoring data from the past for frame arm_wrist_3_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380555666]: TF_OLD_DATA ignoring data from the past for frame arm_forearm_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380582803]: TF_OLD_DATA ignoring data from the past for frame arm_upper_arm_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ WARN] [1624001595.380611405]: TF_OLD_DATA ignoring data from the past for frame arm_shoulder_link at time 1.624e+09 according to authority /robot_state_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
```
My understanding is that the timestamp that publishes the joint state of the ur robot is late than the current timestamp of tf has.

# Versions
 - ROS Driver version: master
 - Affected Robot Software Version(s): master 
 - Affected Robot Hardware Version(s): UR10e
 - Robot Serial Number: 
 - UR+ product(s) installed: UR10e
 - URCaps Software version(s):

# Impact
No impact everything is working except the warning.

## Use Case and Setup
ur driver is on master, ur client is from Debian.

## Project status at point of discovered
*When did you first observe the issue?*
- The issue comes not so often, several times per day.

## Steps to Reproduce
It's hard to reproduce, I guess there is some network communication drop?

## Expected Behavior
No warning, or have warning sometimes, but recover soon.
In my case, once the warning happens, it won't disappear unless I restart the robot launch file.

## Workaround Suggestion
restart the robot launch file

fmauch, 2021-06-28:
This should not happen, but I also cannot imagine how this should happen.

Do you run a multi-machine ROS setup? If there are multiple machines part of one ROS network, there could be issues, if their times are not synchronized exactly. But I would expect a different output in that case.

lianghongzhuo, 2021-06-28:
Thanks for the reply. There is only one ros machine.