UniversalRobots/Universal_Robots_ROS_Driver #418: Unstable connection (Connection to reverse interface dropped.)

felixvd, 2021-06-25:
# Summary

On our system, there is an intermittent issue with the connection to the robot. Starting up certain nodes seems to cause it sometimes, but it is not deterministic.

I assume that it has to do with network or system load, but we were not able to pinpoint the issue yet. I'm posting this in case someone has experiences to share or a good idea on how to debug this.

# Versions
 - ROS Driver version: 13db13 on ROS Melodic
 - Affected Robot Software Version(s): 5.9, 5.10
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number:
 - UR+ product(s) installed: None
 - URCaps Software version(s): 1.0.5

# Impact

The robot driver randomly disconnects and trajectory execution is aborted.

# Issue details

We get errors like the following:

```
[ERROR] [1624420575.141752331]: Sending data through socket failed.
[ERROR] [1624420575.141840852]: Sending data through socket failed.
[ERROR] [1624420575.141891377]: Sending data through socket failed.
[ERROR] [1624420575.141932492]: Sending data through socket failed.
[ERROR] [1624420575.141968390]: Sending data through socket failed.
[ERROR] [1624420575.142005647]: Sending data through socket failed.
[ WARN] [1624420575.142424334]: Connection attempt on port 50003 while maximum number of clients (1) is already connected. Closing connection.
[ INFO] [1624420575.142727757]: Connection to reverse interface dropped.

[...]
[ WARN] [1624420355.903048112]: Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.
[ INFO] [1624420355.903115624]: Connection to reverse interface dropped.

[...]
[ INFO] [1624420328.661596839]: Connection to reverse interface dropped.
[ INFO] [1624420328.661912754]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1624420328.661910557]: Connection to reverse interface dropped.
[ INFO] [1624420328.662169994]: Robot connected to reverse interface. Ready to receive control commands.
```

## Project status at point of discovered

We recently added two more Realsense cameras (for a total of 4) and a second PC to the system, but this might have just been the last drop. 

## Steps to Reproduce

I haven't been able to reproduce the issue reliably.

## Workaround Suggestion

--

MatMcT, 2021-07-13:
We too have just ran into this problem. I recently updated to the latest commit today (12/07/2021). Our last update was last year some time...

Hardware-wise nothing has changed. :(

felixvd, 2021-07-13:
There might have been changes to the backend that catch these errors now. FWIW, we're working under the assumption that it's related to #399, and will try to put the robots on a different network. Will update when we get to it.

MatMcT, 2021-07-14:
@felixvd when you say put the robot on a different network, what do you intend on doing and what is your current setup? Do you have a Real-Time Kernel install? 

I'd like to get some insight before I go down a rabbit hole. 

I'm currently running our setup on a NUC-style PC. All our machines (equipment + robot) are connected via a network switch. So I don't have many other options for connecting the UR on a separate network.

The only options I (think) I have are:

1. Connect the UR to the dedicated ethernet jack on the NUC and connect the network switch to a USB-Ethernet drive.
2. Run the UR on a dedicated separate PC and run a multi PC ROS configuration.

This is a commercially sensitive project so either options don't sound too enticing when you need to consider reliability and cost.

Would like to hear what others think.

felixvd, 2021-07-14:
We don't run an RT kernel because our system is in a Docker container.

We were running on a single PC, but added a second PC for vision tasks when the errors started to occur (under the assumption that calculation load was causing the issues). That did not solve the problem. We also connected all robots and PCs to the same network switch.

We will add a network card to the main PC, and another network switch to the system, so the main PC and robots and connected to one network, and the main + vision PCs are connected to another network (switch).

I think your first solution sounds fine, unless cameras overload the network interfaces in some other way I don't understand.

If the problem is intermittent, using the Dashboard services' start/connect/quit+connect to recover can be an option, but the fact that the trajectory is aborted when the problem occurs is definitely bothersome.

gavanderhoorn, 2021-07-14:
> We don't run an RT kernel because our system is in a Docker container.

Just a note: that does not immediately mean you can't use an RT kernel.

It would probably require running the container with either `--privileged` or `SYS_NICE` (be aware of the drawbacks of `--privileged` of course). That would give processes inside the container the possibility to set their scheduling configuration appropriately. I've successfully used this to containerise RT applications.

(it's not really *hard* RT in any sense, but as this is a PREEMPT kernel and a regular Linux distribution that's not really a concern here. Basically things need to run "fast enough")

Of course this will mean you'd need an RT kernel on the host.

MatMcT, 2021-07-14:
> Just a note: that does not immediately mean you can't use an RT kernel.
>
> It would probably require running the container with either `--privileged` or `SYS_NICE` (be aware of the drawbacks of [...]

This is good to know because I plan on doing exactly this.

MatMcT, 2021-07-15:
UPDATE:

I installed RT Kernel 5.9.1-rt20 and the problem hasn't occurred since. 

I'm running a fairly basic pick and place application and before the Kernel update, it would fail as described by OP every 10 or so picks. Very sporadic and hard to reproduce deterministically.  

After the update, I haven't had the issue in at least over 100 picks. 

Some issues I encountered when installing the RT Kernel which may help future readers:

debian/rules:6: recipe for target 'build' failed
This seems to be caused by a previous non-fatal build error I noticed:
"recipe for target 'certs' failed"

Which can be fixed by running the following the linux-x.x.x folder
`sed -ri '/CONFIG_SYSTEM_TRUSTED_KEYS/s/=.+/=""/g' .config`
See [here](https://unix.stackexchange.com/questions/293642/attempting-to-compile-kernel-yields-a-certification-error) for more detail.

After that small fix, follow all instructions as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md) as normal.

fmauch, 2021-07-19:
@MatMcT Thanks for testing this out and good to hear that this solved your problem. 

If there's nothing open anymore, please feel free to close this issue :-)

If you'd like to contribute to our RT-Kernel installation guide, that would be awesome. I've created a separate issue (#436) for this.

felixvd, 2021-07-20:
As a data point, here's the results from our short investigation. We started with a single machine with 4 Realsense SR305 cameras and two UR5e robots connected to it. The errors from the original post would happen intermittently, and often after starting up vision-related nodes. The cameras were running on 640x480 resolution and 10 fps, so we did not expect high network load, but couldn't be sure.

We first added a network card to our machine to connect the robots via a separate network, so the cameras would not affect the transmission. This did not help.

We then added a second PC (vision PC) to the system, connected all the cameras to it, moved all the vision-related nodes to it, as well as the `roscore`. The original PC (robot PC) was only running the UR drivers, Robotiq gripper drivers, a small Dynamixel node and MoveIt/Rviz. There were no subscribers to the camera topics on the robot PC. The error *still* occurred at times on the robot PC, when starting up vision-related nodes on the vision PC. 

We tried installing the real-time kernel and can confirm that the build fails at the point that @MatMcT mentions, and that the instruction fixes it. We also had to remove the Nvidia drivers from our machine to boot with the RT kernel, because it locks up otherwise.

The errors seem to have disappeared, so my conclusion is that only the real-time kernel will reliably fix it if the system is sufficiently complex. I'll close this for now. Thanks for the input!

closed 2021-07-20 by felixvd

fmauch, 2021-07-22:
Thanks for the in-detail summary @felixvd!

zp2546265641, 2024-07-13:
> 感谢您的详细总结@felixvd！

i have installed 5.15.76-rt53, but it didn't work as expected. It still reported an error. Whenever I use urcap to connect, start the digital output and analog output IO signals, start my air pump, and then send the trajectory program planned by moveit to the real robot arm, the following problem will occur, but this time after i install LinuxRT kernel it was a little different:
[ INFO] [1720873969.508932034]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1720873975.622664768]: Received request to compute Cartesian path
[ INFO] [1720873975.624387838]: Attempting to follow 31139 waypoints for link 'tool_ros' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1720873976.552106491]: Computed Cartesian path with 31145 points (followed 100.000000% of requested trajectory)
[ERROR] [1720873977.292264538]: Sending data through socket failed.
[ INFO] [1720873977.292354495]: Connection to reverse interface dropped.
[ INFO] [1720873977.372595596]: Execution request received
[ERROR] [1720873977.400574305]: Can't accept new action goals. Controller is not running.
[ WARN] [1720873977.400910620]: Controller 'scaled_pos_joint_traj_controller' failed with error INVALID_GOAL:
[ WARN] [1720873977.401011718]: Controller handle scaled_pos_joint_traj_controller reports status FAILED
[ INFO] [1720873977.401071288]: Completed trajectory execution with status FAILED ...
[ INFO] [1720873977.402529277]: Execution completed: FAILED

tvu-radicalai, 2025-06-16:
We're still getting this error.

```
[pick_and_place-13] [INFO] [1750113051.171894575] [station_controller]: EXECUTING
[pick_and_place-13] [INFO] [1750113051.171897561] [station_controller]: Executing plan to sample_holder
[pick_and_place-13] [INFO] [1750113051.172276738] [planning_scene_interface_103860876883712.moveit.moveit.ros.move_group_interface]: Execute request accepted
[move_group-11] [INFO] [1750113051.172281357] [move_group.moveit.moveit.ros.move_group.clear_octomap_service]: Execution request received
[move_group-11] [INFO] [1750113051.172345435] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-11] [INFO] [1750113051.172380679] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-11] [INFO] [1750113051.172475794] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.1
[move_group-11] [INFO] [1750113051.174504195] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-11] [INFO] [1750113051.174569665] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-11] [INFO] [1750113051.174609248] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-11] [INFO] [1750113051.174938654] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to scaled_joint_trajectory_controller
[ros2_control_node-1] [INFO] [1750113051.175323221] [scaled_joint_trajectory_controller]: Received new action goal
[ros2_control_node-1] [INFO] [1750113051.175370118] [scaled_joint_trajectory_controller]: Accepted new action goal
[move_group-11] [INFO] [1750113051.175525053] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: scaled_joint_trajectory_controller started execution
[move_group-11] [INFO] [1750113051.175540571] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-1] [INFO] [1750113051.361216061] [UR_Client_Library:]: Connection to reverse interface dropped.
[ros2_control_node-1] [ERROR] [1750113051.443203024] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.445208864] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.447205276] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.449176962] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.451174746] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.453178431] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.455199479] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.457156027] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.459126341] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.461104069] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.463095091] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.465090030] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.467085129] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.469099734] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.471116213] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.473111252] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.475093158] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.477112602] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.479098324] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.481094105] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.483106877] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.485097989] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.487145785] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.489109898] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[ros2_control_node-1] [ERROR] [1750113051.491176358] [UR_Client_Library:]: Pipeline producer overflowed! <RTDE Data Pipeline>
[... cut ...]
[pick_and_place-13] [ERROR] [1750113051.676950029] [planning_scene_interface_103860876883712.moveit.moveit.ros.move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
[pick_and_place-13] [ERROR] [1750113051.677004580] [station_controller]: Motion execution failed with error code: -4
```

The pick_and_place is our own controller, but the `connection to reverse interfaced dropped` keeps appearing.

Has anyone encountered this with more modern versions of the UR driver in ROS 2?