UniversalRobots/Universal_Robots_ROS_Driver #424: suddenly controller error

robint-XNF, 2021-07-02:
I suddenly can't use ur3_moveit, it shows the error:Action client not connected: /scaled_pos_joint_traj_controller/follow_joint_trajectory.
and the ur3_bring_up shows "Robot ready to receive control commands", i have also touched the "play" button. it used to work, but now it does not work. i don't kown why? could'you please help me ?

robint-XNF, 2021-07-03:
i try to use a remote computer to control ur, however, when i use " rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller", the error shows "Unable to communicate with service [/controller_manager/load_controller], address [rosrpc://192.168.199.190:36791]"

closed 2021-07-03 by robint-XNF

gavanderhoorn, 2021-07-03:
If you solved your problem, could you please post the solution and not remove the original question/report?

That would help future users if they ever encounter the same problem.