UniversalRobots/Universal_Robots_ROS_Driver #427: Difference between this repo and ros-industrial/universal_robot

nberliner, 2021-07-07:
I've asked the same a couple of days ago on stackoverflow (see [here](https://stackoverflow.com/questions/68258402/difference-between-the-ros-universalrobots-universal-robots-ros-driver-and-the-r)) but didn't seem to reach the right audience. Sorry for cross-posting but it would be great to have some insights.

Copying the original post below:

----
I've tried to understand the difference between the two ROS Universal Robot drivers and decide which one to use. So far, I'm mainly confused. As I am new to ROS and robot control I'd appreciate any explanation and hints where to start looking for more details.

From what I've seen, there are two Universal Robot ROS drivers available

(1) https://github.com/ros-industrial/universal_robot

(2) https://github.com/UniversalRobots/Universal_Robots_ROS_Driver

with (1) being "supplemented" by (3) https://github.com/fmauch/universal_robot/tree/calibration_devel.

(1) wasn't updated in while whereas (2) seems to get regular updates. Yet (3) seems to suggest that (1) is still being worked on (see https://github.com/ros-industrial/universal_robot/issues/573 as well). Which one is the "active" one? Which one should be used for which use case?

It also seems to me that I can't use (2) with gazebo. However, as I said, I'm new to ROS and might be misunderstanding something entirely. What I'd like to be able to do is simulate an UR10e to develop my application within gazebo and then swap to the real robot as transparently as possible.

Thanks to all the maintainers of the UR ROS drivers (from both/all repos :))!

fmauch, 2021-07-07:
https://github.com/ros-industrial/universal_robot is a repository containing a lot of UR-related ROS packages. `ur_driver` from there has been deprecated a long time ago and if you look at the [current devel branch](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging) it is gone.

https://github.com/fmauch/universal_robot/tree/calibration_devel is just a fork of that repository (which also doesn't contain) the `ur_driver` package anymore. It is currently being synched back in https://github.com/ros-industrial/universal_robot/pull/562 and https://github.com/ros-industrial/universal_robot/pull/538.

So, in short: This repo here is the (only official) ROS driver from and for Universal Robots which depends on (1) (respectively (3) if you want MoveIt! support) and https://github.com/ros-industrial/ur_msgs. [`ur_driver`](http://wiki.ros.org/ur_driver) and [`ur_modern_driver`](http://wiki.ros.org/ur_modern_driver) have been deprecated.

closed 2021-07-07 by fmauch

nberliner, 2021-07-07:
Thanks for the clarification and the quick reply!