UniversalRobots/Universal_Robots_ROS_Driver #430: cant use moveit to plan

cz-hit, 2021-07-13:
here is the info
[ 82%] Built target ur3_kin
Scanning dependencies of target ur3_moveit_plugin
[ 83%] Building CXX object fmauch_universal_robot/ur_kinematics/CMakeFiles/ur3_moveit_plugin.dir/src/ur_moveit_plugin.cpp.o
In file included from /home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:75:0:
/opt/ros/melodic/include/class_loader/class_loader.h:36:2: warning: #warning Including header <class_loader/class_loader.h> is deprecated, include <class_loader/class_loader.hpp> instead. [-Wcpp]
 #warning Including header <class_loader/class_loader.h> is deprecated, \
  ^~~~~~~
/home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp: In member function ‘virtual bool ur_kinematics::URKinematicsPlugin::initialize(const string&, const string&, const string&, const string&, double)’:
/home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:174:88: warning: ‘virtual void kinematics::KinematicsBase::setValues(const string&, const string&, const string&, const string&, double)’ is deprecated [-Wdeprecated-declarations]
 s(robot_description, group_name, base_frame, tip_frame, search_discretization);
                                                                              ^
In file included from /home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/include/ur_kinematics/ur_moveit_plugin.h:100:0,
                 from /home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:88:
/opt/ros/melodic/include/moveit/kinematics_base/kinematics_base.h:344:31: note: declared here
   [[deprecated]] virtual void setValues(const std::string& robot_description, const std::string& group_name,
                               ^~~~~~~~~
[ 83%] Linking CXX shared library /home/cz/ros_stuff/catkin_ws/devel/lib/libur3_moveit_plugin.so
[ 83%] Built target ur3_moveit_plugin
Scanning dependencies of target ur5_moveit_plugin
[ 83%] Building CXX object fmauch_universal_robot/ur_kinematics/CMakeFiles/ur5_moveit_plugin.dir/src/ur_moveit_plugin.cpp.o
In file included from /home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:75:0:
/opt/ros/melodic/include/class_loader/class_loader.h:36:2: warning: #warning Including header <class_loader/class_loader.h> is deprecated, include <class_loader/class_loader.hpp> instead. [-Wcpp]
 #warning Including header <class_loader/class_loader.h> is deprecated, \
  ^~~~~~~
/home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp: In member function ‘virtual bool ur_kinematics::URKinematicsPlugin::initialize(const string&, const string&, const string&, const string&, double)’:
/home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:174:88: warning: ‘virtual void kinematics::KinematicsBase::setValues(const string&, const string&, const string&, const string&, double)’ is deprecated [-Wdeprecated-declarations]
 s(robot_description, group_name, base_frame, tip_frame, search_discretization);
                                                                              ^
In file included from /home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/include/ur_kinematics/ur_moveit_plugin.h:100:0,
                 from /home/cz/ros_stuff/catkin_ws/src/fmauch_universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:88:
/opt/ros/melodic/include/moveit/kinematics_base/kinematics_base.h:344:31: note: declared here
   [[deprecated]] virtual void setValues(const std::string& robot_description, const std::string& group_name,
                               ^~~~~~~~~
[ 84%] Linking CXX shared library /home/cz/ros_stuff/catkin_ws/devel/lib/libur5_moveit_plugin.so
[ 84%] Built target ur5_moveit_plugin

I installed the ros-melodic and do all the step as "readme.md"
but catkin_make report this
and I can't use moveit to plan 
[image]

and when I run roslaunch ur5_moveit_config demo.launch,it reported this:

You can start planning now!

[ INFO] [1626177524.129451743]: Stereo is NOT SUPPORTED
[ INFO] [1626177524.129602709]: OpenGL device: SVGA3D; build: RELEASE;  LLVM;
[ INFO] [1626177524.129868572]: OpenGl version: 3.3 (GLSL 3.3) limited to GLSL 1.4 on Mesa system.
[ INFO] [1626177527.649795322]: Loading robot model 'ur5_robot'...
[ WARN] [1626177527.688472365]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1626177527.815380437]: Starting planning scene monitor
[ INFO] [1626177527.817833274]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1626177528.106293439]: No active joints or end effectors found for group 'endeffector'. Make sure that kinematics.yaml is loaded in this node's namespace.

pibaopi, 2021-07-14:
hi,guy.I have encountered the same problem as you in the past two days. Have you solved it?

fmauch, 2021-07-19:
The catkin_make output mentioned does not show any error, but merely deprecation warnings.

As I do not see any problem regarding this driver, but a problem setting this up using MoveIt!, I will close this, but add it to the MoveIt tutorial project.

closed 2021-07-19 by fmauch