UniversalRobots/Universal_Robots_ROS_Driver #432: I cannot install UR ROS driver on ROS Noetic 20.04

MartinJensen37, 2021-07-15:
# Summary
The problem arises when I try to install the module from source. I get an issue with the package 'ur_kinematics', it's specifically a cmake error saying the following: 

```
Boost_INCLUDE_DIR
   used as include directory in directory /home/martin/Documents/repositories/catkin_ws/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/martin/Documents/repositories/catkin_ws/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/martin/Documents/repositories/catkin_ws/src/fmauch_universal_robot/ur_kinematics
```
The error has many more lines that this but it's simply repeating the same thing. I looked the package 'ur_kinematics and it only seems to be supported by ROS kinetic. 
I'm wondering if anyone has had the same problems and if you have managed to resolve them.

Thanks in advance!

Martin

fmauch, 2021-07-19:
First of all, `ur_kinematics` is not a part of this repository, so the more appropriate place to open this would be https://github.com/fmauch/universal_robot. Secondly, please post your complete CMake output, as the missing `Boost_INCLUDE_DIR` probably results from an earlier error.

FinnLinxxx, 2021-08-08:
I do encounter the very same problem. I've overcome it once, so i got it build on my system already, but I do not know why, its odd.

On my side the problem seams to be raised by 'ur_kinematics' as well, but I believe there is some more to it, that's why I choosed to stay inside this issue for now, but I might not read the output the right way. 

Here it is:

`Description:	Ubuntu 20.04.2 LTS, noetic` 
`$ sudo apt update on 08.08.2021`

```bash
( I've followed the install instructions inside the Readme.md)
$ source /opt/ros/noetic/setup.bash
$ mkdir -p workspace_ur/src
$ cd workspace_ur
$ git clone https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git src/Universal_Robots_ROS_Driver
$ git clone -b calibration_devel https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot
$ sudo apt update -qq && rosdep update
$ rosdep install --from-paths src --ignore-src -y
( #All required rosdeps installed successfully )
$ catkin_make
```

```cmake
Base path: /home/finn/workspace_ur
Source space: /home/finn/workspace_ur/src
Build space: /home/finn/workspace_ur/build
Devel space: /home/finn/workspace_ur/devel
Install space: /home/finn/workspace_ur/install
####
#### Running command: "cmake /home/finn/workspace_ur/src -DCATKIN_DEVEL_PREFIX=/home/finn/workspace_ur/devel -DCMAKE_INSTALL_PREFIX=/home/finn/workspace_ur/install -G Unix Makefiles" in "/home/finn/workspace_ur/build"
####
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/finn/workspace_ur/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/finn/workspace_ur/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 16 packages in topological order:
-- ~~  - universal_robots (metapackage)
-- ~~  - ur_dashboard_msgs
-- ~~  - controller_stopper
[... 138 lines cut ...]
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics

-- Configuring incomplete, errors occurred!
See also "/home/finn/workspace_ur/build/CMakeFiles/CMakeOutput.log".
See also "/home/finn/workspace_ur/build/CMakeFiles/CMakeError.log".
```

The most odd line may be 
`   '/home/finn/workspace_ur/src/Universal_Robots_ROS_Driver/ur_robot_driver//include'. `
because of the double slash, but then again I might not read this the right way.

Hope this clears things up, @MartinJensen37 might be able to recognize this as the same Error.

I'd be glad for any hint, even in means in moving the issue over to `universal_robot`

MartinJensen37, 2021-08-09:
Hello @FinnLinxxx.

I have the exact same error message however I still don't know how to fix it. I suspect that the ${include} definition is getting changed from include to /include, but I can't find out where it is.

MartinJensen37, 2021-08-09:
I found this piece of code in the ur_robot_driverConfig.cmake file that is raising the error. It can be seen that a path just called /include is added in the include_dirs. I'm not sure how to remove this since the path does not exist let alone if it is the problem to begin with. The code is here:  

    # flag project as catkin-based to distinguish if a find_package()-ed project is a catkin project
    set(ur_robot_driver_FOUND_CATKIN_PROJECT TRUE)

    if(NOT "/home/martin/Documents/repositories/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include;/include;/opt/ros/noetic/include;/usr/include " STREQUAL " ")
      set(ur_robot_driver_INCLUDE_DIRS "")
      set(_include_dirs "/home/martin/Documents/repositories/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include;/include;/opt/ros/noetic/include;/usr/include")
      if(NOT "https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues " STREQUAL " ")
        set(_report "Check the issue tracker 'https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues' and consider creating a ticket if the problem has not been reported yet.")
      elseif(NOT "http://wiki.ros.org/ur_robot_driver " STREQUAL " ")
        set(_report "Check the website 'http://wiki.ros.org/ur_robot_driver' for information and consider reporting the problem.")
      else()
        set(_report "Report the problem to the maintainer 'Felix Exner <exner@fzi.de>' and request to fix the problem.")
      endif()
      foreach(idir ${_include_dirs})
        if(IS_ABSOLUTE ${idir} AND IS_DIRECTORY ${idir})
         set(include ${idir})
        elseif("${idir} " STREQUAL "include ")
          get_filename_component(include "${ur_robot_driver_DIR}/../../../include" ABSOLUTE)
          if(NOT IS_DIRECTORY ${include})
            message(FATAL_ERROR "Project 'ur_robot_driver' specifies '${idir}' as an include dir, which is not found.  It does not exist in '${include}'.  ${_report}")
          endif()
        else()
          message(FATAL_ERROR "Project 'ur_robot_driver' specifies '${idir}' as an include dir, which is not found.  It does neither exist as an absolute directory nor in '/home/martin/Documents/repositories/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/${idir}'.  ${_report}")
        endif()
        _list_append_unique(ur_robot_driver_INCLUDE_DIRS ${include})
      endforeach()
    endif()
I'll try and contact Felix Exner as written in the file if I don't find a fix to the current build error.

FinnLinxxx, 2021-08-09:
Hey Martin, 

Felix Exner is the maintainer of the https://github.com/fmauch/universal_robot repository
_aka. @fmauch_ 

(FZI Forschungszentrum Informatik)

I hope he has time to look over the issue

stefanscherzinger, 2021-08-09:
@MartinJensen37 @FinnLinxxx Until a proper solution is proposed to fix this issue, you could try to create `/include` as an empty folder under `/` root, since that existence seems to be required. That's extremely hacky and should only serve as a temporal fix for you to continue.

In addition, is a system install of the client_library a requirement for your use case (via `sudo apt install ros-noetic-ur-client-library`) or is building that [from source](https://github.com/UniversalRobots/Universal_Robots_Client_Library/tree/boost) (branch boost) also an option?
It's just a wild guess, but comparing my current setup with [your ur_robot_driverConfig.cmake](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/432#issuecomment-895127308) is different by that mine includes the ur_client_library's headers instead:

```cmake
# flag project as catkin-based to distinguish if a find_package()-ed project is a catkin project
set(ur_robot_driver_FOUND_CATKIN_PROJECT TRUE)

if(NOT "/home/scherzin/src/robot_folders/checkout/ur_rosin/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include;/usr/include;/home/scherzin/src/robot_folders/checkout/ur_rosin/catkin_ws/devel_isolated/ur_client_library/include " STREQUAL " ")

```

gavanderhoorn, 2021-08-09:
It's possible this is caused by [this line](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/125f4f7a48b299b07e298cf479e3ba6882c3af3a/ur_client_libraryConfig.cmake#L9) in the `ur_client_libraryConfig.cmake`.

I'd suggest printing the `ur_client_library_INCLUDE_DIRS` variable to see whether it contains `/include`.

FinnLinxxx, 2021-08-09:
@stefanscherzinger thank you very much for your suggestions and hints. 
I created the `/include` folder as suggested (`$ sudo mkdir /include`) , it looks like at least this error-message is aided for now, as it doesnt show up anymore.

```cmake
...
-- +++ processing catkin package: 'ur_robot_driver'
-- ==> add_subdirectory(Universal_Robots_ROS_Driver/ur_robot_driver)
ur_robot_driver: You did not request a specific build type: selecting 'RelWithDebInfo'.
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Installing devel-space wrapper /home/finn/workspace_ur/src/Universal_Robots_ROS_Driver/ur_robot_driver/scripts/tool_communication to /home/finn/workspace_ur/devel/lib/ur_robot_driver
-- +++ processing catkin package: 'ur_calibration'
-- ==> add_subdirectory(Universal_Robots_ROS_Driver/ur_calibration)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Boost_INCLUDE_DIR
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
   used as include directory in directory /home/finn/workspace_ur/src/fmauch_universal_robot/ur_kinematics
...
```

Unfortunately the main issue is still there. As I have no further constrains, I've followed up on your second thought:
 * I removed the ur-client-library `$ sudo apt remove ros-noetic-ur-client-library`
 * setup a new workspace `$ mkdir -p ~/workaround_ur/src`
 * cloned the ur-client-library there `$ git clone -b boost https://github.com/UniversalRobots/Universal_Robots_Client_Library.git ~/workaround_ur/src`
 * build it with `$ catkin_make_isolated` 
 * sourced it `$ source ~/workaround_ur/devel_isolated/setup.bash` 
 * and tried to build my `workspace_ur` again. 
 

_The Error Resists._

My `ur_robot_driverConfig.cmake` now looks similar to yours (with the `/include`-Entry still there):

```cmake 
# flag project as catkin-based to distinguish if a find_package()-ed project is a catkin project
set(ur_robot_driver_FOUND_CATKIN_PROJECT TRUE)

if(NOT "/home/finn/workspace_ur/src/Universal_Robots_ROS_Driver/ur_robot_driver/include;/include;/home/finn/workaround_ur/devel_isolated/ur_client_library/include " STREQUAL " ")
```

FinnLinxxx, 2021-08-09:
@gavanderhoorn 
Thank you to to provide a further suggestion.

I would like to provide the given variable. Unfortunately I don't know how to output the parameter.
With the simple cmake message function I had no luck:

```bash
$ mkdir ~/buildURCL
$ cd ~/buildURCL
$ git clone -b boost https://github.com/UniversalRobots/Universal_Robots_Client_Library.git
$ cd ~/buildURCL/Universal_Robots_Client_Library
$ vim ur_client_libraryConfig.cmake
```
```cmake
include(CMakeFindDependencyMacro)

if(NOT TARGET ur_client_library::urcl)
  include("${CMAKE_CURRENT_LIST_DIR}/urclTargets.cmake")
endif()

# This is for catkin compatibility. Better use target_link_libraries(<my_target> ur_client_library::ur_client_library)
set(ur_client_library_LIBRARIES ur_client_library::urcl)
get_target_property(ur_client_library_INCLUDE_DIRS ur_client_library::urcl INTERFACE_INCLUDE_DIRECTORIES)
message(${ur_client_library_INCLUDE_DIRS})
```
```bash
$ mkdir build && cd build
$ cmake ..
```
(no: `message(${ur_client_library_INCLUDE_DIRS})` given in cmake output)

If its not too much, I kindly ask you to advise me on how to do it the right way.

stefanscherzinger, 2021-08-09:
Thanks @FinnLinxxx for trying the suggestions and your detailed output. It's probably two issues. The first one being the obscure `/include` whose origin we need to clarify and the second being the unset `Boost_INCLUDE_DIR` that cmake is complaining about.

Concerning the second issue, from your build output, I get that Boost is found via 

> /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake

which only defines `Boost_INCLUDE_DIRS` via `get_target_property(Boost_INCLUDE_DIRS Boost::headers INTERFACE_INCLUDE_DIRECTORIES)` but not `Boost_INCLUDE_DIR`. Note the additional `S`.

I'm not sure right now, but I could imagine that [this line ](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_kinematics/CMakeLists.txt#L26)dates back to older Boost versions.
Inside your `fmauch_universal_robot/ur_kinematics/CMakeLists.txt`, try replacing the include (line 26) with this:

```yaml
include_directories(SYSTEM ${Boost_INCLUDE_DIRS})
```

to see if this fixes the build issue.

stefanscherzinger, 2021-08-09:
Concerning the first issue and what @gavanderhoorn suggested, you need to add the cmake message to your `ur_robot_driver/CMakeLists.txt` because that is using the client library. Put it after line `36`:

```yaml
find_package(ur_client_library REQUIRED)

# Print client library includes of the driver
message(WARNING "############### My ur_client_library_INCLUDE_DIRS: ${ur_client_library_INCLUDE_DIRS}")

```
and build it with
```bash
catkin_make_isolated --pkg ur_robot_driver
```
You might need to build the client library before (individually with `--pkg`).
Let's see in the build output if that's the reason for populating `/include`.

FinnLinxxx, 2021-08-10:
@stefanscherzinger 
_Success!_
The additional `S` has resulted in the code being able to be built

`include_directories(SYSTEM ${Boost_INCLUDE_DIRS})`

Thank you very much for your expertise and patience! 

This might be interesting for @MartinJensen37 as well.

---
@stefanscherzinger @gavanderhoorn 

Following up the second problem in this issue, I add up the lines as prompted.
This is the resulting output:

```cmake
...
-- +++ processing catkin package: 'ur_robot_driver'
-- ==> add_subdirectory(Universal_Robots_ROS_Driver/ur_robot_driver)
ur_robot_driver: You did not request a specific build type: selecting 'RelWithDebInfo'.
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
CMake Warning at Universal_Robots_ROS_Driver/ur_robot_driver/CMakeLists.txt:34 (message):
  ############### My ur_client_library_INCLUDE_DIRS:
  /home/finn/workaround_ur/devel_isolated/ur_client_library/include;/include

-- Installing devel-space wrapper /home/finn/workspace_ur/src/Universal_Robots_ROS_Driver/ur_robot_driver/scripts/tool_communication to /home/finn/workspace_ur/devel/lib/ur_robot_driver
-- +++ processing catkin package: 'ur_calibration'
...
```

Containing the ominous `/include`.

If you are not able to reproduce this behavior, let me now how I can help any further to get to the root of this "second" issue.

gavanderhoorn, 2021-08-10:
The root cause is the fact that `ur_kinematics` references `Boost_INCLUDE_DIR` instead of `Boost_INCLUDE_DIRS`.

MartinJensen37, 2021-08-10:
Hello again. 

I managed to reproduce the same results as @FinnLinxxx did. So all the packages can be built now. 

I'm guessing the issue has been resolved now (with exception of the workaround with the /include dir in the root folder). Thanks for the help @stefanscherzinger @gavanderhoorn @FinnLinxxx!

stefanscherzinger, 2021-08-10:
@gavanderhoorn Do you mean that using CMake's `cache entries` instead of its `result variables` (as recommended in this [hints section](https://cmake.org/cmake/help/latest/module/FindBoost.html#hints)) lets `ur_kinematics` cause the unwanted side effect of adding `/include` to the  `ur_client_library`s include dirs?

closed 2021-08-12 by MartinJensen37

pkoprov, 2023-04-29:
Same issue is persistent. When the line 26 in src/universal_robot/ur_kinematics/CMakeLists.txt contains `include_directories(SYSTEM ${Boost_INCLUDE_DIR})` instead of `include_directories(SYSTEM ${Boost_INCLUDE_DIR**S**})
then the following cmake error pops up:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Boost_INCLUDE_DIR
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
[... cut ...]
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
   used as include directory in directory /home/pkoprov/ur_ws/src/universal_robot/ur_kinematics
```
I created a pull request that adds S to DIR  
https://github.com/ros-industrial/universal_robot/pull/635

pkoprov, 2023-04-29:
I still need to manually create include directory in root. I don't think that it is supposed to be like that. BTW I am running Ubuntu 20.04 and ROS Noetic

JKH-ConeTec, 2024-03-18:
I have noticed that if I  `catkin_make` right from the computer it fully compiles and the errors are absent.  However, if I `catkin_make` from a remote connection such as SSH or RDP, then both of these errors arise.

**Solution:**

- If possible, try and `catkin_make` right from the computer, and _not_ from a remote connection such as SSH or RDP.

- If method accessing the computer directly is not possible for your situation, then running the following commands in your `catkin_ws` root will fix or "bypass" these errors:
  ```
  sudo mkdir /include
  sed -i "s/Boost_INCLUDE_DIR/Boost_INCLUDE_DIRS/" src/universal_robot/ur_kinematics/CMakeLists.txt
  ```