UniversalRobots/Universal_Robots_ROS_Driver #434: Can't build the latest beta UR Robot Driver

AlcatrAAz, 2021-07-16:
# Versions
 - ROS Driver version: 0.0.3, Melodic

# Problem

Hello 

I have a problem with building the latest ur_robot_driver ROS package. We got yesterday our new UR10e and after I noticed that you are restructured the new cartesian repositories in github, I decided to build my UR packages completely from scratch. So I deleted all UR packages and in my catkin folder the devel and build folders. I cloned the following gits:

git clone https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian.git
git clone https://github.com/UniversalRobots/Universal_Robots_ROS_passthrough_controllers.git
git clone -b beta-testing https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git
git clone -b beta-testing-boost https://github.com/UniversalRobots/Universal_Robots_Client_Library.git
git clone https://github.com/UniversalRobots/Universal_Robots_ROS_cartesian_control_msgs.git
git clone https://github.com/ros-industrial/ur_msgs.git
git clone https://github.com/UniversalRobots/Universal_Robots_ROS_scaled_controllers.git
git clone https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers.git
git clone https://github.com/fmauch/universal_robot

And when I try to build the ur_robot_driver I got the following error (see end of this post). It's quite confusing which packages we should now use for the latest beta driver. Because the main page here recommends [this](https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control/tree/beta-testing) and this contains for example the cartesian_ros_control package which [this](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian) repo doesn't contain.

I hope someone can help me or did I mixed up some of the packages?

Thanks in advance

> Errors     << ur_robot_driver:make /home/lg/catkin_ws/logs/ur_robot_driver/build.make.000.log                                                                                
In file included from /home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:28:0:
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:244:3: error: ‘cartesian_ros_control’ does not name a type
   cartesian_ros_control::CartesianStateInterface cart_interface_;
   ^~~~~~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:245:3: error: ‘cartesian_ros_control’ does not name a type
   cartesian_ros_control::TwistCommandInterface twist_interface_;
   ^~~~~~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:246:3: error: ‘cartesian_ros_control’ does not name a type
   cartesian_ros_control::PoseCommandInterface pose_interface_;
   ^~~~~~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h: In constructor ‘ur_driver::HardwareInterface::HardwareInterface()’:
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:282:20: warning: ‘ur_driver::HardwareInterface::cartesian_pose_command_’ will be initialized after [-Wreorder]
   urcl::vector6d_t cartesian_pose_command_;
                    ^~~~~~~~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:271:20: warning:   ‘urcl::vector6d_t ur_driver::HardwareInterface::joint_positions_’ [-Wreorder]
   urcl::vector6d_t joint_positions_;
                    ^~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:56:1: warning:   when initialized here [-Wreorder]
 HardwareInterface::HardwareInterface()
 ^~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function ‘virtual bool ur_driver::HardwareInterface::init(ros::NodeHandle&, ros::NodeHandle&)’:
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:336:23: error: ‘JointTrajectoryHandle’ is not a member of ‘hardware_interface’
   hardware_interface::JointTrajectoryHandle joint_trajectory_handle = hardware_interface::JointTrajectoryHandle(
                       ^~~~~~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:336:23: note: suggested alternative: ‘JointTrajectory’
   hardware_interface::JointTrajectoryHandle joint_trajectory_handle = hardware_interface::JointTrajectoryHandle(
                       ^~~~~~~~~~~~~~~~~~~~~
                       JointTrajectory
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:340:23: error: ‘using JointTrajectoryInterface = class hardware_interface::TrajectoryInterface<control_msgs::FollowJointTrajectoryGoal_<std::allocator<void> >, control_msgs::FollowJointTrajectoryFeedback_<std::allocator<void> > > {aka class hardware_interface::TrajectoryInterface<control_msgs::FollowJointTrajectoryGoal_<std::allocator<void> >, control_msgs::FollowJointTrajectoryFeedback_<std::allocator<void> > >}’ has no member named ‘registerHandle’; did you mean ‘registerGoalCallback’?
   jnt_traj_interface_.registerHandle(joint_trajectory_handle);
                       ^~~~~~~~~~~~~~
                       registerGoalCallback
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:340:38: error: ‘joint_trajectory_handle’ was not declared in this scope
   jnt_traj_interface_.registerHandle(joint_trajectory_handle);
                                      ^~~~~~~~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:343:23: error: ‘CartesianTrajectoryHandle’ is not a member of ‘hardware_interface’
   hardware_interface::CartesianTrajectoryHandle cartesian_trajectory_handle =
                       ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:343:23: note: suggested alternative: ‘CartesianTrajectory’
   hardware_interface::CartesianTrajectoryHandle cartesian_trajectory_handle =
                       ^~~~~~~~~~~~~~~~~~~~~~~~~
                       CartesianTrajectory
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:348:24: error: ‘using CartesianTrajectoryInterface = class hardware_interface::TrajectoryInterface<cartesian_control_msgs::FollowCartesianTrajectoryGoal_<std::allocator<void> >, cartesian_control_msgs::FollowCartesianTrajectoryFeedback_<std::allocator<void> > > {aka class hardware_interface::TrajectoryInterface<cartesian_control_msgs::FollowCartesianTrajectoryGoal_<std::allocator<void> >, cartesian_control_msgs::FollowCartesianTrajectoryFeedback_<std::allocator<void> > >}’ has no member named ‘registerHandle’; did you mean ‘registerGoalCallback’?
   cart_traj_interface_.registerHandle(cartesian_trajectory_handle);
[... 139 lines cut ...]
   pose_interface_.registerHandle(
   ^~~~~~~~~~~~~~~
   js_interface_
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:358:7: error: ‘cartesian_ros_control’ has not been declared
       cartesian_ros_control::PoseCommandHandle(cart_interface_.getHandle("tool0_controller"), &pose_command_));
       ^~~~~~~~~~~~~~~~~~~~~
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function ‘virtual void ur_driver::HardwareInterface::read(const ros::Time&, const ros::Duration&)’:
/home/lg/catkin_ws/src/universal_robots/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:562:27: error: ‘using JointTrajectoryInterface = class hardware_interface::TrajectoryInterface<control_msgs::FollowJointTrajectoryGoal_<std::allocator<void> >, control_msgs::FollowJointTrajectoryFeedback_<std::allocator<void> > > {aka class hardware_interface::TrajectoryInterface<control_msgs::FollowJointTrajectoryGoal_<std::allocator<void> >, control_msgs::FollowJointTrajectoryFeedback_<std::alloca [... cut ...]
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/ur_robot_driver_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/lg/catkin_ws/build/ur_robot_driver; catkin build --get-env ur_robot_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.............................................................................................................................................................................
Failed     << ur_robot_driver:make                              [ Exited with code 2 ]

stefanscherzinger, 2021-07-19:
Hi @AlcatrAAz 

You are right. It's a little confusing at the moment. Skimming over your list, I noticed that you need to switch branches on the driver and library repos:

- [Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/staging) (staging)
- [Library](https://github.com/UniversalRobots/Universal_Robots_Client_Library/tree/boost) (boost)

I'm not completely sure about the UR kinematics/calibration part at the current state. Maybe @fmauch can help.

AlcatrAAz, 2021-07-19:
Hey @stefanscherzinger 

thank you very much for your answer. Yes with Driver (staging) and Library (boost) I could build everything and the driver is working with the UR10e. I use now UR kinematics/calibration (calibration_devel) like in the Readme mentioned, that was my fault.

It would be nice (also for others) if you could update your readme when you change the necessary branches. [Here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/beta-testing#building) it is still beta-testing-boost and beta-testing for Driver and Library.

In the calibration [Readme](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/beta-testing/ur_calibration/README.md) there is also a wrong command for the beta Driver. It should be "roscp _**ur_robot_driver**_ ur3e_bringup.launch ur3e_1.launch"

fmauch, 2021-07-22:
the `beta-testing` branch is deprecated, hence the instructions there are wrong. I'll update the README accordingly.

PR #439 (2021-07-22): Fix package name in calibration example
The example was referencing the library package where it should have been
referencing the driver package.

Closes #434 

Thanks @AlcatrAAz for spotting this.
closed 2021-07-22 by fmauch