UniversalRobots/Universal_Robots_ROS_Driver #436: Improve RT installation guide

fmauch, 2021-07-19:
As risen [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/418#issuecomment-880348250) it might be a good idea to look at our RT-kernel installation guide, whether it could / should be improved.

felixvd, 2021-07-20:
FWIW, I didn't manage to follow the RT patch verification on first try, but would have felt better not skipping that step.

roy-augmentus, 2021-09-20:
Hi for the guide, it was important for me before compiling to change the 2 values to empty for CONFIG_SYSTEM_TRUSTED_KEYS=""
CONFIG_SYSTEM_REVOCATION_KEYS=""

Also I had to turn off secure boot in BIOS to choose the kernel. I skipped the section setup grub to boot rt kernel always. Maybe that part can be put optional?