UniversalRobots/Universal_Robots_ROS_Driver #440: Port Issues

Asad-Shahid, 2021-07-24:
# Summary
Hi, could someone explain how does the ports configuration work in the ur_robot_driver?
I am having an issue with connecting the ur10e to the remote computer due to the ports. All the IP configurations are set 
properly. 

# Issue details
When I try to bring up the robot using:

$ roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=192.168.100.195

It fails almost all the time giving an error about the communication, like connection refused or could not be established. 
I looked at the similar issues in some threads but was not able to bring up the robot. 

So I started changing the ports randomly e.g. changing 'script_sender_port' from default 50002 to 50003 or 50001 and it worked. External control started successfully. But then with the exact same configuration of the ports, I was not able to bring up the robot the second time. The strange thing is that sometimes it works with 30001 or 30002 but other times it doesn't work with the same ports but works with e.g. 50002. Similarly, I also changed the 'reverse_port' from 50001 to 50003 or 30001 etc and it was the similar effect. 

I am really confused about whats going on. The only way I am able to bring up the robot now is by keeping the 'reverse_port' fixed to 50001 and continuously trying changing the 'script_sender_port' to numbers like 50002, 50003, 30001, 30002, 30003. If I have enough luck, one configuration would work. 

Any suggestions, would be very much appreciated, Thanks

fmauch, 2021-07-27:
Mixing the ports around like you try doesn't seem to be a good idea. All the ports you tried are being used by other parts of the driver (at least the 5000x on the ROS machine).

The [`ur_client_library`](https://github.com/UniversalRobots/Universal_Robots_Client_Library) has an [architecture overview](https://raw.githubusercontent.com/UniversalRobots/Universal_Robots_Client_Library/master/doc/dataflow.svg) of the ports being involved (The trajectory port on 50003 is missing there, see https://github.com/UniversalRobots/Universal_Robots_Client_Library/issues/82).

I would suggest not to change the ports as a start and find out why the default configuration doesn't work. For example reasons could be a firewall or other processes conflicting with the port definitions.

You can see the currently opened ports e.g. by using `ss -alpn`. An example output (filtered by port numbers 5000x) could be:

```
> $ ss -alpn | grep 5000
tcp   LISTEN   0     1        0.0.0.0:50001       0.0.0.0:*     users:(("ur_robot_driver",pid=78863,fd=18))
tcp   LISTEN   0     1        0.0.0.0:50002       0.0.0.0:*     users:(("ur_robot_driver",pid=78863,fd=14))
tcp   LISTEN   0     1        0.0.0.0:50003       0.0.0.0:*     users:(("ur_robot_driver",pid=78863,fd=21))
```

Secondly, the full output of the driver could be helpful to assist you with finding the source of the connection problems.

Asad-Shahid, 2021-07-27:
Thanks for the comment. If I try with the default ports configurations below is the error: It seems that the connection with the robot cannot be established right from the start

Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 204, in main
    resp = switch_controller(loaded, [], 2, False, 0.0)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1627374880.066841]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1627374880.067684]: Stopping all controllers...
[WARN] [1627374880.069203]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
================================================================================REQUIRED process [ur_hardware_interface-3] has died!
process has died [pid 27872, exit code -11, cmd /home/idsia/assassinn_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/idsia/.ros/log/79e374c4-eeb5-11eb-949a-9c7bef429eb8/ur_hardware_interface-3.log].
log file: /home/idsia/.ros/log/79e374c4-eeb5-11eb-949a-9c7bef429eb8/ur_hardware_interface-3*.log
Initiating shutdown!
================================================================================
[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[controller_stopper-6] killing on exit

Asad-Shahid, 2021-07-27:
Thanks for the comment. If I try with the default ports configurations below is the error: It seems that the connection with the robot cannot be established right from the start

```console
Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 204, in main
    resp = switch_controller(loaded, [], 2, False, 0.0)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1627374880.066841]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1627374880.067684]: Stopping all controllers...
[WARN] [1627374880.069203]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
================================================================================REQUIRED process [ur_hardware_interface-3] has died!
process has died [pid 27872, exit code -11, cmd /home/idsia/assassinn_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/idsia/.ros/log/79e374c4-eeb5-11eb-949a-9c7bef429eb8/ur_hardware_interface-3.log].
log file: /home/idsia/.ros/log/79e374c4-eeb5-11eb-949a-9c7bef429eb8/ur_hardware_interface-3*.log
Initiating shutdown!
================================================================================
[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[controller_stopper-6] killing on exit

```

Asad-Shahid, 2021-07-27:
Also the `ss -alpn | grep 5000` gives no output. Does it mean that firewall is keeping these ports closed?

```terminal
ss -alpn | grep 5000
 
```

fmauch, 2021-07-27:
Could you please provide the **full** output of the driver? The snippet you posted is not very much helpful.

Asad-Shahid, 2021-07-27:
Below is the full output:

```terminal
roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=192.168.100.195
... logging to /home/idsia/.ros/log/4a4e00a4-eec8-11eb-9d6c-9c7bef429eb8/roslaunch-idsia-32033.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://idsia:34967/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: melodic
 * /rosversion: 1.14.11
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_joint_traj_controller/state_publish_rate: 500
 * /scaled_pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /scaled_pos_joint_traj_controller/type: position_controll...
 * /scaled_vel_joint_traj_controller/action_monitor_rate: 20
 * /scaled_vel_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_vel_joint_traj_controller/constraints/elbow_joint/trajectory: 0.1
[... 216 lines cut ...]
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1627382960.683731]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1627382960.684573]: Stopping all controllers...
[WARN] [1627382960.686038]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
================================================================================REQUIRED process [ur_hardware_interface-3] has died!
process has died [pid 32067, exit code -11, cmd /home/idsia/assassinn_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/idsia/.ros/log/4a4e00a4-eec8-11eb-9d6c-9c7bef429eb8/ur_hardware_interface-3.log].
log file: /home/idsia/.ros/log/4a4e00a4-eec8-11eb-9d6c-9c7bef429eb8/ur_hardware_interface-3*.log
Initiating shutdown!
================================================================================
[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[controller_stopper-6] killing on exit
[ur_hardware_interface-3] killing on exit
[ros_control_stopped_spawner-5] killing on exit
[ros_control_controller_spawner-4] killing on exit
[robot_state_publisher-2] killing on exit
[INFO] [1627382960.804656]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1627382960.805723]: Stopping all controllers...
[WARN] [1627382960.807768]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
[controller_stopper-6] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

fmauch, 2021-07-27:
The `connection refused` messages you read are ROS-internal errors following by the driver crashing earlier.

In the past we experienced crashes due to ABI incompatibilities e.g. after updates. Before trying to debug why the driver crashes you could try to 
* Update your system using ´sudo apt update && sudo apt dist-upgrade`
* Rebuild your workspace (delete the `build` and `devel` folders beforehand)

If that doesn't help, you can try creating a stack trace as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/395#issuecomment-852374641).

xiaxiaxaia, 2025-02-18:
我遇到了类似的问题，通过tty清空了示教器的的防火强规则，已经可以正常连接了```bash sudo iptables -F```

urrsk, 2025-04-25:
> 我遇到了类似的问题，通过tty清空了示教器的的防火强规则，已经可以正常连接了

@xiaxiaxaia  thanks for the suggestion. Though, I will not suggest that. Since this issue was originally made, we have added some security features. One of them is that all the ports are closed by default.
You can turn them on using the security setting in the Teach pendant.

[image]

See https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/setup/robot_setup.html#robot-setup under the Polyscope 5 (E-series)