UniversalRobots/Universal_Robots_ROS_Driver #443: URCL log handler error

ganapathy-89, 2021-07-29:
# Summary

On building the package, I am getting multiple errors on the URCL Log handler file,

> Errors     << ur_robot_driver:make /home/aoiipc/catkin_ws/logs/ur_robot_driver/build.make.005.log                                                                                                          
In file included from /home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/urcl_log_handler.cpp:32:0:
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:43:31: error: ‘urcl’ has not been declared
 class UrclLogHandler : public urcl::LogHandler
                               ^
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:43:37: error: expected ‘{’ before ‘LogHandler’
 class UrclLogHandler : public urcl::LogHandler
                                     ^
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:45:1: error: expected primary-expression before ‘public’
 public:
 ^
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:45:1: error: expected ‘}’ before ‘public’
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:45:1: error: expected ‘,’ or ‘;’ before ‘public’
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:59:40: error: ‘urcl’ has not been declared
   void log(const char* file, int line, urcl::LogLevel loglevel, const char* message) override;
                                        ^
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:59:55: error: expected ‘,’ or ‘...’ before ‘loglevel’
   void log(const char* file, int line, urcl::LogLevel loglevel, const char* message) override;
                                                       ^
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:59:86: error: virt-specifiers in ‘log’ not allowed outside a class definition
   void log(const char* file, int line, urcl::LogLevel loglevel, const char* message) override;
                                                                                      ^
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:61:1: error: expected unqualified-id before ‘private’
 private:
 ^
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/urcl_log_handler.h:78:1: error: expected declaration before ‘}’ token
 }  // namespace ur_driver
 ^
make[2]: *** [CMakeFiles/urcl_log_handler.dir/src/urcl_log_handler.cpp.o] Error 1
make[1]: *** [CMakeFiles/urcl_log_handler.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In lambda function:
/home/aoiipc/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:396:39: error: ‘ur_msgs::SetPayload::Request {aka struct ur_msgs::SetPayloadRequest_<std::allocator<void> >}’ has no member named ‘mass’
             << " set_payload(" << req.mass << ", [" << req.center_of_gravity.x << ", " << req.center_of_gravity.y
                                       ^

> 

# Versions
 - ROS Driver version: ROS kinetic
 - ubuntu version : 14

 - Robot Serial Number: UR5e
 - Robot software version 5.1.1.40229
 - UR+ product(s) installed:  UniversalRobots/Universal_Robots_ROS_Driver
 - URCaps Software version(s): 1.0.5

fmauch, 2021-08-03:
Please note that ROS kinetic is no longer supported. For building a recent driver version, please use ROS melodic or ROS noetic.

closed 2021-08-03 by fmauch

ganapathy-89, 2021-08-03:
Thanks for the response 
Appreciated.

urrsk, 2021-08-03:
@ganapathy-89 it might be possible to checkout an older version of the driver to make it work with kinetic. (@fmauch please correct be if I am wrong)
In addition, I will also suggestion you to consider upgrading the robot's software. A lot of new thinks have changed since version 5.1

fmauch, 2021-08-03:
Yes, this might be possible, though you would also have to checkout an older version of the description repository, as well.

Additionally, it might be possible to compile the library from source, as well, as described in the [README](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#alternative-all-source-build), but I am not sure whether this might work.

If you can arrange it, I highly recommend to upgrade your ROS system to a supported version.

ganapathy-89, 2021-08-04:
Thanks @urrsk @fmauch 

I have upgraded my system to Ubuntu 20 and ROS to noetic and the package works fine to control with Moveit RVIZ.

May I know, does the package has any action server client script in C++ to control the UR Robot arm ? Thanks

fmauch, 2021-08-05:
We offer a python example in #437 

A c++ example should be rather similar. You can take the [action client tutorial](http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionClient) as a base and see https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/146195b3496857985b5b12a3eb33a0df928c2d81/ur_robot_driver/scripts/test_move#L70-L74 and https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/146195b3496857985b5b12a3eb33a0df928c2d81/ur_robot_driver/scripts/test_move#L97-L101 as a reference how to fill the action goal (The code is python, but it is rather similar in C++).