UniversalRobots/Universal_Robots_ROS_Driver #444: Is there any method to reset ftsensor in external control?

Ryoga723, 2021-08-01:
Hello,

I would like to reset ftsensor without disconnection such as pressing footer stop botton in teach pendant.
Previously, I knew rosservice which is named zero_ftsensor is server to reset ftsensor, and I try it now.

But for some reason I cannot reset ftsensor by call the service.
Concretely, I implement rosservice call /ur_hardware_interface/zero_ftsensor in terminal when I control robots by external pragram.
And I monitored the ftsensor data in rqt plot. But the value is not reseted.

rosservice call /ur_hardware_interface/zero_ftsensor return below.
success: True
message: ''"

So just by looking, I can reset the ftsensor value.
In the case I client service in program, the result is same as rosservice call.

I have no idea to reset ftsensor with controlling robots by external program.

In the first place, can I reset ftsensor by rosservice with controlling robots?

Please teach me any method if you know anything.
Thank you for your helping.

fmauch, 2021-08-03:
As the [docs state](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#zero_ftsensor-std_srvstrigger) an e-Series robot has to be in remote_control mode in order to use this service. Do you have n e-Series robot and did you do that?

Ryoga723, 2021-08-03:
Thank you for answering.

I use e-series robots and I'm aware of that you point out. I thought external mode and remote mode mean the same thing. I'm sorry I
misinterpreted.

Then, is there any way to reset ftsensor with controlling e-series robots by external control mode?

It doesn't have to be ros service.
Please teach me any method if you know anything.

fmauch, 2021-08-03:
See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/README.md#remote-control-mode for a short introduction to remote_control mode.

There is no such thing as "external control mode". The URCap delivered with this driver simply starts a program node on the robot that executes a set of script code that is provided by the ROS driver initially. The script code itself is running on the robot as any other program.

Taring the sensor sends a short script code snippet to the robot that will interrupt any running program on the robot.

Ryoga723, 2021-08-03:
In addition, I use ftsensor as  the meaning of the sensor, which measures the force published as /wrench.

urrsk, 2021-08-03:
Resetting/taring the force-torque sensor is not implement in the current version of the driver.

A workaround could be to do a local taring in ROS by simple save the current reading and subtract it afterwards going forward.
Another work around is to use the [ROS as a Service URCap](https://github.com/UniversalRobots/Universal_Robots_ROS_as_a_Service_URCap) and make a call back to the robot program, which you could program to zero the force-torque sensor as well.

Ryoga723, 2021-08-04:
Thank you for your answer.

I checked my execution environment and I understood what I running was local control mode. I use ROS usage in URCap and write main program by C++. So I understood the ftsensor is not reseted by rosservice in my environment.

Now then, is there any method to reset ftsensor in local control mode?

Currently I'm thinking as follows.
I would like to know what is excuted if ftsensor reset service is called. If its process is first case below, I can reset ftsensor by implementing the process in my program.

1. Without turnnig off the sensor, to correct the sensor's raw value numerically.
2. To turn off the sensor.

If its process is second case, I will put up with the inconvenience to operate teach pendant whenever I would like to reset the ftsensor.
Please teach me if you know which process is excuted.