UniversalRobots/Universal_Robots_ROS_Driver #460: UR5 external control cannot connect to ROS. Issue: No route to host.

Andrija987, 2021-09-01:
# Summary

*Introduction to the issue*

# Versions
 - ROS Driver version: Ubuntu 20.04, ROS noetic
 - Affected Robot Software Version(s): CB3
 - Affected Robot Hardware Version(s): UR5
 - URCaps Software version(s): externalcontrol-1.0.5.urcap

# Impact
I cannot connect the UR5 robot that we have in our lab to ROS. As it seems the communication between the UR5 and ROS PC could not be established, thus we cannot continue with our experiments and building important demo for real costumers.

# Issue details
When I run the UR cap for external control at TP it outputs: The connection to the remote PC at 192.160.0.100:5002 could not be established. Reason: No route to host.
I am struggling with this for couple of days. As it seems I cannot establish connection between the UR5 robot and PC. The connection other way around works i.e. I can connect to the robot from PC, but I cannot make the UR cap for external control to connect to the PC thus I cannot control the UR robot from ROS PC.

## Use Case and Setup
One UR5 robot and ROS PC. The use case is trying to control the UR5 robots with our own programs and python script we wrote. Ultimately we also want to control the UR robot by using Moveit.  

## Project status at point of discovered
I observe the issue as soon as come to a point where I want to control the UR5 robot.

## Steps to Reproduce
I tried literately everything and spend a few days searching on the web some solution, but all I tried failed. Below are the steps what I have tried and tested.

IP address of ROS PC: 192.160.0.100, mask: 255.255.255.0
I wrote the same IP address inside the External Control UR cap.
IP address of UR5 robot: 192.160.0.101, mask: 255.255.255.0

I connected the robot LAN cable directly to ROS PC, where Ubuntu is running (not a virtual machine).

First I wrote this line of command in Terminal:
`roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.160.0.101 [reverse_port:=REVERSE_PORT] kinematics_config:=$(rospack find ur_calibration)/etc/ur5_fact_cl.yaml`

It seemed that everting is fine. Here is the output from the terminal:
`SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 125
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
[... 390 lines cut ...]
`ss -atnp | grep 5000`

and got this:
`LISTEN      0        1                  0.0.0.0:50001            0.0.0.0:*       users:(("ur_robot_driver",pid=8238,fd=18))                                     
LISTEN      0        1                  0.0.0.0:50002            0.0.0.0:*       users:(("ur_robot_driver",pid=8238,fd=12))                                     
LISTEN      0        1                  0.0.0.0:50003            0.0.0.0:*       users:(("ur_robot_driver",pid=8238,fd=21))                                     
TIME-WAIT   0        0            192.160.0.100:34446      192.160.0.100:50002    `

I am not sure what all this meeans.

Then I thought maybe a firewall is blocking some ports, but this is not the case. I tested the firewal:
`sudo ufw status`
and it output this
`inactive`

Then I tired different IP address like:
ROS PC: 192.168.0.77 mask: 255.255.255.0
UR5: 192.168.0.100 mask: 255.255.255.0
didnt work.
Then I also tried:
ROS PC: 192.168.2.13 mask: 255.255.0.0
UR5: 192.168.2.11 mask: 255.255.0.0
didn't work.

No matter what I tired I could not make to work. I hope someone can help me. I am really struggling with this. Many thanks!

Andrija987, 2021-09-02:
Hi @urrsk . Many thanks for writing me and trying to help I really apricate it!

I am not so advanced user of ROS, thus please sorry if some the things I ask are a but silly. I don't know how to do `ping` from the robot to the PC and from the PC to the robot? Can you maybe write me what I should do?

If you think if I `ping` the IP address of a robot, by typing in the terminal `ping 192.160.0.101` then yes I have tried this and it received some packages. I did the same for the IP address of the PC i.e. the one that I wrote in the UR cap, I typed `ping 192.160.0.100` and again it seems that it works. 

Maybe I am not understanding correctly what you are suggesting that I do, thus feel free to tell me. Many thanks! Hope we will find some solution, I really cannot understand why it does not work. It seems that something is blocking the communication. I also don't know how to look inside the UR cap to check what is going on there?

fmauch, 2021-09-02:
> are you literally including [reverse_port:=REVERSE_PORT]?

The roslaunch output says 

>```
> * /ur_hardware_interface/reverse_port: 50001
> * /ur_hardware_interface/robot_ip: 192.160.0.101
> * /ur_hardware_interface/script_file: /home/andrija/cat...
> * /ur_hardware_interface/script_sender_port: 50002
>```

so normal port configuration seems to apply. As @gavanderhoorn wrote: If your popup really complains about `192.160.0.100:5002` then the port setup is missing a `0`.

Andrija987, 2021-09-02:
Hi @gavanderhoorn . Thanks as well for trying to help out.

Sorry I know that sometimes I am not making issues clear enough. Apologize for this.

I will start again to work with the UR robot soon and will post a picture from the TP what is written, a bit later. But I will try to answer your questions.

I have install the UR cap called: `externalcontrol-1.0.5.urcap` in the TP, following the instructions given [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md). Then  I followed all the steps that were written there. After installation and restarting the UR5 robot, I create a new program and insert the External Control program node into the program tree. I saved the program. Then when I loaded and run this program the message on the TP screen appears saying: `The connection to the remote PC at 192.160.0.100:50002 could not be established. Reason: No route to host.`

Sorry the 5002 was a typo in my comment and post here. I check several times in the UR cap. I used the default port 50002 in the UR cap.

For the second part of your question. Yes I literally included the `[reverse_port:=REVERSE_PORT]`. Because I followed the tutorial [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/301)

Maybe this is a mistake? Should I write something else for `REVERS_PORT`? To make things more clear I pasted this line in my Ubuntu PC terminal:
`roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.160.0.101 [reverse_port:=REVERSE_PORT] kinematics_config:=$(rospack find ur_calibration)/etc/ur5_fact_cl.yaml`

(the file `ur5_fact_cl.yaml` is okay, I have this).

fmauch, 2021-09-02:
If you don't require any special usecase then omitting the `reverse_port` parameter is fine.

Andrija987, 2021-09-02:
@fmauch thanks. No it was my typo in the comment. I assure you I have type the right port number in the UR cap, and I check this many times. But I can post a picture from TP so you can see. I really cannot make sense what I am doing wrong.

gavanderhoorn, 2021-09-02:
> Maybe this is a mistake? Should I write something else for `REVERS_PORT`? To make things more clear I pasted this line in my Ubuntu PC terminal:
> `roslaunch ur_robot_driver ur5_bringup.launch [...]

it's not necessarily going to fix your issue, but: drop the `[reverse_port:=REVERSE_PORT]` part from your command line.

Andrija987, 2021-09-02:
Thanks @fmauch and @gavanderhoorn  . I will try to omit once more and see if it will work. But since I tested so many things, I am pretty sure that I also tried to omit this as well.

I also tried to change the port numbers and see what will happen. I did this by typing this:
`roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.160.0.101 [script_sender_port:=50003] kinematics_config:=$(rospack find ur_calibration)/etc/ur5_fact_cl.yaml`
And tested also ports 50004, but again it didn't work.

Thanks the last suggestion I never tried. I will try what you have suggested and post what is the output.

> Then, we should take the path @urrsk suggested. You can login to your robot:
> 
> ```
> ssh root@192.160.0.101
> ```
> 
> The default password is `easybot`
> 
> There you can do `ping 192.160.0.100` and also the `nc 192.160.0.100 50002` test you did earlier.

gavanderhoorn, 2021-09-02:
> I also tried to change the port numbers and see what will happen. I did this by typing this:
> `roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.160.0.101 [script_sender_port:=50003] kinematics_config:=$(rospack find ur_calibration)/etc/ur5_fact_cl.yaml`

don't add those `[` and `]`. They are not part of the argument name.

They are used to indicate that `script_sender_port` is an *optional* argument.

Andrija987, 2021-09-02:
Thanks! I will experiment with is as well. Many many thanks for trying to help out. I will post soon what happened, also post some images of the setup and output form the TP.

fmauch, 2021-09-02:
Note that this very much sounds like a firewall problem. If your `nc` test on the controller fails I would be almost certain that it is a firewall problem. `ufw` is not the only firewall possible, though. It could also be `iptables` or `firewalld`, or even something else that I am not aware of.

Andrija987, 2021-09-02:
Hm....I see @fmauch . Thanks for noting this as well! I will do the test in hour or so, and post what happened. My felling is that you are right, as it all points out like something is blocking the communication, and it could be some firewall. I also tried to see if ports are open or closed (with some command that could not remember now, but will dig it up) and all ports were open. But I will post all that I tired.

I am really gratefully to all of you for trying to help!

gavanderhoorn, 2021-09-02:
@fmauch: "no route to host" suggests a routing problem. A firewall is not involved in routing (typically).

Routing problems would arise when the CB can't actually route a packet to the specific destination.

fmauch, 2021-09-02:
@gavanderhoorn you are probably right. I just tested using `iptables` and `ufw` to block 50002 and both times I got the popup "Connect timed out" from polyscope. Though it is URSim, so things could be different, there.

@Andrija987 running `ip route` on the robot controller might bring some insights.

Andrija987, 2021-09-02:
Hi again. It took time until I test out all the thing. Here is what I have tired.

Here is the image about IP address of the UR5 robot
[image]

Then image about IP address of the UR cap:
[image]

Then image about the message that is displayed at the TP when I try to run the external command program:
[image]

A screen shot from setting the IP address
[image]
 of the PC:

Then I tired to ping the IP address of the PC as described by @fmauch. Here is the output of that:
[image]

Then I tired the `nc 192.160.0.100 50002` test and it output the message `no route to host` here is the screen shot:
[image]

Then I wanted to tested out the `firewalld` and see what will output. I tired typing this:
`sudo firewall-cmd --list-all`
and this is the output form the terminal
```
public (active)
  target: default
  icmp-block-inversion: no
  interfaces: enp2s0 wls1
  sources: 
  services: dhcpv6-client ssh
  ports: 
  protocols: 
  masquerade: no
  forward-ports: 
  source-ports: 
  icmp-blocks: 
  rich rules: 
```
I don't know how to read this, but it seems to my like really that ports are not open and that only ipv6 services is allowed?

Because when I type this
`firewall-cmd --list-ports`
to see the list of open ports it seems that all the ports are closed, as noting is printed out of terminal

or I wanted to see which ssh services are open and then I typed this
`grep ssh /etc/services`
and the output from the terminal is
```
ssh		22/tcp				# SSH Remote Login Protocol
```
I also tried typing this:
`sudo systemctl status firewalld`
and the output from the terminal is this
```
firewalld.service - firewalld - dynamic firewall daemon
     Loaded: loaded (/lib/systemd/system/firewalld.service; enabled; vendor pre>
     Active: active (running) since Thu 2021-09-02 15:41:10 CEST; 1h 17min ago
       Docs: man:firewalld(1)
   Main PID: 989 (firewalld)
      Tasks: 2 (limit: 9469)
     Memory: 29.9M
     CGroup: /system.slice/firewalld.service
             └─989 /usr/bin/python3 /usr/sbin/firewalld --nofork --nopid

sep. 02 15:41:09 ifi-iMac systemd[1]: Starting firewalld - dynamic firewall dae>
sep. 02 15:41:10 ifi-iMac systemd[1]: Started firewalld - dynamic firewall daem>

```

Then I moved to test the `iptables` so I typed this:
`sudo iptables -S`

And this is the output:
```
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT
-N FORWARD_IN_ZONES
-N FORWARD_OUT_ZONES
-N FORWARD_direct
-N FWDI_public
-N FWDI_public_allow
-N FWDI_public_deny
-N FWDI_public_log
-N FWDI_public_post
-N FWDI_public_pre
-N FWDO_public
-N FWDO_public_allow
-N FWDO_public_deny
-N FWDO_public_log
-N FWDO_public_post
-N FWDO_public_pre
-N INPUT_ZONES
-N INPUT_direct
-N IN_public
-N IN_public_allow
-N IN_public_deny
-N IN_public_log
-N IN_public_post
-N IN_public_pre
-N OUTPUT_direct
-A INPUT -m conntrack --ctstate RELATED,ESTABLISHED,DNAT -j ACCEPT
-A INPUT -i lo -j ACCEPT
-A INPUT -j INPUT_direct
-A INPUT -j INPUT_ZONES
-A INPUT -m conntrack --ctstate INVALID -j DROP
-A INPUT -j REJECT --reject-with icmp-host-prohibited
-A FORWARD -m conntrack --ctstate RELATED,ESTABLISHED,DNAT -j ACCEPT
-A FORWARD -i lo -j ACCEPT
-A FORWARD -j FORWARD_direct
-A FORWARD -j FORWARD_IN_ZONES
-A FORWARD -j FORWARD_OUT_ZONES
-A FORWARD -m conntrack --ctstate INVALID -j DROP
-A FORWARD -j REJECT --reject-with icmp-host-prohibited
-A OUTPUT -o lo -j ACCEPT
-A OUTPUT -j OUTPUT_direct
-A FORWARD_IN_ZONES -g FWDI_public
-A FORWARD_OUT_ZONES -g FWDO_public
-A FWDI_public -j FWDI_public_pre
-A FWDI_public -j FWDI_public_log
-A FWDI_public -j FWDI_public_deny
-A FWDI_public -j FWDI_public_allow
-A FWDI_public -j FWDI_public_post
-A FWDI_public -p icmp -j ACCEPT
-A FWDO_public -j FWDO_public_pre
-A FWDO_public -j FWDO_public_log
-A FWDO_public -j FWDO_public_deny
-A FWDO_public -j FWDO_public_allow
-A FWDO_public -j FWDO_public_post
-A INPUT_ZONES -g IN_public
-A IN_public -j IN_public_pre
-A IN_public -j IN_public_log
-A IN_public -j IN_public_deny
-A IN_public -j IN_public_allow
-A IN_public -j IN_public_post
-A IN_public -p icmp -j ACCEPT
-A IN_public_allow -p tcp -m tcp --dport 22 -m conntrack --ctstate NEW,UNTRACKED -j ACCEPT
```

I presume this all means that my ports are not open and firewall is blocking the Ipv4 communication. I am trying to figure out if and how I can disable `firewall` or allow the communication thought some ports. But I would really appreciate if someone has further ideas and suggestions.

Ah lastly I don't know how to run  `ip route` on the robot controller?
Thanks!

fmauch, 2021-09-02:
You can run `ip route` just as the `ping` command in the ssh session.

To (temporarily) disable firewalld you can probably do 
```
sudo systemctl stop firewalld
```

or use some `firewall-cmd` command (I am not very familiar with firewalld).

Andrija987, 2021-09-02:
Thanks @fmauch . 

I disabled the `firewall` and it worked!!! Many thanks!!! Now everything works and I can control the UR5 from the PC.

At the end apparently this was the issue and the firewall blocked the ports. Thanks again to all of you for helping out.

I will close this issue now.

closed 2021-09-02 by Andrija987

gavanderhoorn, 2021-09-02:
> I disabled the `firewall` and it worked

hm, interesting.

Apparently the firewall worked so well it made the UR CB's ip-stack conclude the destination IP didn't exist / was unreachable.

Andrija987, 2021-09-02:
Thanks @gavanderhoorn 

Yes I really don't know what is that. When I typed `sudo systemctl stop firewalld` and stopped the firewalld then everything worked.

It could be as I am using a University computer that this is somehow set to be a strong protection, but luckily they give me admin access so I could modify all this.

Many thanks to all of you!

Andrija987, 2021-09-02:
Hope the issues I had and all of your help to solve it will help others in the future.

hf-zhao, 2022-03-21:
I have encountered the same problem，when I execute the following command,
```
root@ur-2021351267:~# nc 192.168.124.102 50002
(UNKNOWN) [192.168.124.102] 50002 (?) : Connection refused
```
At the same time, this command is useless to me.
```
sudo systemctl status firewalld
Unit firewalld.service could not be found.
```
This is my environment:
ROS驱动版本：Ubuntu 20.04，ROS noetic
受影响的机器人软件版本：CB3
受影响的机器人硬件版本：UR5
URCaps 软件版本：externalcontrol-1.0.5.urcap

I need help

fmauch, 2022-03-24:
@hf-zhao please have a look at the connection debugging guide I [recently posted}(https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/510#issuecomment-1062879270). If that doesn't help you, then please open a new issue.

urrsk, 2022-03-24:
> Thanks @gavanderhoorn
> 
> Yes I really don't know why is that. When I typed `sudo systemctl stop firewalld` and stopped the firewalld then everything worked.
> 
> It could be as I am using a University computer that this is somehow set to be a strong protection, but luckily they give me admin access so I could modify all this.
> 
> Many thanks to all of you!

@Andrija987 Just to ensure that I understand you correctly; You disabled the firewall on your Ubuntu computer, running the UR ROS Driver, to make it work?

hf-zhao, 2022-03-25:
@fmauch Although I have solved, but I am very grateful for your help.
I connected the network cable directly between my computer and UR5, with wifi turned off, and this worked

Andrija987, 2022-03-25:
Hi @urrsk ,

Thanks for the question. Yes that is exactly what I did. When I disabled the firewall using the command I posted above, my Ubuntu computer, running the UR ROS Driver worked. And all worked nicely since then. I never had a problem after that or when I turned off the firewall.

@hf-zhao Nice that you made it work! Sorry for not answering your messages before.