UniversalRobots/Universal_Robots_ROS_Driver #461: Adding extra axis control on the ROS driver

akdhandy, 2021-09-02:
Hello,

I'm trying to add a linear slider onto which my UR5e arm is attached in my lab. How should I go about passing on the 7DoF solution from MoveIt Planner onto the ur_driver since the driver could take only the 6DoF solution?

fmauch, 2021-09-16:
For the MoveIt! part I cannot really help you, I'm afraid. However, as you seem to have gotten it running in simulation it seems that MoveIt! is not your problem.

I further assume that you are missing a trajectory action server controlling all 7 actuators?

If that is the case, you should probably create a custom hardware interface using a [combined_robot_hw](http://wiki.ros.org/combined_robot_hw) (See [this example](https://github.com/ros-controls/ros_control/wiki/Writing-CombinedRobotHW)). The `HardwareInterface` class of this node should be prepared for getting integrated into a combined interface.

Once this is done, this should be usable together with MoveIt!

gavanderhoorn, 2022-01-25:
Out of curiosity: has anyone implemented / experimented with @fmauch's suggestion?

Most (all?) external tracks available for URs seem to come with urcaps which can't guarantee / provide synchronised motion.

The `CombinedRobotHW` approach could potentially achieve it (although it won't be as good as a native implementation (ie: everything controlled by `URControl`)).

I'm curious to know whether anyone has tried this and what their experiences were.

aatb-ch, 2022-02-02:
We are currently developing such thing, but are also at the same stage, we have a custom hardware interface for a servo drive controlling the track, and have been running trajectories separately by sending the UR and track to each's action servers with identical start times. It's definitely a temporary solution currently as for our use-case we must guarantee coordinated motion or risk severe crashes if one or the other stops during trajectory execution, so we are about to start writing a CombinedRobotHW for it.

Now that I get to think about it, we are also starting to integrate cartesian controllers for our UR10s, how would that behave alongside a CombinedRobotHW (just curious if you have any thoughts) ?

@akdhandy which servo manufacturer, drive, fieldbus, plc are you using?

Indeed all commercial 7-axis linear track solutions are not synchronised so useless for our use-case, we've actually suggested multiple times to UR that it would be fantastic to have native external axis support (track and rotary). We've recently had to use Elite robots for a temporary project in Shanghai (we had lot of reserves about it, basically customer forced us to use these), but were quite amused to see that they supported natively 2 external axises and VNC server out-of-the-box :)

Mohatashem, 2022-12-23:
Hello,

Has anyone succeeded in adding a linear axis using `CombinedRobotHW`?

I have tried to add an external axis to the UR and need some help/suggestions from anyone who has succeeded. I’ve already written a ROS-Driver for the linear axis and am able to control it in ROS. After going through [combined_robot_hw]( http://wiki.ros.org/combined_robot_hw), and the [example](https://github.com/ros-controls/ros_control/wiki/Writing-CombinedRobotHW), I was able to implement ‘CombinedRobotHardware’ in a simple simulation joining two linear axes into a 2-DOF planar mechanism.

However, looking at how elaborate the driver’s [main node](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/hardware_interface_node.cpp) and [`hardware_interface`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/hardware_interface.cpp) are, I was wondering how different would it be from the `combo_control_node.cpp` in the example implementation [here](https://github.com/ros-controls/ros_control/wiki/Writing-CombinedRobotHW)?

Do we use the ‘hardware_interface` of the UR **as is**? Do we need to make changes? What do we need to consider?

I have had difficulty implementing the `CombinedRobotHardware` for a UR10e  + external axis. On launching the bringup, the following error is thrown continuously:

```
[ERROR] [1671728447.435445624] [/combined_hardware_interface] [ros.zerog_ur_robot_driver.ur_client_library]: Pipeline producer overflowed! <RTDE Data Pipeline>

```

# My Setup:
- Ubuntu 18.04
- ROS Melodic

# What I have so far:

1.	What my terminal looks like:
```terminal
... logging to /home/pp1744/.ros/log/2b6e739e-821a-11ed-8127-a0291986156e/roslaunch-pp1744-Precision-5550-3360.log

started roslaunch server http://localhost:41803/

SUMMARY
========

PARAMETERS
 * /combined_hardware_interface/ceiling_robot/joints: ['ceiling_arm_sho...
 * /combined_hardware_interface/ceiling_robot/type: ur_driver/Hardwar...
 * /combined_hardware_interface/cr/joints: ['ceiling_track_s...
 * /combined_hardware_interface/cr/type: ur_driver/cr
 * /combined_hardware_interface/headless_mode: False
 * /combined_hardware_interface/input_recipe_file: /home/pp1744/rosw...
 * /combined_hardware_interface/kinematics/forearm/pitch: 0
 * /combined_hardware_interface/kinematics/forearm/roll: 0
 * /combined_hardware_interface/kinematics/forearm/x: -0.6127
 * /combined_hardware_interface/kinematics/forearm/y: 0
 * /combined_hardware_interface/kinematics/forearm/yaw: 0
 * /combined_hardware_interface/kinematics/forearm/z: 0
 * /combined_hardware_interface/kinematics/hash: calib_51197013707...
 * /combined_hardware_interface/kinematics/shoulder/pitch: 0
 * /combined_hardware_interface/kinematics/shoulder/roll: 0
 * /combined_hardware_interface/kinematics/shoulder/x: 0
 * /combined_hardware_interface/kinematics/shoulder/y: 0
 * /combined_hardware_interface/kinematics/shoulder/yaw: 0
 * /combined_hardware_interface/kinematics/shoulder/z: 0.1807
 * /combined_hardware_interface/kinematics/upper_arm/pitch: 0
 * /combined_hardware_interface/kinematics/upper_arm/roll: 1.570796327
 * /combined_hardware_interface/kinematics/upper_arm/x: 0
 * /combined_hardware_interface/kinematics/upper_arm/y: 0
 * /combined_hardware_interface/kinematics/upper_arm/yaw: 0
 * /combined_hardware_interface/kinematics/upper_arm/z: 0
 * /combined_hardware_interface/kinematics/wrist_1/pitch: 0
 * /combined_hardware_interface/kinematics/wrist_1/roll: 0
 * /combined_hardware_interface/kinematics/wrist_1/x: -0.57155
 * /combined_hardware_interface/kinematics/wrist_1/y: 0
 * /combined_hardware_interface/kinematics/wrist_1/yaw: 0
 * /combined_hardware_interface/kinematics/wrist_1/z: 0.17415
 * /combined_hardware_interface/kinematics/wrist_2/pitch: 0
 * /combined_hardware_interface/kinematics/wrist_2/roll: 1.570796327
 * /combined_hardware_interface/kinematics/wrist_2/x: 0
 * /combined_hardware_interface/kinematics/wrist_2/y: -0.11985
 * /combined_hardware_interface/kinematics/wrist_2/yaw: 0
 * /combined_hardware_interface/kinematics/wrist_2/z: -2.45816459076e-11
[... 237 lines cut ...]
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

2. I am able to load the combined controller for both the robot and the rail.
[image]

3. Here is my rqt_graph:
[image]

4. And here is the robot behaviour (apologies for the vertical filming):

https://user-images.githubusercontent.com/40743268/209245347-8b721bf0-b2a4-490d-92f1-f9c15fa20bbf.mp4

In the video, I was playing with the `rqt_joint_trajectory_controller`. The arm shakes when I try to move a joint on the arm (In another trial, I was able to move the `wrist_3_joint` to a desired position but it kept shaking). The rail joint, however, moves smoothly to the desired position.

# What I’ve done so far:

I created a separate package as shown in the example for `CombinedRobotHW`. I’ve used the `hardware_interface.cpp` along with other files from the `ur_robot_driver` package, required to run the robot successfully. Once I was sure that my package worked fine, and I could control both pieces of hardware from within this package. I proceeded to work towards a combined implementation. I am willing to share the main node of the code if it can help solve the problem.

Any suggestion/nudge in the right direction is appreciated.
Thank you

closed 2023-04-29 by github-actions[bot]