UniversalRobots/Universal_Robots_ROS_Driver #463: Use of invalid name(s): __limits

yangzion, 2021-09-09:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

# Versions
 - ROS Driver version: master
 - Affected Robot Software Version(s): URSoftware 3.15.2.106179
 - Affected Robot Hardware Version(s): UR10
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*
I am trying to connect UR10 with ROS driver. 
When I input 
roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=xx.xx.xx.xx

A long error pumped up:
# Issue details
*This part is important in order to clarify the severity of the issue.*

Use of invalid name(s):  __limits 
when evaluating expression '__limits['shoulder_pan']['min_position']'
when instantiating macro: read_model_data (/home//urdrive_ws/src/fmauch_universal_robot/ur_description/urdf/inc/ur_common.xacro)
instantiated from: ur_robot (/home//urdrive_ws/src/fmauch_universal_robot/ur_description/urdf/inc/ur_macro.xacro)
in file: /home//urdrive_ws/src/fmauch_universal_robot/ur_description/urdf/ur.xacro
RLException: while processing /home//urdrive_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/launch/ur_common.launch:
while processing /home//urdrive_ws/src/fmauch_universal_robot/ur_description/launch/load_ur10.launch:
while processing /home//urdrive_ws/src/fmauch_universal_robot/ur_description/launch/load_ur.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '/home/zhaolab/urdrive_ws/src/fmauch_universal_robot/ur_description/urdf/ur.xacro', 'joint_limit_params:=/home//urdrive_ws/src/fmauch_universal_robot/ur_description/config/ur10/joint_limits.yaml', 'kinematics_params:=/home//urdrive_ws/src/fmauch_universal_robot/ur_description/config/ur10/default_kinematics.yaml', 'physical_params:=/home//urdrive_ws/src/fmauch_universal_robot/ur_description/config/ur10/physical_parameters.yaml', 'visual_params:=/home//urdrive_ws/src/fmauch_universal_robot/ur_description/config/ur10/visual_parameters.yaml', 'transmission_hw_interface:=hardware_interface/PositionJointInterface', 'safety_limits:=false', 'safety_pos_margin:=0.15', 'safety_k_position:=20']] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro '$(find ur_description)/urdf/ur.xacro'     joint_limit_params:=$(arg joint_limit_params)     kinematics_params:=$(arg kinematics_params)     physical_params:=$(arg physical_params)     visual_params:=$(arg visual_params)     transmission_hw_interface:=$(arg transmission_hw_interface)     safety_limits:=$(arg safety_limits)     safety_pos_margin:=$(arg safety_pos_margin)     safety_k_position:=$(arg safety_k_position)" name="robot_description"/>
The traceback for the exception was written to the log file

## Use Case and Setup
I am using Ubuntu 18.04 + ROS melodic.
*What does the setup look like and what are the objectives?*

## Project status at point of discovered
*When did you first observe the issue?*
 - *Before I started?*
 - *In first couple of tries?*
 - *In normal use?*
 - *etc...*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

fmauch, 2021-09-09:
Yes, I can confirm this. Will fix it asap

gavanderhoorn, 2021-09-09:
@fmauch: https://github.com/ros/xacro/commit/9941961b2ac5eb91822c6c50c251ed094af24d6b

fmauch, 2021-09-09:
As this is description related, will track this in https://github.com/fmauch/universal_robot/issues/18

closed 2021-09-09 by fmauch

fmauch, 2021-09-09:
@yangzion could you please try out https://github.com/fmauch/universal_robot/pull/19?

udopa, 2021-09-10:
@fmauch I had the same problem and the fix works for me.