UniversalRobots/Universal_Robots_ROS_Driver #464: I have something wrong with main thread

tttshao666, 2021-09-10:
[ERROR] [1631239742.680577697]: Unsuccessful in setting main thread realtime priority. Error code: 1
[ERROR] [1631239742.680666131]: Main thread: Scheduling is NOT SCHED_FIFO!

fmauch, 2021-09-10:
As the error states, it can't schedule the main thread with real-time priority. Please follow our [real-time installation guide](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md) in order to setup a real-time enabled kernel correctly.

Note: You can choose to skip this step and ignore the error, but that might lead to undesired side-effects in some cases. It might work without a real-time kernel for testing purposes, but for a production environment we really encourage you to setup a real-time kernel.

roy-augmentus, 2021-09-20:
I had this issue previously also, What I had to do was to sudo gedit /etc/security/limits.conf to make sure it had the 
@realtime soft rtprio 99
@realtime soft priority 99
@realtime soft memlock 102400
@realtime hard rtprio 99
@realtime hard priority 99
@realtime hard memlock 102400

closed 2023-07-28 by github-actions[bot]