UniversalRobots/Universal_Robots_ROS_Driver #465: Remapping within included launch file is ignored

mqt0029, 2021-09-14:
# Summary

When trying to spawn Gazebo and MoveIt! simultaneously from a single launch file via `<include>`, it seems the `<remap>` in `urx_moveit_planning_execution.launch` is ignored, causing MoveIt to failed to spawn appropriate controllers.

# Versions
 - ROS Version: Noetic
 - ROS Driver version: master
 - OS: Ubuntu 20.04 Docker container

# Impact
MoveIt! cannot spawn necessary controllers to control the arm in Gazebo.

# Issue details
Receiving `[ WARN] [1631588665.519291600, 5.165000000]: Waiting for scaled_pos_joint_traj_controller/follow_joint_trajectory to come up` when using the `sim:=true` options with `urx_moveit_planning_execution.launch` in the nested launch file. All launch files worked normally when running using separated terminals.

## Use Case and Setup
Follow standard procedure on the main `README.md` and all files worked accordingly when using separate terminals, but when combined to one launch file, received described error.

Here's what the combined launch file looks like:
```XML
<launch>
    <arg name="robot" default="ur5"/>
    <arg name="gui"   default="false"/>
    <arg name="sim"   default="true"/>
    <arg name="debug" default="false" />

    <include file="$(find ur_gazebo)/launch/$(arg robot)_bringup.launch" pass_all_args="true"/>

    <include file="$(eval find(arg('robot') + '_moveit_config') + '/launch/ur5_moveit_planning_execution.launch')" pass_all_args="true"/>
</launch>
```
## Expected Behavior
Everything spawns correctly as it would in independent terminals i.e. the `scaled_pos_joint_traj_controller` should be remapped to `pos_joint_traj_controller`

## Actual Behavior
Controllers failed to spawn and cannot control simulated arm. The following error was received:
in which the following error were received:
```
[ WARN] [1631588665.519291600, 5.165000000]: Waiting for scaled_pos_joint_traj_controller/follow_joint_trajectory to come up
[ WARN] [1631588671.540738300, 11.167000000]: Waiting for scaled_pos_joint_traj_controller/follow_joint_trajectory to come up
[ERROR] [1631588677.559732600, 17.167000000]: Action client not connected: scaled_pos_joint_traj_controller/follow_joint_trajectory
[ INFO] [1631588677.582869000, 17.192000000]: Returned 0 controllers in list
[ INFO] [1631588677.591651400, 17.202000000]: Trajectory execution is managing controllers
[ INFO] [1631588677.591705900, 17.202000000]: MoveGroup debug mode is OFF
```

## Workaround Suggestion
Patiently waiting for a single include file answer... 🙏 Otherwise proceed as you would with using multiple terminals. Purpose of trying to use single include file is to systematically launch both the simulator and the MoveIt! backend along with some other software from a single executable i.e. streamlining the spawning process.

fmauch, 2021-09-15:
As this seems to be only concerning moveit and gazebo, this seems the wrong repository.

fmauch, 2021-09-15:
You can try breaking your problem down into a small example, where you connect two small launchfiles with a remap in one of them and see if they behave the same. Next, you could ask on ROS answers for help.

closed 2021-09-15 by fmauch