UniversalRobots/Universal_Robots_ROS_Driver #468: UR10 'base' frame in URDF does not match teach pendant Move screen coordinates or URSim coordinates

mzucker, 2021-09-16:
# Summary

There seems to be a discrepancy of approximately 400mm in the Z axis between the forward kinematics that appear in the teach pendant "Move" display, and the forward kinematics reported to ROS by the driver when requesting the transform between `'base'` and `'tool0_controller'`

# Versions
 - ROS Driver version: ROS Noetic, driver commit 54e24c03f07e1a3c6c92569dcb24e51cc013b976
 - Affected Robot Software Version(s): URSoftware 3.15.1.106171 (Mar 29 2021) ; 
 - Affected Robot Hardware Version(s): UR10
 - Robot Serial Number: 2018300027
 - UR+ product(s) installed:
 - URCaps Software version(s): external control

# Impact

Ideally, we would like 100% consistency between the ROS world and what is seen on the teach pendant. It is reasonably important that a world-frame XYZ position reported on the pendant be meaningful when doing planning, etc. in ROS.

# Issue details

## Project status at point of discovered

Discovered this when writing some joystick teleop code for UR10. I wanted to make sure that the FK was being computed correctly by my ROS code and could not get my FK to match the reported robot coordinates.

## Steps to Reproduce

On the ROS side, obtained a robot calibration [per the instructions](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information). Then ran the bringup for UR10:

    roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=10.44.3.55 kinematics_config:=/path/to/ur10_calibration.yaml

Look at the pendant and see the following:

  * Installation -> TCP Configuration: All zeros except for Z = 250.0 mm
  * Move -> Joints: -90, -110, -130, 60, 90, 0; X: -164.86 mm, Y: -836.14 mm, Z: -77.33 mm

Then echo the transform from `base` to `tool0_controller`:

    rosrun tf tf_echo base tool0_controller 

The output contains`Translation: [-0.165, -0.836, 0.323]`. 

X & Y agree with the teach pendant. Z is off by ~400mm.

I can similarly reproduce this in the simulator as well, although FK values are slightly different than real robot for the same joint configuration (possibly due to calibration offsets?).

Simulated teach pendant FK: [-174.15 mm, -861.88 mm, -18.68mm], ROS Translation: [-0.174, -0.862, 0.381].

## Expected Behavior

ROS TF output matches reported XYZ on Move screen in teach pendant or simulation window.

## Actual Behavior

Seemingly persistent offset of ~400 mm in Z between ROS transforms and Move screen.

urrsk, 2021-09-17:
@mzucker I believe you will get the expected tool position, If you go to the move tab of the robot's teach pendant and select the 'Base' frame/feature instead of the 'View'.

It is a good question why the Z position is off by 400mm in the view screen. Thanks for asking!

mzucker, 2021-09-20:
I can confirm on the real teach pendant that changing the Feature from "View" to "Base" results in FK that matches what is in ROS, thanks. 

Do you happen to know what coordinate frame the robot's safety planes [nx, ny, nz, distance] are defined relative to? Is it Base?

Also, what function does the "View" frame provide?

Thanks very much!

-Matt

urrsk, 2021-09-21:
> Do you happen to know what coordinate frame the robot's safety planes [nx, ny, nz, distance] are defined relative to? Is it Base?

Yes, it is the Base frame

> Also, what function does the "View" frame provide?

In the "Move" tab's "View" feature mode, you can change the angle of view of the virtual robot by using the touch screen and drag it around. 
Handy way to make the Cartesian move arrows correspond to the operators point of view of the robot

mzucker, 2021-09-21:
Got it, thanks!

closed 2021-09-21 by mzucker