UniversalRobots/Universal_Robots_ROS_Driver #470: Repeated "No trajectory defined at current time" error message when replanning

mzucker, 2021-09-20:
# Summary

When sending overlapping trajectory snippets to the UR Robot Driver via the `/pos_joint_traj_controller/follow_joint_trajectory` interface, occasionally there is an error message printed in red in the UR Robot Driver output that reads:

    Unexpected error: No trajectory defined at current time. Please contact the package maintainer.

From some research into the source code, the error message looks like it is defined in [JointTrajectoryController::update](https://github.com/ros-controls/ros_controllers/blob/noetic-devel/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L381), which seems to be called (indirectly) by the UR driver's [hardware interface node](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/hardware_interface_node.cpp#L151).

# Versions
  - ROS Driver version: Both ROS Melodic and ROS Noetic, driver commit 54e24c0
  - Affected Robot Software Version(s): URSoftware 3.15.1.106171 (Mar 29 2021) ;
  - Affected Robot Hardware Version(s): UR10
  - Robot Serial Number: 2018300027
  - UR+ product(s) installed:
  - URCaps Software version(s): external control

# Impact
It is unclear whether this error message indicates any important any bugs/malfunctions, or whether it is benign.

# Issue details

## Use Case and Setup

The use case is sending trajectory snippets to the `FollowJointTrajectory` interface at a fixed replanning interval (e.g. 0.05 s) Each trajectory snippet starts exactly one replanning interval after the previously sent snippet, and overlaps the previous one (e.g. each is 0.20 s long). 

So for example, the first trajectory snippet could go from t=0.00 to t=0.20 seconds, the second from t=0.05 to t=0.25, the third from t=0.10 to t=0.30, etc.

This buffering/replanning strategy is not uncommon in robotics (e.g. receding horizon planning or [model predictive control](https://en.wikipedia.org/wiki/Model_predictive_control)).

It ought to also be fully compatible with the trajectory replacement policy outlined in the [ROS wiki](http://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement).

## Project status at point of discovered

We were developing an interactive replanner for teleoperation and saw this error message sporadically. We then tried to make a simplified standalone ROS node which could reliably reproduce the error.

## Steps to Reproduce

  1. Download and build the `joint_replanner` node from <https://github.com/positronix-ai/joint_replanner>.

  2. Then run the UR robot driver on a real or simulated robot (both can trigger the error message but it seems more frequent with real hardware than the simulator). Make sure the pos_joint_traj_controller is active. For example: 

    roslaunch ur_robot_driver ur10_bringup.launch \
      robot_ip:=MY_ROBOT_IP kinematics_config:=MY_KINEMATICS.yaml \
      controllers:="joint_state_controller pos_joint_traj_controller" \
      stopped_controllers:="scaled_pos_joint_traj_controller"

  3. Run the program on the teach pendant that uses the "External Control" URCap.

  4. Finally, run the node you just built:

    rosrun joint_replanner joint_replanner

  5. You should see the wrist 1 joint move and then see the error message printed before too long. If not, try running the `stress` command (`sudo apt install stress`) with increasing numbers of CPU threads until you see the error message (e.g. `stress --cpu 2`).

## Expected Behavior

With the linked test program, the robot should wiggle its wrist 1 joint plus/minus 0.1 rad (approx. 6 deg) at a period of 2 seconds until Ctrl + C is pressed.

## Actual Behavior

The robot does move smoothly as programmed; however, the error message appears in the UR Robot Driver console output sporadically.

## Workaround Suggestion

No workarounds found yet but it does seem like reducing the frequency of trajectory points from 1000 Hz to 100 Hz (adjust `traj_dt` in `joint_replanner.cpp`) makes the message substantially less likely to appear. This, combined with the fact that the message seems to be correlated to CPU load, could possibly suggest some type of race condition or timing error.

fmauch, 2021-09-28:
I'm not sure whether this is inside this driver's scope. Could you maybe verify that this works with another hardware interface implementation, such as a [ros_control_boilerplate](https://github.com/PickNikRobotics/ros_control_boilerplate) interface?

mzucker, 2021-09-28:
Confirmed that this reproduces against ros_control_boilerplate. Steps to reproduce:

    roslaunch ros_control_boilerplate rrbot_simulation.launch
    rosrun joint_replanner joint_replanner \
      _joint_states_name:=/rrbot/joint_states \
      _traj_client_name:=/rrbot/position_trajectory_controller/follow_joint_trajectory \
      _wiggle_joint:=joint1

Though anecdotally it seems much harder / less frequent to reproduce with ros_control_boilerplate than with the UR driver.

I do agree that if the error message is printed by both projects, it is more likely that the underlying issue is in the common ros_controllers code. I will close this and go file an issue over there!

closed 2021-09-28 by mzucker

mzucker, 2021-09-28:
Reported at https://github.com/PickNikRobotics/ros_control_boilerplate/issues/56 and also at https://github.com/ros-controls/ros_controllers/issues/143#issuecomment-929542856.