UniversalRobots/Universal_Robots_ROS_Driver #471: Does moveit_config launch file override calibration offsets?

mzucker, 2021-09-20:
# Summary

I noticed that by default, many of the moveit config launch files from <https://github.com/fmauch/universal_robot/tree/calibration_devel> re-load the robot description.

I'm not sure whether this interferes with the calibration offsets loaded by the UR Robot Driver if the moveit config is launched after the robot driver.

# Versions
 - ROS Driver version: Both ROS Melodic and ROS Noetic, driver commit 54e24c0; universal_robot calibration_devel commit 1ffd6918
 - Affected Robot Software Version(s): URSoftware 3.15.1.106171 (Mar 29 2021) 
 - Affected Robot Hardware Version(s): UR10
 - Robot Serial Number: 2018300027
 - UR+ product(s) installed:
 - URCaps Software version(s): external control

# Impact

Trying to establish best practices for launching robot driver + MoveIt.

# Issue details

There seem to be two independent robot descriptions from the driver and from MoveIt. Not sure which one should be the final one for planning/monitoring of robot execution.

## Use Case and Setup

Basically just trying to figure out whether it matters to pass load_robot_description to the moveit launch file.

## Steps to Reproduce

Run the robot bringup on real hardware, i.e. with

    roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=MYROBOTIP kinematics_config:=/path/to/mykinematics.yaml

Note the hash of the robot description:

    rosparam get /robot_description | md5sum

Then launch the moveit config:

    roslaunch ur10_moveit_config move_group.launch 

Note the hash of the robot description has changed:

    rosparam get /robot_description | md5sum

If I re-run everything but then do not load the robot description (i.e. the second launch command is now)

    roslaunch ur10_moveit_config move_group.launch load_robot_description:=false

I note the robot description does not change; however I now see the error in my console:

    [ERROR] Semantic description is not specified for the same robot as the URDF

## Expected Behavior

Not sure what to expect – just want to make sure I'm launching in the correct order and not overwriting any important calibration offsets.

## Actual Behavior

Did not observe any ill effects regardless of load_robot_description but curious whether there are any.

fmauch, 2021-09-28:
Thanks for bringing this up, although this isn't the correct repository for this, but it should rather be https://github.com/fmauch/universal_robot or https://github.com/ros_industrial/universal_robot.

I'll look into this...

mzucker, 2021-09-28:
Thanks for looking into it!

The error seems to be happening here: <https://github.com/ros-planning/srdfdom/blob/noetic-devel/src/model.cpp#L614>

Looks like it's triggered because the name of the [SRDF robot](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur10_moveit_config/config/ur10.srdf#L6) (in this case "ur10_robot") does not match the name of the [generic URDF](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_description/urdf/ur.xacro#L2) (just "ur") loaded by the bringup.

If I change the name attribute in the generic XML to "ur10_robot", the error goes away. Obviously that's not a good long-term fix, but it confirms that it's a name conflict.

It seems like one possible fix would be to somehow use xacro to ensure that the top-level `<robot>` tag in the generic URDF has the same name as the SRDF? 

Also, it seems like the only differences between the SRDF files for the different UR robot versions is their names. So possibly just having a single additional "generic" SRDF, or using xacro to eliminate the redundancy there would be better.

Should I file an issue against <https://github.com/fmauch/universal_robot>?