UniversalRobots/Universal_Robots_ROS_Driver #472: Does the ROS driver expose the safety planes?

mzucker, 2021-09-20:
# Summary

Was hoping to get some information about safety setup of the robot including safety planes on the ROS side.

# Versions
 - ROS Driver version: Both ROS Melodic and ROS Noetic, driver commit 54e24c0; universal_robot calibration_devel commit 1ffd6918
 - Affected Robot Software Version(s): URSoftware 3.15.1.106171 (Mar 29 2021) 
 - Affected Robot Hardware Version(s): UR10
 - Robot Serial Number: 2018300027
 - UR+ product(s) installed:
 - URCaps Software version(s): external control

# Impact

It would be great if it was possible to read safety planes defined in the robot installation so they could be used for planning (e.g. in MoveIt, etc).

## Use Case and Setup

The use case would be a ROS topic related to the safety parameters of the robot installation, including a list of up to the maximum number of defined safety planes (8 I believe). Then the safety planes could be defined as primitives in MoveIt and added to the planning scene, perhaps with some additional safety margin to avoid tripping a protective stop on the robot side.

Is this possible? If not, please consider this a feature request!

fmauch, 2021-09-29:
I cannot think of an "easy" implementation of this and we currently do not have the plan or resources for this. So, I'll mark it as "help wanted".

mzucker, 2021-09-29:
I also asked on the UR forums: https://forum.universal-robots.com/t/are-safety-planes-exposed-to-ros/16546

Maybe someone at UR will get excited about making this possible?

urrsk, 2021-10-08:
@mzucker  I like the idea, it could be the next step of making integration easier.
See my reply on the forum https://forum.universal-robots.com/t/are-safety-planes-exposed-to-ros/16546

How do you in vision the use. An export of safety boundaries to an URDF file or?