UniversalRobots/Universal_Robots_ROS_Driver #476: Connecting UR3e and gripper in top level launch file

iosb-ina-mr, 2021-09-27:
# Summary

Hello everybody,
I try to launch all relevant nodes and launch files of the UR3e and the Robotiq Hand-E gripper using a new launch file. What I want to achieve is a unified joint_states topic (/joint_states) containing the joint values of the UR3e along with the values of the gripper and having all those transformations being published to TF by a robot_state_publisher (the attached figure visualizes the launch file chain). And here is the problem, calling ur3e_bringup.launch creates a robot_state_publisher which subscribes to the default joint_states topic and a ur_hardware_interface which publishes to the same topic.
So, remapping the joint_states topic in my launch file changes the topic for both, the robot_state_publisher and the ur_hardware_interface. This, however, leads to the transformations of the gripper not being published to TF, because the robot_state_publisher does not listen to the unified topic anymore due to remapping.

I wonder, if there is a more elegant way to bring all things together other than modifying ur_control.launch (hardcode a remap to the ur_hardware_interface node) or copying the content of ur_common.launch over to my own launch file basically calling ur_control.launch myself with a remapping in place?

[image]

# Versions
 - ROS Driver version: Tag v2.0.0
 
 ## Workaround Suggestion

Add a new launch-file argument to the launch files that allows to alter the joint_states topic the ur_hardware_interface publishes to.

fmauch, 2021-09-29:
This is exactly, why this is splitted up that way. `ur_common.launch` starts everything that is needed for a standalone application with `ur3e.launch` specifying this for the specific robot model. `ur_control.launch` starts the robot only, so that would be the file that you would include in a more complex situation:
```
my_file.launch
  - include ur_control.launch
  - start gripper node
  - start robot_state_publisher
  - load description (custom description containing the robot and the gripper)
```
This way, the ur hardware interface and the gripper will publish their joint states on the same `/joint_states` topic and one global `robot_state_publisher` will transform those into tf messages using your common description.

iosb-ina-mr, 2021-09-30:
@fmauch, thank you for your clarification. That's the solution I already have in place, but with a joint_state_publisher to fuse both joint states coming from the UR and the gripper.

closed 2021-09-30 by iosb-ina-mr