UniversalRobots/Universal_Robots_ROS_Driver #479: I get an error when roslaunching ur_ros_driver

osumilab, 2021-10-05:
# Summary

When I roslaunch ur_ros_driver, I get an error if there is no scaled controller.

# Versions
 - ROS Driver version:ROS noetic
 - Affected Robot Software Version(s):URSoftware 5.9.3.1031212(Sep 22 2020)
 - Affected Robot Hardware Version(s):UR3e
 - Robot Serial Number:20205300834
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Impact
UR3e cannot be controlled using ROS because ur_ros_driver does not start.

# Issue details
When I try to start it using roslaunch, I get the following error
It used to work fine, but suddenly I started getting errors.

`roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=XXX limited:=true kinematics_config:="${HOME}/my_robot_calibration.yaml"`

    [ERROR] [1633410437.724233056]: This controller requires a hardware interface of type 'scaled_controllers::ScaledPositionJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
    [ERROR] [1633410437.724256060]: Initializing controller 'scaled_pos_joint_traj_controller' failed

## Project status at point of discovered
Half a year ago, I got it as follows.
`$ git clone -b calibration_devel https://github.com/fmauch/universal_robot.git`
`$ git clone https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git`

Currently, https://github.com/fmauch/universal_robot does not have ur_e_description, so it cannot be updated.

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

osumilab, 2021-10-05:
`... logging to /home/osumi/.ros/log/2aa9d1b4-25a1-11ec-bada-3b5469e2de83/roslaunch-osumi-System-Product-Name-135363.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://133.91.72.99:37675/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: noetic
 * /rosversion: 1.15.13
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_joint_traj_controller/state_publish_rate: 500
 * /scaled_pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /scaled_pos_joint_traj_controller/type: position_controll...
 * /scaled_vel_joint_traj_controller/action_monitor_rate: 20
 * /scaled_vel_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_vel_joint_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /scaled_vel_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_vel_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_vel_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.1
[... 192 lines cut ...]
[ INFO] [1633413471.119954488]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1633413471.119976168]: Service available.
[ INFO] [1633413471.119984956]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1633413471.120410961]: Service available.
[ INFO] [1633413471.302252472]: Robot mode is now RUNNING
[ INFO] [1633413471.302964174]: Robot's safety mode is now NORMAL
[INFO] [1633413471.341713]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1633413471.342695]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1633413471.343178]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1633413471.344155]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1633413471.344396]: Loading controller: pos_joint_traj_controller
[INFO] [1633413471.345405]: Loading controller: joint_state_controller
[INFO] [1633413471.360780]: Loading controller: joint_group_vel_controller
[INFO] [1633413471.362741]: Loading controller: scaled_pos_joint_traj_controller
[... cut ...] eir own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[INFO] [1633413472.374675]: Loading controller: force_torque_sensor_controller
[INFO] [1633413472.378691]: Controller Spawner: Loaded controllers: joint_state_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1633413472.380697]: Started controllers: joint_state_controller, speed_scaling_state_controller, force_torque_sensor_controller
`

closed 2021-10-05 by osumilab

fmauch, 2021-10-05:
Seems to be a duplicate of #480 which has a solution proposed.