UniversalRobots/Universal_Robots_ROS_Driver #481: Method for Stopping Suddenly (i.e. E-Stop)

rolling2020, 2021-10-12:
I'm curious about if it is possible to have the driver make the robot stop suddenly, similar to how the robot stops when the E-STOP button on the Teach Pendant. We want a way to programmatically stop the robot in its current position, without much concern about acceleration or jerk, etc.

Currently, to make the robot stop, we use the cancelAllGoals() function of the ROS SimpleActionClient (specifically actionlib::SimpleActionClient<control_msgs::FollowJointTrajectoryAction>) when controlling the robot through the scaled_pos_joint_traj_controller, which does stop the robot, but not immediately; the robot will continue moving slightly, which might be outside the trajectory it had been commanded to follow. In movement through open space, this is fine. But when approaching a table, this could potentially result in the robot continuing to move into the table, which might lead to breaking the attached gripper or worse. 

We want a way to duplicate the behavior of pressing the E-Stop, so as to stop the robot suddenly, without having to worry about whether it continues moving or not.

gavanderhoorn, 2021-10-12:
I would like to reinforce @fmauch's comment on "e-stop like behaviour".

What I believe you're looking for @rolling2020, is a controlled stop with a very short trajectory length.

An e-stop removes power from the actuators and engages the breaks. That's not a controlled stop any more.

gavanderhoorn, 2021-10-12:
> I believe that if we simply changed the stop_trajectory_duration to 0 for the controller we use to send trajectories (`scaled_pos_joint_traj_controller`), that would remove the smooth stopping behavior that we have now (that we want to keep for our open-space movements).

Smooth stops when executing trajectories should not be affected by `stop_trajectory_duration`. The `JointTrajectoryController` (or the scaled variant) uses `stop_trajectory_duration` to initialise the `TrajectoryBuilder` (which generates the stopping/holding trajectory). That builder is only used when cancelling goals, or when the JTC receives an empty trajectory as a new goal (which is sort-of the same as cancelling, without cancelling).

Smooth stops for normal trajectories are a function of the trajectories, not of the JTC.