UniversalRobots/Universal_Robots_ROS_Driver #482: UR5 MoveIt! Execute Fails on Real Robot

sumiboss95, 2021-10-12:
Hello, 

I'm running ROS Melodic on Ubuntu 18.04 with a 5.4.143-rt64 kernel. I wanted to use ROS to connect my computer with a real UR5 robot with CB3 controller and 3.13 Polyscope version. The robot has a Robotiq camera, and torque sensor mounted, as well as a custom end effector. I followed [your tutorial](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) to install the driver and to calibrate the robot, which is done correctly, and then [this one](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/usage_example.md) to try to use the driver. I wanted to control the real robot with MoveIt! I did all the steps in the second tutorial, and when I get to the rviz gui, I plan the path but can't execute it. That is, I cannot execute it on the real robot; it won't move. The robot is connected to the UR5 controller and after I start the External Control program on the UR5 i get this info:

`[ INFO] [1634038019.712352877]: Robot connected to reverse interface. Ready to receive control commands.`

This is the output on the terminal from which I launched roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch (I tried planning and executing path a couple of times):

```
[ INFO] [1634038162.327979298]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1634038162.328769093]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1634038162.329233621]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.329262222]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.329293093]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.329324419]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.340358627]: RRTConnect: Created 6 states (4 start + 2 goal)
[ INFO] [1634038162.340834941]: RRTConnect: Created 7 states (3 start + 4 goal)
[ INFO] [1634038162.342155666]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1634038162.342183359]: RRTConnect: Created 12 states (5 start + 7 goal)
[ INFO] [1634038162.342245492]: ParallelPlan::solve(): Solution found by one or more threads in 0.013184 seconds
[ INFO] [1634038162.342397387]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.342427398]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.342468110]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.342509692]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.343200811]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1634038162.343238479]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1634038162.343707175]: RRTConnect: Created 6 states (2 start + 4 goal)
[ INFO] [1634038162.344029835]: RRTConnect: Created 6 states (3 start + 3 goal)
[ INFO] [1634038162.344097804]: ParallelPlan::solve(): Solution found by one or more threads in 0.001736 seconds
[ INFO] [1634038162.344201981]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.344223002]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1634038162.345305828]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1634038162.346399065]: RRTConnect: Created 8 states (3 start + 5 goal)
[ INFO] [1634038162.346571596]: ParallelPlan::solve(): Solution found by one or more threads in 0.002399 seconds
[ INFO] [1634038162.352383640]: SimpleSetup: Path simplification took 0.005699 seconds and changed from 4 to 2 states
[ INFO] [1634038169.102762801]: Execution request received
```

Finally, this is the output from the terminal where I launch RViz:

```
[ INFO] [1634038114.577987585]: rviz version 1.13.19
[ INFO] [1634038114.578014765]: compiled against Qt version 5.9.5
[ INFO] [1634038114.578023306]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1634038114.580495495]: Forcing OpenGl version 0.
[ INFO] [1634038114.656398726]: Stereo is NOT SUPPORTED
[ INFO] [1634038114.656442584]: OpenGL device: Mesa DRI Intel(R) UHD Graphics (CML GT2)
[ INFO] [1634038114.656463324]: OpenGl version: 3,0 (GLSL 1,3).
[ INFO] [1634038117.838167550]: Loading robot model 'ur5_robot'...
[ WARN] [1634038117.851798728]: Kinematics solver doesnt support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1634038117.893602461]: Starting planning scene monitor
[ INFO] [1634038117.894853255]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1634038118.077698016]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1634038119.144668526]: Ready to take commands for planning group manipulator.
[ INFO] [1634038119.144816552]: Looking around: no
[ INFO] [1634038119.144862536]: Replanning: no
```

I don't know what I'm doing wrong. Is the real robot even supposed to move? Or do I have to make a specific setting via MoveIt! Setup Assistant so that the real robot could move? I'd really appreciate any help or ideas you could give me.

Thanks in advance!

urrsk, 2021-10-19:
@sumiboss95  Can you try to uninstall all other URCaps, than the External Control, to see if the e.g. Robotiq URCap is blocking the RTDE protocol?

Though, I think we need some more clues. Can you also post the full driver terminal output?

sumiboss95, 2021-10-19:
@urrsk I can't uninstall other URCaps as we're using the torque sensor info at the moment but will try to do it as soon as possible and see if it will maybe fix the problem. 

Meanwhile, here is the output from the driver terminal (it is the complete output, after also starting the external control program on the robot side):

```
started roslaunch server http://robotika-G3-3500:34277/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 125
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
 * /joint_based_cartesian_traj_controller/type: position_controll...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /pose_based_cartesian_traj_controller/base: base
 * /pose_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /pose_based_cartesian_traj_controller/tip: tool0_controller
 * /pose_based_cartesian_traj_controller/type: pose_controllers/...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
[... 239 lines cut ...]
```
[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[controller_stopper-6] killing on exit
[ros_control_stopped_spawner-5] killing on exit
[ros_control_controller_spawner-4] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
[INFO] [1634564358.003834]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1634564358.003873]: Shutting down spawner. Stopping and unloading controllers...
Interrupt signal (2) received.
[INFO] [1634564358.005313]: Stopping all controllers...
[INFO] [1634564358.005554]: Stopping all controllers...
[ERROR] [1634564358.008365511]: Could not stop controller 'pos_joint_traj_controller' since it is not running
[INFO] [1634564358.008554]: Unloading all loaded controllers...
[INFO] [1634564358.009653]: Trying to unload joint_group_vel_controller
[INFO] [1634564363.037961]: Unloading all loaded controllers...
[WARN] [1634564363.039197]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1634564363.041873]: Trying to unload force_torque_sensor_controller
[WARN] [1634564363.044473]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

urrsk, 2021-10-20:
Can I ask you to verity that the External Control node is working?
You can use this example to verify: https://github.com/urrsk/Universal_Robots_ExternalControl_URCap/tree/master/examples/simple_external_control_server
Please use the same ip and port settings as on the ROS setup.