UniversalRobots/Universal_Robots_ROS_Driver #487: Can't connect to UR5e because of "[ur_hardware_interface-3] has died!"

CodeDok, 2021-11-01:
# Summary

Hello,
I'm unable to connect to the URSim robot. It's the first time I'm working with ROS or with a robotic arm in general. I already looked at other issues (closed) which have similiar messages and tried the solutions but they did unfortunately not work.
Thank you for your help! :)
#327 

# Versions
 - ROS Driver version: melodic (1.14.12)
 - Affected Robot Software Version(s): 5.11.4.108353
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number: 20205501392
 - UR+ product(s) installed: none
 - URCaps Software version(s): none

# Impact
I cant connect to the robotic arm (URSim).

# Issue details
When executing "roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.56.101" I get an error even though I followed the quick start. I build it like the README without problems.

## Use Case and Setup
I'm using URSim for the e-Series and set everything up like in "Using the ur_robot_driver"

## Steps to Reproduce
New OS and fresh VM (URSim)

## Actual Behavior
without debug

```
Summary

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 500
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
 * /joint_based_cartesian_traj_controller/type: position_controll...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /pose_based_cartesian_traj_controller/base: base
[... 628 lines cut ...]
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[ INFO] [1635745116.864789475]: Robot mode is now RUNNING
[ INFO] [1635745116.864850956]: Robot's safety mode is now NORMAL
[INFO] [1635745116.867873]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[New Thread 0x7fffb7542700 (LWP 2476)]
[INFO] [1635745116.983931]: Loading controller: speed_scaling_state_controller
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[New Thread 0x7fffb6d41700 (LWP 2478)]
[INFO] [1635745116.988292]: Loading controller: force_torque_sensor_controller
[tcsetpgrp failed in terminal_inferior: Invalid argument]
[New Thread 0x7fffb6325700 (LWP 2480)]
[INFO] [1635745117.204041]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1635745117.216047]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range

Thread 1 "ur_robot_driver" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) 
```

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

urrsk, 2021-11-03:
Hi @CodeDok 
Sorry to here that you have problems to get started.

>     * URCaps Software version(s): none
I noted that you did not mentioned which URCap you have installed. 
Have you installed the [External Control URCap](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/releases) and add it to a program for inject the ROS control, or are you running the driver in headless mode with the robot in ```Remote Control Mode```

CodeDok, 2021-11-08:
> Hi @CodeDok Sorry to here that you have problems to get started.
>
>
> I noted that you did not mentioned which URCap you have installed. Have you installed the [External Control [...]

Hi @urrsk 

I apologize for the late answer. I was on vacation and didn't look into my mails or anything. 

I didn't install the URCap because my goal is just visualize the robot and not controlling it.

Today (back in the office) I tried to set everything up onto a different device (with a fresh ubuntu) and there it works without any complications. My OS might be the cause but its more practical to use another device in my case than to reinstall the os.

urrsk, 2021-11-10:
@CodeDok Happy to here that you succeed at the end.
Could a docker approach had helped you, as in having this ROS driver running in a docker container or would that also have been inconvenient?

closed 2021-11-10 by urrsk

CodeDok, 2021-11-10:
It think it might have worked but I'm okay with the current setup. Maybe I'll try it in the future :)