UniversalRobots/Universal_Robots_ROS_Driver #488: Can build the driver on noetic

jagomo, 2021-11-12:
# Summary
I'm experiencing an issue similar to the ones mentioned here reported here in this issues:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/255
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/235

I trying build it for noetic on ubuntu 20.04. Looks like some mismatch with ur_msgs is preventing the driver to build, but I can't figure out a solution.

# Versions
 - ROS Driver version:
   - master branch from git@github.com:UniversalRobots/Universal_Robots_ROS_Driver.git
   - calibration_devel branch from  https://github.com/fmauch/universal_robot
 - Affected Robot Software Version(s): N/A
 - Affected Robot Hardware Version(s): N/A
 - Robot Serial Number: N/A
 - UR+ product(s) installed: N/A
 - URCaps Software version(s): N/A

# Impact
Severe

# Issue details
Trying to port current codebase from kinetic to noetic the following build error appears

```
/home/jagomo/src/noetic-port/ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:453:39: error: ‘ur_msgs::SetPayload::Request’ {aka ‘struct ur_msgs::SetPayloadRequest_<std::allocator<void> >’} has no member named ‘mass’
  453 |             << " set_payload(" << req.mass << ", [" << req.center_of_gravity.x << ", " << req.center_of_gravity.y
      |                                       ^~~~
/home/jagomo/src/noetic-port/ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:453:60: error: ‘ur_msgs::SetPayload::Request’ {aka ‘struct ur_msgs::SetPayloadRequest_<std::allocator<void> >’} has no member named ‘center_of_gravity’
  453 |             << " set_payload(" << req.mass << ", [" << req.center_of_gravity.x << ", " << req.center_of_gravity.y

```
I've followed the instructions in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building

But the mentioned issue prevents the build.

Checked just in case that the dependency can be resolved and it is installed. Here ur_msgs is set as a dependency (also in my local checkout)
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/package.xml#L46
```
❯ cat src/Universal_Robots_ROS_Driver/ur_robot_driver/package.xml -n | grep ur_msgs
    46	  <depend>ur_msgs</depend>

❯ rosdep resolve ur_msgs --rosdistro noetic                                
#apt
ros-noetic-ur-msgs

❯ apt list | grep ur-msgs                                                  

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-noetic-ur-msgs/focal,now 1.3.4-1focal.20210528.210935 amd64 [installed]
```

Any suggestions about what could I have overlooked?

urrsk, 2021-11-15:
@jagomo The "payload" field name in the ur_msgs was at some point changed to "mass".
You might use an old version of the ur_msgs?

jagomo, 2021-11-15:
First, thanks for taking the time to reply. I suspect the same but, right now, by following https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#building, looks like ur_msgs is pulled by using rosdep. Which resolves it to be a dependency that can be solved through apt like I showed in the OP.

```
❯ apt list -a ros-noetic-ur-msgs
Listing... Done
ros-noetic-ur-msgs/focal,now 1.3.4-1focal.20210528.210935 amd64 [installed]
```

Just return one available version of the library (maybe in 18.04 are several to choose from). Therefore I don't have many more options here. Either I'm using the wrong branches or ur_msgs should be pulled from somewhere else.

urrsk, 2021-11-16:
@jagomo ros-noetic-ur-msgs 1.3.4 seems to be the latest. My guess it that your compiler are using another the old one.

gavanderhoorn, 2021-11-16:
I can't reproduce this: using `ros:noetic` (Docker image) and following the instructions in the *Building* section of the `README.md` linked by @jagomo I only get some warnings about initialisation order of a few member variables.

But no build failures.

@jagomo: perhaps you could try and find out whether you have a source clone of `ur_msgs` somewhere in your workspace, or installed somewhere in an underlay?

jagomo, 2021-11-17:
@urrsk @gavanderhoorn  You guys were absolutely on point. There was no src clone, but there was a header in /devel (that I erroneously assumed as being part of this last build iteration). This was causing the issue. After some cleanup everything built as expected. Thank you very much for your time and the hint!

closed 2021-11-17 by jagomo