UniversalRobots/Universal_Robots_ROS_Driver #489: Error compiling Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp

fglaurent, 2021-11-25:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

# Versions
 - ROS Driver version: Noetic
 - Affected Robot Software Version(s): 
 - Affected Robot Hardware Version(s): 
 - Robot Serial Number: 
 - UR+ product(s) installed: 
 - URCaps Software version(s): 

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*
I followed the instructions to installed the UR driver on my computer running with Ubuntu 20.04 and Ros Noetic. Getting to the building part, when i do "catkin_make", some errors shows up in the 'hardware_interface.cpp' file;

‘ur_msgs::SetPayload::Request’ {aka ‘struct ur_msgs::SetPayloadRequest_<std::allocator<void> >’} has no member named ‘mass’
‘ur_msgs::SetPayload::Request’ {aka ‘struct ur_msgs::SetPayloadRequest_<std::allocator<void> >’} has no member named 'center_of_gravitiy’

I checked in the SetPayload.h, SetPayloadRequest.h and in the SetPayloadResponse.h file and there is indeed no member called 'mass' and 'center of gravity'. 

What can i do to make it right ?

# Issue details
*This part is important in order to clarify the severity of the issue.*

## Use Case and Setup
*What does the setup look like and what are the objectives?*

## Project status at point of discovered
*When did you first observe the issue?*
 - *Before I started?*
 - *In first couple of tries?*
 - *In normal use?*
 - *etc...*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

urrsk, 2021-11-28:
@fglaurent The ur_msgs changed at some point the topic name on payload to mass. My quess is that you most likely have an old version of ur_msgs library on your computer.
Maybe this issue can help op out: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/488

fglaurent, 2021-11-29:
@urrsk Thanks for your help ! I looked into it and I found a solution

fmauch, 2021-11-29:
@fglaurent could you maybe post what your solution was? That might help future readers of this thread. Also, if it is solved, feel free to close this issue :-)

fglaurent, 2021-11-29:
Yes. All I did was create another catkin workspace to download the ur_driver on

closed 2021-11-29 by fglaurent