UniversalRobots/Universal_Robots_ROS_Driver #491: Preempt_rt patch installation problem

fglaurent, 2021-11-29:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*
While trying to implement the kernel with the PREEMPT_RT patch, coming to the step : 
"$ make -j `getconf _NPROCESSORS_ONLN` deb-pkg" 

The error : 
make[2]: *** [debian/rules:7: build-arch] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:77: deb-pkg] Error 2
make: *** [Makefile:1528: deb-pkg] Error 2

shows up and stop the building process of the kernel.

My currently kernel version is 5.11.0-40-generic. I chose the 5.11 and then the 5.10 and 5.4 kernel for the patch but the same error written few ligns up keeps showing up while building process.  

# Versions
 - ROS Driver version: Noetic 
 - Affected Robot Software Version(s): N/A
 - Affected Robot Hardware Version(s): N/A
 - Robot Serial Number: N/A
 - UR+ product(s) installed: N/A
 - URCaps Software version(s): N/A

# Impact
*What is the impact of the issue. Try to describe it well enough, in order for us to prioritise the issues.*

# Issue details
*This part is important in order to clarify the severity of the issue.*

## Use Case and Setup
*What does the setup look like and what are the objectives?*

## Project status at point of discovered
*When did you first observe the issue?*
 - *Before I started?*
 - *In first couple of tries?* 
 - *In normal use?*
 - *etc...*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

gavanderhoorn, 2021-11-30:
Probably similar to https://github.com/frankaemika/libfranka/issues/62.

penbl4, 2022-02-08:
I was getting the same error. It worked after I followed Franka [installation procedures](https://frankaemika.github.io/docs/installation_linux.html) and commented out these lines in .config
CONFIG_SYSTEM_TRUSTED_KEY
CONFIG_MODULE_SIG_KEY
CONFIG_DEBUG_INFO_BTF

closed 2023-04-02 by github-actions[bot]