UniversalRobots/Universal_Robots_ROS_Driver #493: Can't accept new action goals. Controller is  not running.

Elise-J, 2021-12-07:
# Summary

I'm trying to move a UR5 arm with a python script.
I'm sending a goal using a /control_msg/FollowJointTrajectoryAction action as suggested here https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/122#issuecomment-591861319 
I also use trac_ik to get the corresponding joint post from a giver cartesian pose.

I first run the driver for the robot, then I start the external control on the TP and I get as I should:

```
[ INFO] [1638877506.629783602]: Robot mode is now RUNNING 
[ INFO] [1638877506.631649479]: Robot's safety mode is now NORMAL     
[ INFO] [1638877516.672911345]: Robot requested program          
[ INFO] [1638877516.673097589]: Sent program to robot 
[ INFO] [1638877516.750458563]: Robot connected to reverse interface. Ready 
to receive control commands. 
```
But when I start my python script to send a goal, I get the following error:
```
[ERROR] [1638877525.988082260]: Can't accept new action goals. Controller is
 not running.                                                               

```
The controllers should be running since I have this in the terminal:
```
[INFO] [1638877506.550673]: Started controllers: joint_state_controller, sca
led_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_
sensor_controller                                                           

```

I saw this issue: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/4#issue-506648356  but I correctly started the external control after the driver started.

This is how I send the goal : 
```
 client = actionlib.SimpleActionClient(
        '/pos_joint_traj_controller/follow_joint_trajectory', 
        control_msgs.msg.FollowJointTrajectoryAction)
    client.wait_for_server()
    goal = control_msgs.msg.FollowJointTrajectoryGoal()
    goal.trajectory.joint_names  = ["shoulder_pan_joint", "shoulder_lift_joint", "elbow_joint","wrist_1_joint", "wrist_2_joint", "wrist_3_joint"]
    point = trajectory_msgs.msg.JointTrajectoryPoint()
    point.positions = position
    point.time_from_start = rospy.Duration([6.0])
    goal.trajectory.points.append(point)
    client.send_goal(goal)
    client.wait_for_result()
```

I don't understand what I did wrong. Has someone any ideas?

gavanderhoorn, 2021-12-07:
`scaled_pos_joint_traj_controller` != `pos_joint_traj_controller`.

To make your action client work, it's likely you should update the namespace of the `follow_joint_trajectory` server to `scaled_pos_joint_traj_controller`. So it should read:

```python
 client = actionlib.SimpleActionClient('/scaled_pos_joint_traj_controller/follow_joint_trajectory', ...)
```

Elise-J, 2021-12-07:
Hi @gavanderhoorn 

Thank you, it was indeed the problem ! I changed to `/scaled_pos_joint_traj_controller` and it works :)

For the `Duration` you are right, it takes a value.

closed 2021-12-07 by Elise-J