UniversalRobots/Universal_Robots_ROS_Driver #497: URScript does not execute, except when killing driver

gycn, 2021-12-21:
I am trying to send a URScript to /ur_hardware_interface/script_command topic. However, when I send the script, nothing occurs unless I kill the UR robot driver, at which point the script starts executing. The command I send is `force_mode(tool_pose(), [1, 0, 1, 0, 0, 0], [5.0, 0.0, 5.0, 0.0, 0.0, 0.0], 2, [0.1, 0.005, 0.1, 0.05, 0.05, 0.05])`. I am launching the UR robot driver via `roslaunch ur_robot_driver ur10e_bringup.launch robot_ip:=192.168.1.101 kinematics_config:=$(rospack find mypackage)/data/ur10e_calibration.yaml`

# Versions
 - ROS Driver version: 2.0.0
 - Affected Robot Software Version(s): 5.11.5.1010327
 - Affected Robot Hardware Version(s): UR10E
 - Robot Serial Number:
 - UR+ product(s) installed: Robotiq Hand-e
 - URCaps Software version(s): Robotiq Hand-e

closed 2021-12-21 by gycn

fmauch, 2022-01-13:
@gycn did you figure something out? It would be interesting hearing about your experience using `force_mode` together with this driver.

gycn, 2022-01-13:
I realized that I had to wrap the command in a function and include an infinite loop like
```
def use_force_mode():
    force_mode(...)
    while true:
        sync()
    end
end
```
After I finish using force mode, I'd send another command to end it before asking the driver to resend the program to go back to normal position/velocity control.