UniversalRobots/Universal_Robots_ROS_Driver #500: About zero_ftsensor()  and remote control mode

ksjz, 2021-12-30:
Ubuntu18.04 + melodic 

I found that the value of the six-dimensional force sensor of ur10e will become larger and larger, and the value needs to be constantly zeroing by using zero_ftsensor() when using it.    I'm a fresher in ros , I don't know how to program in remote conteol mode even i see the doc(https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#zero_ftsensor-std_srvstrigger).  Is there an example of zeroingthe sensor in the remote control mode?  thank you very much!

urrsk, 2022-01-07:
@ksjz Thanks for your question
To enter the Remote Control mode of the robot, please see the manual. http://help.universal-robots.com/SW_5_11/UR5e/Content/from-tex/software_manual/g5_sections/hamburger_menu_g5/settings_en.htm?Highlight=Remote%20control

@ksjz Though, It seems that you have found a case that is not yet well supported in the driver.
If the e-seris robot is in 'remote control' mode, you can call the service at you referar to. But it seems we lack in supporting zeroing the FT sensor if you are using the robot with the external control URCap.

A workaround, to be able to zero the sensor in 'Normal mode' using the external control URCap, could be to make an extra UR thread calling the zero_ftsensor() URScript function when e.g. an digital output have changed

ksjz, 2022-01-09:
> @ksjz Thanks for your question To enter the Remote Control mode of the robot, please see the manual. [...]

thank you!

159chenJie159, 2025-07-03:
> [@ksjz](https://github.com/ksjz) Thanks for your question  谢谢你的问题 To enter the Remote Control mode of the robot, please see the manual. http://help.universal-robots.com/SW_5_11/UR5e/Content/from-tex/software_manual/g5_sections/hamburger_menu_g5/settings_en.htm?Highlight=Remote%20control要进入机器人的远程控制模式，请参阅手册。http://help.universal-robots.com/SW_5_11/UR5e/Content/from-tex/software_manual/g5_sections/hamburger_menu_g5/settings_en.htm?Highlight=Remote%20control
> 
> [@ksjz](https://github.com/ksjz) Though, It seems that you have found a case that is not yet well supported in the driver.不过，您似乎发现了一个驱动程序尚未得到充分支持的情况。 If the e-seris robot is in 'remote control' mode, you can call the service at you referar to. But it seems we lack in supporting zeroing the FT sensor if you are using the robot with the external control URCap.如果 e-seris 机器人处于“远程控制”模式，您可以调用您推荐的服务。但是，如果您将机器人与外部控制 URCap 一起使用，我们似乎缺乏对 FT 传感器归零的支持。
> 
> A workaround, to be able to zero the sensor in 'Normal mode' using the external control URCap, could be to make an extra UR thread calling the zero_ftsensor() URScript function when e.g. an digital output have changed为了能够使用外部控制器 URCap 在“正常模式”下将传感器归零，一种解决方法是，当数字输出发生变化时，创建一个额外的 UR 线程调用 zero_ftsensor（） URScript 函数

hello,i have a question about gravity compensation.
based on my observation, the load compensation effect achieved by setting the payload and zero for UR5e and FT300 seems to be bad— I'm not sure if this is reasonable.

With a 0.8 kg payload, changing the robot's posture results in an error of several Newtons.
So I tried reading the ft_raw_wrench for gravity compensation, but the values I got were very large, over 20,000. I'm not sure if this is correct or what the units are.

urfeex, 2025-07-04:
@159chenJie159 The ROS driver doesn't take any action on the gravity compensation, so I cannot say anything about that.

The original question has been resolved by https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/567, so I'm closing this.

closed 2025-07-04 by urfeex