UniversalRobots/Universal_Robots_ROS_Driver #502: Problem about control frequency, how to achieve 125 Hz

1825735116, 2022-01-21:
# Versions
 - ROS Driver version: ubuntu 20
 - Affected Robot Software Version(s): URSoftware 3.8.0.61663
 - Affected Robot Hardware Version(s): UR10
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

By using the newest Universal_Robots_ROS_Driver, how to use the port: 50003 of UR10?
Or how to realize 125 Hz control by "/scaled_pos_joint_traj_controller/command" and port: 50002?

urrsk, 2022-02-01:
@1825735116 Port 50002 is the standard port for Polyscope (the UI) to receive the scriptcode to communicate to the robot.
For using this, you need to install the "External Control" URCap.
It should be explained how to do in  https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_cb3.md

Suggestion for improving the documentation is always welcome
Hope you find it helpful.

1825735116, 2022-02-15:
> @1825735116 Port 50002 is the standard port for Polyscope (the UI) to receive the scriptcode to communicate to the robot. For using this, you need to install the "External Control" URCap. It should [...]

[](https://fanyi.baidu.com/translate?aldtype=16047&query=&keyfrom=baidu&smartresult=dict&lang=auto2zh###)[](https://fanyi.baidu.com/translate?aldtype=16047&query=&keyfrom=baidu&smartresult=dict&lang=auto2zh###)[](javascript:void(0);)By using the 50002 port, does the robot receive and execute the angle command at 125Hz

fmauch, 2022-02-23:
When you use the driver as explained in the README, it will open multiple connections.

Script code will be sent to port 50002 and get executed on the robot using the mentioned URCap. The script will connect to the ROS driver on another custom socket. Motion commands will be sent though this socket at 125Hz.

Ontop of that are the ROS controllers. There are different types of controllers providing different interfaces. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/controllers.md for details.

If you really want to send individual commands using a frequency of 125 Hz you *could* use a position_controllers/JointGroupPositionController or velocity_controllers/JointGroupVelocityController. The latter is also included in our default set of controllers: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/c3378599d5fa73a261328b326392e847f312ab6b/ur_robot_driver/config/ur10_controllers.yaml#L129-L131

Use the mechanisms described in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/controllers.md to activate this controller.

However, be aware that you will be far away from a real-time system there. If you send commands to a topic this includes an unpredictable delay and / or jitter. Hence, using controllers that abstract the real-time interface are more beneficial in most use-cases. This is also the reason why we didn't include the position-based group controller into our default set, as this would be expected to generate jerky motions, as it will most certainly not send position commands in a clean 125Hz rate. For velocity control this won't have these negative effects, but you will experience a certain amount of lag nevertheless.

closed 2023-03-28 by github-actions[bot]