UniversalRobots/Universal_Robots_ROS_Driver #503: *The Universal_Robot_ROS_Driver Builid Error* - Cannot locate rosdep definition for [speed_scaling_state_controller]

Chen-Ronaldo, 2022-01-24:
# Summary

*The Universal_Robot_ROS_Driver Builid Error* -------- Cannot locate rosdep definition for [speed_scaling_state_controller]
 

# Versions
 - ROS Driver version: kinect
 - PC System: ubuntu 16.04
 -  Robot  : UR10_CB3

# Issue details
*This part is important in order to clarify the severity of the issue.*
when I run the command `rosdep install --from-paths src --ignore-src -y`  , it reports error as below:
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_robot_driver: Cannot locate rosdep definition for [speed_scaling_state_controller]`. 
I have tried many ways to solve this problem but all failed. So I put this issue to ask for help 

## Steps to Reproduce

I just follow the building steps in The Universal_Robot_ROS_Driver document.

Thank you for helping with my problem!

medashan, 2022-01-25:
I think this should fix it
[https://answers.ros.org/question/382809/problems-installing-ur-driver-on-ros-kinetic/](url)

fmauch, 2022-02-02:
closing then. Please note that kinetic is no longer supported.

closed 2022-02-02 by fmauch