UniversalRobots/Universal_Robots_ROS_Driver #505: error: not valid rs485.urcap when trying to install it to the teach pendant

vickyvou, 2022-01-26:
# Summary

Hello
I am trying to install the rs485-1.0.urcap  to the teach pendant, but I get the following error:
"rs485.urcap" is not a valid URCap file.  The manifest file ("META-INF/MANIFEST.MF") is missing.

I am running 5.11.0.108249 UR sw for the UR5e.

# Versions
 - ROS Driver version: melodic
 - Robot Serial Number: 20215580082

The reason i need to install it is because we have the UR5e and the robotiq 2f85 and I need to control the gripper via ROS and moveit, and it seems this is the right way to go. 

Any help is highly appreciated

gavanderhoorn, 2022-01-26:
How did you download it?

vickyvou, 2022-01-26:
ok that was a great question that already led to the solution. I am not sure because we had it already in the USB stick along with other apps that some, we had already installed at the pendant. Since i also had the repo also installed, i copied the files from the repo and it worked. Thanks a million!

closed 2022-01-26 by vickyvou