UniversalRobots/Universal_Robots_ROS_Driver #506: ROS + moveit for robotiq 2f_85 with UR5e (next steps and errors)

vickyvou, 2022-01-28:
# Summary
Hello all. I need to control both the UR5e and robotiq 2f_85 with ROS and moveit!. I have two basic issues and the most important is: I am not which is the next step so I can have access to the gripper. I can control the UR5e with moveit. Also, a bit of a n00b when it comes to manipulators, so apologies in advance in case I have missed something. NOTE: 2f_85 is connected to the control box via USB and not to the flange of the robot.

So I run the  `ur_robot_driver ur5e_bringup.launch` and things are ok. I test with moveit and it works. Now from what I understand, the next step is to somehow set the communication tools. This is where things get blurry and also I get errors. 

When I run the `rosrun ur_robot_driver tool_communication` command I get the following:
```
[INFO] [1643296476.184035]: Remote device is available at '/tmp/ttyUR'
[INFO] [1643296476.185039]: Starting socat with following command:
socat pty,link=/tmp/ttyUR,raw,ignoreeof,waitslave tcp:10.161.5.10:54321
```

So the thing is: I am a bit lost. I have read most of the other threads but it looks like there is information that I am somehow missing because I just don't seem to be able to move forward.

From my understanding, the next step would be to run `rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /tmp/ttyUR`
 but when I run that I get the following error:

```
 rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /tmp/tty
URTraceback (most recent call last):
  File "/home/control/control_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 90, in <module>
    mainLoop(sys.argv[1])
  File "/home/control/control_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 76, in mainLoop
    status = gripper.getStatus()
  File "/home/control/control_ws/src/robotiq/robotiq_2f_gripper_control/src/robotiq_2f_gripper_control/baseRobotiq2FGripper.py", line 107, in getStatus
    status = self.client.getStatus(6);
  File "/home/control/control_ws/src/robotiq/robotiq_modbus_rtu/src/robotiq_modbus_rtu/comModbusRtu.py", line 95, in getStatus
    output.append((response.getRegister(i) & 0xFF00) >> 8)
AttributeError: 'ModbusIOException' object has no attribute 'getRegister'
```

I understand that this has been an issue to several people. I have tried a couple of solutions and even tried to post this error to the [robotiq](https://github.com/ros-industrial/robotiq/issues/197) issues, but what they kindly told me is that (and this is part of the probable thing that I was missing as info) the communication tools is for when the gripper is connected to the flange of the robot and I should probably post it here. 

So then the main question i have is the following:
If I connect the robotiq 2f_85 gripper to control box via USB, how can I have access to it? Is it still via the communication tools? Because if this is the case there is an error when i run the gripper control.
Or is it better to connect it to the ROS external computer? 

I am a bit lost and apologies if there is any similar thread I might have missed.

Thank you very much for your time and any help is highly appreciated!

*Introduction to the issue*

# Versions
 - ROS Driver version: melodic
 - Affected Robot Software Version(s): 5.11.0
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number: x
 - UR+ product(s) installed: ?
 - URCaps Software version(s): external control (1.0.5), RS485 (1.0.0)

## Use Case and Setup
*What does the setup look like and what are the objectives?*
I am trying to get our system (UR5e+ robotiq 2f_85) to work, especially via ROS and moveit. The gripper is connected to the control box of the UR5e via USB and to the 24v. 
I have uninstalled (given the advice of several posts) the robotiq URCaps and I have installed the following: external control and RS485 on the teach pendant. And the RS-485 daemon runs. 

When I ssh the system and I do:
```
dmesg | grep tty
[    0.000000] Kernel command line: BOOT_IMAGE=(hd0,3)/vmlinuz root=LABEL=rootfs.0 rootfstype=ext3 console=ttyS0,115200 consoleblank=0 net.ifnames=0 biosdevname=0 panic=60 intel_idle.max_cstate=0 processor.max_cstate=0 sdhci_pci.max_freq=0 idle=poll splash quiet rauc.slot=A isolcpus=3
[    0.000000] console [ttyS0] enabled
[    3.553684] 00:02: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    7.531730] systemd[1]: Expecting device dev-ttyS0.device...
[    8.341233] usb 1-2: FTDI USB Serial Device converter now attached to ttyUSB0
```

I run the  external control and by running the `ur_robot_driver ur5e_bringup.launch ` I can actually control the arm with rviz and moveit. The problem is the gripper. 

The ur5e bringup has the following parameters:
```
use_tool_communication = true
tool_voltage = 24
tool_parity = 0
tool_baud_rate = 115200
tool_stop_bits = 1
tool_rx_idle_chars = 1.5
tool_tx_idle_chars = 3.5
tool_device_name = =/tmp/ttyUR
```

Also:
```
ls -l /tmp/ttyUR
lrwxrwxrwx 1 control control 10 Jan 27 19:33 /tmp/ttyUR -> /dev/pts/5
```

I have installed in the same workspace the robotiq packages from the ros robotiq repo.

vickyvou, 2022-02-03:
hello and thank you for your reply. I understand that connecting the gripper to the control box via USB is not as straightforward as connecting it to the tool I/O. The only solution (after having fought with the previous setup) was to change the connection to the external ROS machine via USB instead of the control box, so at least having access to the gripper based on [this](http://wiki.ros.org/robotiq/Tutorials/Control%20of%20a%202-Finger%20Gripper%20using%20the%20Modbus%20RTU%20protocol%20%28ros%20kinetic%20and%20newer%20releases%29) tutorial  worked. Now trying to make them both work via moveit, but that is another story.

Another option is to order  the robotiq Gripper Coupling GRP-ES-CPL-062, but for now I will stick to the USB until I find another solution.

Again, thank you for your time :)  I can only imagine how hard it is, so if you ever have some time and want to have fun with it, please do let me know.

rrapi, 2022-04-06:
Hi everyone!!

I am doing the same thing for my master thesis and I got the same problem as @vickyvou. But unfortunately I cannot connect the gripper directly to the ROS pc. So I am able to control the robot arm but not the robotiq gripper.

@fmauch Do you know if there is a way to make it work directly connected to ur controller??

Could someone help me please to solve this problem??
I am trying a lot of possible solutions this days but no one works.

Thank you very much in advance for your time!!

Franci

firesurfer, 2022-09-19:
Hi everyone. I just ran into the same error with the ROS2 driver and and realized that the `rx_idle_chars`  and `tx_idle_chars` are defined as floating point in the config where as in the hardware interface it is converted from a param to an integer. This issue also exists in the ROS1 driver:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/3853a30b939b0b1554ea17c9948a6b639a8f2bbf/ur_robot_driver/src/hardware_interface.cpp#L244
```c++

    int rx_idle_chars;
    // Number of idle chars for the RX unit used for tool communication. Will be set as soon as the UR-Program on the
    // robot is started. Valid values: min=1.0, max=40.0
    //
    // Note: This parameter is only evaluated, when the parameter "use_tool_communication"
    // is set to TRUE.  Then, this parameter is required.
    if (!robot_hw_nh.getParam("tool_rx_idle_chars", rx_idle_chars))
    {
      ROS_ERROR_STREAM("Required parameter " << robot_hw_nh.resolveName("tool_rx_idle_chars") << " not given.");
      return false;
    }
    tool_comm_setup->setRxIdleChars(rx_idle_chars);
    tool_comm_setup->setParity(static_cast<urcl::Parity>(parity));

    int tx_idle_chars;
    // Number of idle chars for the TX unit used for tool communication. Will be set as soon as the UR-Program on the
    // robot is started. Valid values: min=0.0, max=40.0
    //
    // Note: This parameter is only evaluated, when the parameter "use_tool_communication"
    // is set to TRUE.  Then, this parameter is required.
    if (!robot_hw_nh.getParam("tool_tx_idle_chars", tx_idle_chars))
    {
      ROS_ERROR_STREAM("Required parameter " << robot_hw_nh.resolveName("tool_tx_idle_chars") << " not given.");
      return false;
    }
    tool_comm_setup->setTxIdleChars(tx_idle_chars);
```

As I am in homeoffice today I cannot test if fixing this issue helps with the not working Modbus communication but it is surely related!

fmauch, 2022-09-21:
@firesurfer thanks for pointing that out, I'll look into that!

firesurfer, 2022-09-23:
@fmauch  I tested it on my setup this week and unfortunately the Modbus communication still doesnt work.

The solution I am using now is the official Robotiq urcap which is reachable via tcp port 65332.

A great example of how to control the gripper can be found here:
https://gitlab.com/sdurobotics/ur_rtde/-/blob/master/doc/_static/robotiq_gripper.py
https://gitlab.com/sdurobotics/ur_rtde/-/blob/master/doc/_static/robotiq_gripper_control.py

fmauch, 2022-09-23:
Regarding this thread:

Using the Robotiq_Gripper URCap and connecting to Port 63352 (at least that was the case for me, maybe 65332 was a type @firesurfer?) currently seems the best option to use a Robotiq-2f  gripper.

However, is there a proper ROS abstraction using this method?

Nevertheless this currently leaves me with a non-working tty device forwarding which is rather frustrating as I definitively had this in the past, so I'll further debug this...

firesurfer, 2022-09-23:
@fmauch  you are right about the port!

> However, is there a proper ROS abstraction using this method?

There are some quickly written ros nodes on github for that method. So it depends on what you call a proper ROS abstraction.

rrapi, 2022-09-23:
Hi everyone!!

I had the same problem at the beginning, but at the end I was able to control the gripper directly attached to the robot from the ROS PC.

This is what I did to achieve this:

1. In case there is the Robotiq Gripper URCap, UNINSTALL it.
2. Robot **bringup**: `roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=<ROBOT_IP> kinematics_config:=${HOME}/path_to/ur5e_calibration.yaml use_tool_communication:=true tool_device_name:=/tmp/ttyUR`
3. Start **RTU** gripper connection: `rosrun robotiq_2f_gripper_control RobotiFGripperRtuNode.py /tmp/ttyUR`. REPEAT this until the blue light on gripper is ON.
4. Start gripper CONTROLLER: `rosrun robotiq_2f_gripper_control Robotiq2FGripperSimpleController.py`

Now you should be able to control the gripper.

I used the most updated libraries and drivers. [universal_robot](https://github.com/fmauch/universal_robot) package "calibration-devel" of @fmauch and [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) main branch.

fmauch, 2022-09-23:
@rrapi thanks for sharing this with us! This seems like a very inconvenient workaround, but as I stated earlier: I experienced occasional multiple starts in the past as well, but in my recent tests this went way out of something that should be necessary. Also, this never occurs to me when I use an RS485 adapter on the ROS machine directly.

kirstyellis, 2022-10-07:
Hi all, I am experiencing similar issues with trying to run the Robotiq hand-e gripper on a UR5e robot. I have been able to get things working with a UR5 and robotiq 85 gripper but so far no joy with the UR5e and hand-e. The gripper that I have on the UR5e is connected directly to the wrist, not the control box, does this make a difference as to how to communicate with the robot? I have tried with the robotiq gripper urcap and with the rs-485 urcap. Does anyone have experience of running the gripper when connected directly to the wrist and should that even make a difference? @rrapi is your gripper connected to the ur5e wrist?

rrapi, 2022-10-07:
Hi @kirstyellis, yes It is connected to the UR5e wrist. I think the problem is because you have installed also the RobotiqGripper URCap. Uninstall it and try again.

kirstyellis, 2022-10-07:
Hi @rrapi, thank you for your response! So no urcaps should be installed except the external control urcap? And then you launch the ur5e with the use_tool_communication set to true, run the external control urcap in a program on the pendant and then you are able to connect to the gripper?

rrapi, 2022-10-07:
You must install only external-control and rs485 urcap. Be sure that you installed the right drivers and packages as explained [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).
Yes that's right. You run the robot bringup, start the external-control from pendant, and once the connection is established you can run the gripper control.

tattrongvu, 2022-10-26:
@vickyvou Did you get it to work? After reading multiple source finally i managed to control the gripper. Special thanks to @firesurfer. 
So recap, the method of @rrapi  is only work if the Gripper connect to the RS485 on the flange of UR5e, which is not the case that @vickyvou mention since he connect the gripper to the control box via USB. In this case it just forward the rs485 tool communication to the port 54321 but NOT the USB port which is connected in the control box for the gripper. 
I have a same setup on my lab, which contain a Robotiq Wrist Camera (behave as a coupler) and a 2F85 gripper. So this USB cable come out of the Wrist Camera connect both the camera and the gripper to the control box. 
So for any one have the same setup as me: 
1, install Robotiq Gripper URcap, Wrist Camera URcap and a external_control urcap. (no, do not install the rs485 ur cap)
2, use the Robotiq Gripper class of @firesurfer
[https://gitlab.com/sdurobotics/ur_rtde/-/blob/master/doc/_static/robotiq_gripper.py](url)
 How to use that class is mention here: https://dof.robotiq.com/discussion/comment/5858/#Comment_5858

closed 2023-03-28 by github-actions[bot]

RyanPaulMcKenna, 2024-01-25:
Here is an error message that got logged from attempting to communicate with rs485 serial device /tmp/ttyUR

It is showing messages about an xmlrpc server running on various different ports. My gripper is the on robot rg2 and I notice when the gripper runs its own functions on the teach pendant it showws connections to a xmlrpc server on localhost there as well.

Which means the ur_robot calls functions on another service (also in the robot I guess or inside the gripper although that seems unlikely) through remote procedure call urscript function: rpc_factory.

Maybe this is only happening because I have both urcaps loaded; rs485 and onrobot urcap.

robotlab@ur5enuc:/opt/ros/noetic/share/ur_robot_driver$ more /home/robotlab/.ros/log/dffbc184-bb87-11ee-a80f-e9e4bc85766f/ur_tool_communication_bridge-4*.log
[rospy.client][INFO] 2024-01-25 13:44:35,778: init_node, name[/ur_tool_communication_bridge], pid[2858478]
[xmlrpc][INFO] 2024-01-25 13:44:35,778: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2024-01-25 13:44:35,779: Started XML-RPC server [http://ur5enuc:35985/]
[rospy.impl.masterslave][INFO] 2024-01-25 13:44:35,779: _ready: http://ur5enuc:35985/
[rospy.init][INFO] 2024-01-25 13:44:35,779: ROS Slave URI: [http://ur5enuc:35985/]
[rospy.registration][INFO] 2024-01-25 13:44:35,779: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2024-01-25 13:44:35,779: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2024-01-25 13:44:35,880: registered with master
[rospy.rosout][INFO] 2024-01-25 13:44:35,880: initializing /rosout core topic
[rospy.rosout][INFO] 2024-01-25 13:44:35,892: connected to core topic /rosout
[rospy.simtime][INFO] 2024-01-25 13:44:35,894: /use_sim_time is not set, will not subscribe to simulated time [/cl
ock] topic
[rosout][INFO] 2024-01-25 13:44:35,914: Remote device is available at '/tmp/ttyUR'
[rosout][INFO] 2024-01-25 13:44:35,922: Starting socat with following command:
socat pty,link=/tmp/ttyUR,raw,ignoreeof,waitslave tcp:192.168.0.99:54321
[rospy.internal][INFO] 2024-01-25 13:44:36,099: topic[/rosout] adding connection to [/rosout], count 0
[rospy.core][INFO] 2024-01-25 13:46:56,474: signal_shutdown [atexit]
[rospy.internal][INFO] 2024-01-25 13:46:56,480: topic[/rosout] removing connection to /rosout
[rospy.impl.masterslave][INFO] 2024-01-25 13:46:56,481: atexit

tomqext, 2026-09-01:
> Hi everyone!!
> 
> I had the same problem at the beginning, but at the end I was able to control the gripper directly attached to the robot from the ROS PC.
> 
> This is what I did to achieve this:
> 
> 1. In case there is the RobotiqGripper URCap, UNINSTALL it (must be installed only **external-control** and **rs485** urcap).
> 2. Robot **bringup**: `roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=<ROBOT_IP> kinematics_config:=${HOME}/path_to/ur5e_calibration.yaml use_tool_communication:=true tool_device_name:=/tmp/ttyUR`
> 3. Start the external-control from pendant
> 4. Once the connection is established, start **RTU** gripper connection: `rosrun robotiq_2f_gripper_control RobotiFGripperRtuNode.py /tmp/ttyUR`. REPEAT this until the blue light on gripper is ON.
> 5. Start gripper CONTROLLER: `rosrun robotiq_2f_gripper_control Robotiq2FGripperSimpleController.py`
> 
> Now you should be able to control the gripper.
> 
> I used the most updated libraries and drivers. [universal_robot](https://github.com/fmauch/universal_robot) package "calibration-devel" of [@fmauch](https://github.com/fmauch) and [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) main branch.

This was a big help, thanks 

Regarding `REPEAT this until the blue light on gripper is ON` - this never worked for us. Instead, we found that increasing the timeout here: https://github.com/TAMS-Group/robotiq/blob/b38125c4e25a15e9bdf0f6c63237e56b72e07a6c/robotiq_modbus_rtu/src/robotiq_modbus_rtu/comModbusRtu.py#L65
to `1.0`, let us follow your instructions without any issue

Thanks for the help!