UniversalRobots/Universal_Robots_ROS_Driver #507: Connection to reverse interface dropped - UR10e

DanielBilbao12, 2022-01-31:
# Summary

I have made a ROS program that allows me to send the position of an ArUco marker to the robot, so that it can be placed just above it at a defined distance (in Z). To do this, two nodes are used: One node for the vision system and the transmission of the ArUco’s pose and another node to receive this information and execute the robot’s movement. Everything works correctly, I mean, the robot positions itself satisfactorily until suddenly (I don’t know the cause) the robot stops accepting new poses and the connection is closed. If I run again the external control from the teachpedant, the program still works.

# Versions
 - ROS Driver version: Melodic with Ubuntu 18.04
 - Affected Robot Software Version(s): URSoftware 5.9.1.1031110
 - Affected Robot Hardware Version(s): UR10e
 - URCaps Software version(s): External Control 1.0.5

# Impact
Due to this problem, it is not possible to smoothly use external control for guidance tasks based on visual markers.

# Issue details
I have made a ROS program that allows me to send the position of an ArUco marker to the robot, so that it can be placed just above it at a defined distance (in Z). To do this, two nodes are used: One node for the vision system and the transmission of the ArUco’s pose and another node to receive this information and execute the robot’s movement. Everything works correctly, I mean, the robot positions itself satisfactorily until suddenly (I don’t know the cause) the robot stops accepting new poses and the connection is closed. If I run again the external control from the teachpedant, the program still works.

The controller used is the “fordward_cartesian_traj_controller”. The driver has been downloaded from : https://github.com/UniversalRobots/Universal_Robots_ROS_Driver

In addition, the message that appears on the console is attached in the following image:
[image]

# Extra details

In case you require further information, please do not hesitate to ask for it.

Thanks in advanced!

fmauch, 2022-02-02:
First of all, thank you for posting this.

Basically, what I think is happening here:
 - The control node running on the ROS machine (or the communication between the two machines) is not fast / reliable enough. This is why we suggest using a ROS machine with a real-time patched kernel and a dedicated network connection. I can't see whether you've got that kind of setup running.
 - When the robot does not receive a command from the ROS machine every control cycle, it will end the program (hence, your robot stops)
 - From the screenshot you posted (preferably post output using [syntax hoghlighting](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) instead of screenshots) it looks like your Teach-Pendant program tries to reconnect automatically, but that happens too fast for the current implementation. (That's a bug known to me, but I haven't found the time to address it, yet). A workaround for that would be to add a sleep command at the end of your TP-Program (after the external_control node). Then, your robot will automatically reconnect to your ROS machine. This will however still stop your robot, unfortunately.
 - As a workaround (I do see the problem here and this should be fixed properly), you could try modifying the driver by setting a higher keepalive counter. This can be done by adding `ur_driver_->setKeepaliveCount(5);` [after this line](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/hardware_interface.cpp#L311). You can change the number 5 to anything you like. The higher the number, the more control cycles without any feedback from the ROS machine the robot will allow. Note, that this is an assumption that this workaround **should** work, so please report back in any case.

I hope, this explanation helps and we can address this properly, soon.

aatb-ch, 2022-02-02:
As mentioned on the forum, I also hit this issue when running the driver in a VM. On my side 20.04 and Noetic over a gigabit wired switch, no chance to get a stable connection in a VM (Parallels on Mac), but runs for days with zero hiccups on a real linux machine. I have not inquired further and suspected that it might be something to do with the VM not being responsive enough compared to native hardware, but this prevents me from developing with my laptop while travelling which is quite inconvenient, since I would ideally be able to run ROS in a VM alongside URSim, which is at the moment not possible (at least on my config).

I am not sure what is the error about the "maximum number of clients (1) already connected", could this point rather more to an issue with the network driver of the VM environment?

I have no time to dig further until the end of next week but will happily run more tests after to see if I can isolate the issue to something more specific.

aatb-ch, 2022-02-02:
Maybe a note, our robots are UR10 with CB3.0 running 3.15

DanielBilbao12, 2022-02-03:
Hello again!

First of all, thank you for your quick response!

- Regarding the first point, I am using a virtual machine with ROS Melodic and Ubuntu 18.04. The connection between the robot controller and the PC is done via wired connection and the real time kernel, it is not enabled.

- Regarding the solution proposed in the second point, by introducing a wait sentence of 0.5 s after the external_control node, the connection is re-established without the need to re-execute the program from the TP. The problem, as you say, is that in this way the robot stops (although it reconnects quickly).

- Finally, we have tried to solve the problem by modifying the driver by introducing the sentence ```ur_driver_->setKeepaliveCount(5);``` in the indicated place. We have tried changing the number 5 to higher numbers but the problem persists.

I think the problem is related to the number of poses being sent from the vision system node to the control system node because when the frequency of frames generated by the vision system increases, the connection drops earlier. Could it be because of the topic queue? I have set this to be 1 so that the pose sent by this node is the last one detected by the vision system and so the robot only moves to the final position and not to all the positions that the aruco passes through before being moved through the workspace.

Thanks in advanced,
Daniel.

fmauch, 2022-02-03:
I would expect the problem to correlate with the system load actually going on on the system. I'll be in the lab tomorrow where I will have a look at a quick fix I did last year to a similar problem. I thought, I did the `setKeepaliveCount` thing, but I might remember it wrong.

aatb-ch, 2022-02-06:
I have been doing more testing, we are currently deploying 2 UR10s, which we ideally want to drive off a single Jetson Nano for ease of deloyment, instead of having multiple machines.

I currently am able to run 2 nodes of the ur_robot_driver from the Jetson, and it seems to work well, with both processes hovering around 15% cpu load, but I can still see the occasional dropped connection, about once per hour:

```
[ INFO] [1644150281.232665034]: Robot requested program
[ INFO] [1644150281.232870502]: Sent program to robot
[ INFO] [1644150281.404626379]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1644150429.325002328]: Connection to reverse interface dropped.
[ERROR] [1644150429.325689976]: Sending data through socket failed.
[ INFO] [1644150430.317166557]: Robot connected to reverse interface. Ready to receive control commands.
[ERROR] [1644151231.750051161]: Unexpected error: No trajectory defined at current time. Please contact the package maintainer.
[ INFO] [1644151490.761816208]: Connection to reverse interface dropped.
[ INFO] [1644151491.754148939]: Robot connected to reverse interface. Ready to receive control commands.
```

I have a hunch that this is indeed system load-related, if I am doing other things like installing a package or loading a chrome tab i can see it happening more often. What is the ```setKeepaliveCount``` trick you are mentionning? For my use-case I can allow a pretty loose timeout since what I care about most is stability over time.

aatb-ch, 2022-02-06:
Quick note, I went ahead and compiled a RT kernel for the Jetson Nano and have not seen any dropped connection since then, so it very likely is due to occasional jitter which just gets worse when system load goes up.

I still have these `Unexpected error: No trajectory defined at current time. Please contact the package maintainer.` messages recurring but these are likely an unrelated issue, though I'd prefer to figure out what is causing them (I am publishing trajectory messages to the Topic interface at about 10Hz).

aatb-ch, 2022-03-07:
Just wanted to add some further notes: running 20.04 with RT kernel on a Raspberry Pi 4, but connected to URSim is also very stable with only very occasional dropped connection, at least rare enough to be useable. So the issue is really about running the `ur_robot_driver` node in a VM.

prasuchit, 2022-09-20:
Hi,

I'm facing the exact same problem with UR3e right now. Has anyone found a good fix for this? I just pulled all the latest changes from the [UR ros driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) package and updated to polyscope on the teach pendant to the [most recent version](https://www.universal-robots.com/download/software-e-series/update/latest-polyscope-software-update-sw-5124-e-series/). 

I installed all the dependencies correctly using  rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y -i --verbose, did an update and upgrade after and rebooted both the robot and the PC.

I haven't faced this issue till now and it suddenly started happening today. I'm not sure what changed. I checked and made sure the robot is connected to the network on the teach pendant and my ros driver is connecting to the correct ip address.

Kindly help!

captain-yoshi, 2022-09-30:
It might not be your case, but I had forgotten to update the `Universal_Robots_Client_Library` and had systematic connection drops. Was resolved after the update.

Ramdambo, 2022-10-14:
> * As a workaround (I do see the problem here and this should be fixed properly), you could try modifying the driver by setting a higher keepalive counter. This can be done by adding [...]

Since the file seems to have changed since this comment, could you please update the link or share after which line the `setKeepAliveCount` is supposed to be set ?

Or is there a new workaround for this issue ?

fmauch, 2022-10-14:
Sorry I didn't use a permalink: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/222d67aad0bfb4f35e7acc32d34b4fb74710c0bd/ur_robot_driver/src/hardware_interface.cpp#L315

Ramdambo, 2022-10-18:
Thanks, much appreciated !

fmauch, 2022-11-09:
@alexandrosnic Please see https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/534

jsbyysheng, 2022-11-16:
> Sorry I didn't use a permalink. You can insert that code after this line:
> 
> https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/222d67aad0bfb4f35e7acc32d34b4fb74710c0bd/ur_robot_driver/src/hardware_interface.cpp#L314-L315

It makes UR Simulator (Vmware / Physical Machine) work well ! I suggest adding a parameter like `simulator: =True` or `alive_count: =5` for the ros driver launch file.

fmauch, 2022-11-16:
@jsbyysheng A more descriptive name of what the parameter actually does, such as `max_missed_packages` or something like that would maybe be more suitable. The `keepalive_count` IMHO makes sense internally, but for driver users, this probably isn't self explanatory.

However, when we introduce such a parameter, this would have to be well-documented so users are aware of the implications following from that.

jsbyysheng, 2022-11-16:
> @jsbyysheng A more descriptive name of what the parameter actually does, such as `max_missed_packages` or something like that would maybe be more suitable. The `keepalive_count` IMHO makes sense [...]

In fact, this parameter makes more sense for URSim than real robot. It is not necessary for URSim keeping RT communication. A simple parameter named ‘using_ursim:=true’ with ‘keepalivecount’ fixed to 5 is totally enough. The count number more than 5 may make no sense because the network environment is unstable for any communication.

OltanS, 2023-02-22:
Hi All,

Since this is one of the first results that came up when I personally encountered this issue when I was trying to control a UR5e with ROS for my undergraduate thesis, I thought I would bring up **my solution to the problem**.

The previous answers helped me greatly in understanding the root cause of the issue regarding the network stability.
The software solutions provided are excellent, but I wanted to see if a hardware solution could be reached.

I'll preface this by saying that I am taking no responsibility for the consequences of you following these steps, I am in no way an authority on what I am talking about and I'm purely talking about my personal experience. This is not technical advice.

For reference, I am using **VMware Workstation** 17 with **Ubuntu** 20.04 as my guest host, but I believe the solution steps I took should work in versions of these published in the last 10 or so years. that I was bridging my ethernet connection to the UR5e directly to my virtual machine, but even that did not provide enough network stability to prevent me from losing my connection every ~30 seconds or so.

1. I looked at [this article](https://docs.vmware.com/en/VMware-vSphere/7.0/com.vmware.vsphere.monitoring.doc/GUID-C4EA86FD-95CB-4DE7-A9E3-63F6BFC1A268.html) by vmware talking about some possible causes of network issues. 
2. I realized my virtual network adapters were running e1000 instead of vmxnet3, which is supposed to improve performance a lot. This is not doable by GUI on Workstation, so I had to edit the .vmx file and replace the lines where it says e1000 with vmxnet3. Prudent to make a backup of the vmx before this.
3. I lost my network gui on my guest! Then I got it back by running:
```sudo nmcli networking off```
```sudo nmcli networking on```

Some further reading about adding the network interface back in to the /etc/networks/interface file might be helpful if this does not fix the issue.
4. I increased the resources allocated to the VM from 4GB ram/2 (core * processors) to 6GB ram/4 (core * processors) (not sure if this helped all that much, but my setup can afford it).
5. I disabled side channel mitigation (Which apparently opened my computer up to some potentially very nasty attacks, worth doing research over before doing).

All these steps took me from a place of dropping connection ever 30 seconds to no connections dropped yet. (maybe still possible, I haven't kept it running for over 1 hour at a time but it looks promising).
I suspect the most impactful step was to changing the NIC drivers. I also suspect this combined with the software solution enabling auto-reconnecting in case of a DC as insurance would be sufficient for most use cases.

I hope this is helpful to someone!

JaisonJose241, 2024-01-12:
one of the problem might  be the hardware which is used to connect the system.

zp2546265641, 2024-07-12:
hello, i just meet such problem too.My computer is Ubuntu 20.04, ROS-Noetic, dual system, and 5.15.0-113Generic kernel. After I enabled urcap in the teach pendant, I established a connection with the real robot through the driver. I use the control cabinet io output to control the external air pump and drive ur5 to execute the trajectory planned by moveit at the same time, but the problem occurs. Once the io signal and the planned trajectory are running at the same time, urcap will be interrupted and the sending signal will prompt failure:
[ INFO] [1720767234.769926758]: Completed trajectory execution with status PREEMPTED ...
[ INFO] [1720767234.772389043]: Execution completed: PREEMPTED
[ INFO] [1720767237.051394741]: Robot requested program
[ INFO] [1720767237.051487070]: Sent program to robot
[ INFO] [1720767237.401967035]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1720767257.018619063]: Connection to reverse interface dropped.
But I don't use io signals, and the execution of the track alone can be successful, but sometimes it will be interrupted. I don't know where the problem lies?

zp2546265641, 2024-07-12:
> first thing: if you dont use a RT kernel you'll get constant drops of connection, so you should really fix that first.
> 
> second: i am now on ROS2 but I've seen similar issue, you mention executing a trajectory while toggling an output. I've observed the same: if using the `set_io` service _while_ the robot is executing a trajectory it would drop the connection half the time, forcing to reconnect. I'm thinking that the `set_io` call somehow blocks the controller long-enough to timeout the connection.. didnt have time to dig into it any further, had to find a quick workaround.

hello, do you know which RL-kernel version should i choose for nvidia driver 550.54.14. i am now fix the kernel and the driver, but i meet new problem.
Building module:
cleaning build area...
unset ARCH; [ ! -h /usr/bin/cc ] && export CC=/usr/bin/gcc; env NV_VERBOSE=1 'make' -j16 NV_EXCLUDE_BUILD_MODULES='' KERNEL_UNAME=5.15.76-rt53 IGNORE_XEN_PRESENCE=1 IGNORE_CC_MISMATCH=1 SYSSRC=/lib/modules/5.15.76-rt53/build LD=/usr/bin/ld.bfd modules...(bad exit status: 2)
ERROR (dkms apport): kernel package linux-headers-5.15.76-rt53 is not supported
Error! Bad return status for module build on kernel: 5.15.76-rt53 (x86_64)
Consult /var/lib/dkms/nvidia/550.54.14/build/make.log for more information.
can you give me some advice?

aatb-ch, 2024-07-12:
dont think I can help much.. i've been using a lot the prebuilt raspi ros2 images here https://github.com/ros-realtime/ros-realtime-rpi4-image but that wont be of much help for you i think

zp2546265641, 2024-07-12:
Thank you so much for your help！i will continue to think about it.

zp2546265641, 2024-07-12:
> dont think I can help much.. i've been using a lot the prebuilt raspi ros2 images here https://github.com/ros-realtime/ros-realtime-rpi4-image but that wont be of much help for you i think

Thank you so much for your help！i will continue to think about it.

zp2546265641, 2024-07-13:
> dont think I can help much.. i've been using a lot the prebuilt raspi ros2 images here https://github.com/ros-realtime/ros-realtime-rpi4-image but that wont be of much help for you i think
i have installed 5.15.76-rt53, but it didn't work as expected. It still reported an error, but this time it was a little different:
[ INFO] [1720873969.508932034]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1720873975.622664768]: Received request to compute Cartesian path
[ INFO] [1720873975.624387838]: Attempting to follow 31139 waypoints for link 'tool_ros' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1720873976.552106491]: Computed Cartesian path with 31145 points (followed 100.000000% of requested trajectory)
[ERROR] [1720873977.292264538]: Sending data through socket failed.
[ INFO] [1720873977.292354495]: Connection to reverse interface dropped.
[ INFO] [1720873977.372595596]: Execution request received
[ERROR] [1720873977.400574305]: Can't accept new action goals. Controller is not running.
[ WARN] [1720873977.400910620]: Controller 'scaled_pos_joint_traj_controller' failed with error INVALID_GOAL:
[ WARN] [1720873977.401011718]: Controller handle scaled_pos_joint_traj_controller reports status FAILED
[ INFO] [1720873977.401071288]: Completed trajectory execution with status FAILED ...
[ INFO] [1720873977.402529277]: Execution completed: FAILED

zp2546265641, 2024-07-13:
> i have installed 5.15.76-rt53, but it didn't work as expected. It still reported an error, but this time it was a little different:
> [ INFO] [1720873969.508932034]: Robot connected to reverse [...]

i have installed 5.15.76-rt53, but it didn't work as expected. It still reported an error, but this time it was a little different:
[ INFO] [1720873969.508932034]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1720873975.622664768]: Received request to compute Cartesian path
[ INFO] [1720873975.624387838]: Attempting to follow 31139 waypoints for link 'tool_ros' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1720873976.552106491]: Computed Cartesian path with 31145 points (followed 100.000000% of requested trajectory)
[ERROR] [1720873977.292264538]: Sending data through socket failed.
[ INFO] [1720873977.292354495]: Connection to reverse interface dropped.
[ INFO] [1720873977.372595596]: Execution request received
[ERROR] [1720873977.400574305]: Can't accept new action goals. Controller is not running.
[ WARN] [1720873977.400910620]: Controller 'scaled_pos_joint_traj_controller' failed with error INVALID_GOAL:
[ WARN] [1720873977.401011718]: Controller handle scaled_pos_joint_traj_controller reports status FAILED
[ INFO] [1720873977.401071288]: Completed trajectory execution with status FAILED ...
[ INFO] [1720873977.402529277]: Execution completed: FAILED

zp2546265641, 2024-07-13:
20:40:36.536769 IP 192.168.0.10.60435 > zp-ros.50002: Flags [.], ack 14481, win 455, options [nop,nop,TS val 12400012 ecr 119005074], length 0
20:40:36.536773 IP 192.168.0.10.60435 > zp-ros.50002: Flags [.], ack 23169, win 591, options [nop,nop,TS val 12400012 ecr 119005074], length 0
20:40:36.536774 IP 192.168.0.10.60435 > zp-ros.50002: Flags [.], ack 25312, win 624, options [nop,nop,TS val 12400012 ecr 119005075], length 0
20:40:36.623828 IP 192.168.0.10.60435 > zp-ros.50002: Flags [F.], seq 17, ack 25312, win 624, options [nop,nop,TS val 12400099 ecr 119005075], length 0
20:40:36.623910 IP zp-ros.50002 > 192.168.0.10.60435: Flags [F.], seq 25312, ack 18, win 509, options [nop,nop,TS val 119005162 ecr 12400099], length 0
20:40:36.624020 IP 192.168.0.10.60435 > zp-ros.50002: Flags [.], ack 25313, win 624, options [nop,nop,TS val 12400099 ecr 119005162], length 0
When I listen to port 5002, I start urcap before 20:41, and it does not update the message after running the script