UniversalRobots/Universal_Robots_ROS_Driver #508: Teach pendant freezes when running ROS driver control with move it - UR10e

SCerq, 2022-02-03:
# Summary
We developed a ROS program to control the robot using Move it. Everything runned smoothly until recently, when the teach pendant started to loose the responsiviness, few minutes after the program starts.  However, the robot executes the program perfectly (over ROS) , only the tech pendant stops working. The power off button still works, however, the screen is blocked and I cannot select any option.

# Versions
 - ROS Driver version: 1.0.5, Ubuntu 20.4 with ROS Noetic
 - Affected Robot Software Version(s): 5.11
 - Affected Robot Hardware Version(s): UR 10e
 - URCaps Software version(s): External Control 1.0.5
 
# Impact 
Due to this problem I have to force the robot power off everytime I run a ROS program to control the robot

#Extra Details
In case you need extra information, please do not hesitate to ask for it.

Thanks in advance

fmauch, 2022-02-09:
Does your program (on the TP) include anything else than just the external_control? Do you have any other URCaps installed?

fmauch, 2022-02-11:
That should not cause any problems. 

To further debug this, we would have to have a look at the log files and the generated script. See [this comment](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/74#issuecomment-571949829) for details.

SCerq, 2022-02-11:
I think I have attached all the files that you requested. 

[Here it goes](https://we.tl/t-GA2L9gEzIU)

Thank you!

fmauch, 2022-02-14:
In the files provided the robot program didn't show much, as in the example program included no connection to the ROS machine could be established. However, the most important part is that there is no "weird" configuration in the program's preamble. So, if you did not mess with the script code provided by the ROS driver, there's no indication that there's something wrong with the script code.

In the log file I did not find anything disturbing. The "strangest" thing I found were occasional disconnections such as

```
2021-12-11 16:58:40.748 :: -5 :: C0A0:7 :: null :: 1 ::  :: Program ergoaware starting... (Last saved: 2021-10-11 17:37:22) :: null
2021-12-11 16:58:40.841 :: -3 :: C0A0:7 :: null :: 1 :: ergoaware :: Program ergoaware started :: null
2021-12-11 16:58:40.841 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: steptime=0.002 ::  :: null
2021-12-11 16:58:40.842 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: External control active ::  :: null
2021-12-11 16:58:40.842 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: Starting servo thread ::  :: null
2021-12-11 16:59:24.331 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: Stopping communication and control ::  :: null
2021-12-11 16:59:24.332 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: servo thread ended ::  :: null
2021-12-11 16:59:24.332 :: -3 :: C0A0:0 :: null :: 1 :: socket_read_binary_integer: timeout ::  :: null
2021-12-11 16:59:24.333 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: All threads ended ::  :: null
2021-12-11 16:59:24.333 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: steptime=0.002 ::  :: null
2021-12-11 16:59:24.333 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: External control active ::  :: null
2021-12-11 16:59:24.333 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: Starting servo thread ::  :: null
2021-12-11 16:59:24.334 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: servo thread ended ::  :: null
2021-12-11 16:59:24.433 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: Starting servo thread ::  :: null
2021-12-11 16:59:32.567 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: Stopping communication and control ::  :: null
2021-12-11 16:59:32.568 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: servo thread ended ::  :: null
2021-12-11 16:59:32.568 :: -3 :: C0A0:0 :: null :: 1 :: socket_read_binary_integer: timeout ::  :: null
2021-12-11 16:59:32.568 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: All threads ended ::  :: null
2021-12-11 16:59:32.568 :: -3 :: C0A0:0 :: null :: 1 :: ExternalControl: steptime=0.002 ::  :: null
2021-12-11 18:02:10.557 :: -3 :: C0A0:7 :: null :: 1 :: ergoaware :: Program ergoaware stopped :: null
```

which indicates that connection to the ROS machine was lost (either due to a resource issue (Real-Time kernel? dedicated network connection?)) But that should not lead to this behavior.

Still, could you please try to deactivate looping your program (See [this UR-forum entry](https://forum.universal-robots.com/t/program-loops-forever-default-setting/4418)) and see whether with this the same thing occurs?

@urrsk any ideas where we could dig for more information?

urrsk, 2022-02-15:
@SCerq did this problem start before or after you upgrade the robot software from 5.9.5 to 5.11.8?

fmauch, 2022-02-23:
@SCerq could you pinpoint a time in the log_history when the freezing started?

SCerq, 2022-02-24:
@fmauch, i do not know exactly the instant that the program froze. What I have been doing is to force the robot's shutdown when I try to select an option on the TP screen and the TP does not responds.

In this last test the program froze almost immediatly

******** Log start (2022-02-22 17:43:36) ********

3.5 :: 0009d23h19m37.443s :: 2022-02-22 17:43:46.856 :: -5 :: C0A0:0 :: null :: 1 :: URSoftware 5.11.8.1010444 (Jan 13 2022) s/n: 20215001013 : UR10 ::  :: null
3.5 :: 0009d23h19m37.443s :: 2022-02-22 17:43:48.018 :: -5 :: C0A0:7 :: null :: 1 ::  :: Connected to Controller :: null
3.5 :: 0009d23h19m37.443s :: 2022-02-22 17:43:48.020 :: -2 :: C0A0:3 :: null :: 1 :: URControl 5.11.8.0 ::  :: null
3.5 :: 0009d23h19m37.443s :: 2022-02-22 17:43:48.022 :: -2 :: C0A0:12 :: null :: 1 :: URSafetyA 0: URSafetyB 0 ::  :: null
3.5 :: 0009d23h19m37.443s :: 2022-02-22 17:43:48.234 :: -5 :: C0A0:7 :: null :: 1 ::  :: Safety checksum changed to: A70A :: null
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.924 :: 30 :: C303A2:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.926 :: 118 :: C303A2:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.927 :: 30 :: C303A7:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.928 :: 20 :: C303A2:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.929 :: 20 :: C303A7:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.930 :: 108 :: C303A2:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.931 :: 108 :: C303A7:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.932 :: 30 :: C257A0:6 :: null :: 130 ::  ::  :: 24010E1
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.933 :: -2 :: C100A2:6 :: null :: 1 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.934 :: 108 :: C313A3:6 :: null :: 129 ::  ::  :: 9
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.935 :: 108 :: C313A4:6 :: null :: 129 ::  ::  :: 13
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.937 :: 108 :: C313A5:6 :: null :: 129 ::  ::  :: 52
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.938 :: 108 :: C313A1:6 :: null :: 129 ::  ::  :: 2C6C7AA0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.939 :: 108 :: C313A2:6 :: null :: 129 ::  ::  :: C9086AAD
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.939 :: 108 :: C303A1:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.941 :: 118 :: C313A3:6 :: null :: 129 ::  ::  :: 6
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.942 :: 118 :: C313A4:6 :: null :: 129 ::  ::  :: 10
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.943 :: 118 :: C313A5:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.943 :: 118 :: C313A1:6 :: null :: 129 ::  ::  :: 1D26B60
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.944 :: 118 :: C313A2:6 :: null :: 129 ::  ::  :: 34212C36
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.945 :: 118 :: C303A1:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.946 :: 20 :: C313A3:6 :: null :: 129 ::  ::  :: 25
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.947 :: 20 :: C313A4:6 :: null :: 129 ::  ::  :: 5
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.948 :: 20 :: C313A5:6 :: null :: 129 ::  ::  :: 1
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.949 :: 20 :: C313A1:6 :: null :: 129 ::  ::  :: 3EC2F25A
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.950 :: 20 :: C313A2:6 :: null :: 129 ::  ::  :: 12EB27B1
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.951 :: 20 :: C303A1:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.951 :: 30 :: C313A3:6 :: null :: 129 ::  ::  :: 24
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.952 :: 30 :: C313A4:6 :: null :: 129 ::  ::  :: 1
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.953 :: 30 :: C313A5:6 :: null :: 129 ::  ::  :: 2
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.954 :: 30 :: C313A1:6 :: null :: 129 ::  ::  :: E4D8A5CC
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.955 :: 30 :: C313A2:6 :: null :: 129 ::  ::  :: 420BD380
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.956 :: 30 :: C303A1:6 :: null :: 129 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.957 :: -2 :: C100A0:6 :: null :: 1 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.958 :: -2 :: C100A1:6 :: null :: 1 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.959 :: 118 :: C268A0:6 :: null :: 130 ::  ::  :: 20
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.960 :: -2 :: C100A3:6 :: null :: 1 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.961 :: -3 :: C211A2:6 :: null :: 1 ::  ::  :: 0
3.5 :: 0009d23h19m45.228s :: 2022-02-22 17:43:48.962 :: -2 :: C0A0:5 :: 1 :: 1 ::  ::  :: 0
3.5 :: 0009d23h19m45.229s :: 2022-02-22 17:43:48.965 :: -3 :: C0A0:7 :: null :: 1 :: set_runstate_configurable_digit :: !SECONDARY_PROGRAM_EXCEPTION_set_runstate_configurable_digit! :: null
3.5 :: 0009d23h19m45.230s :: 2022-02-22 17:43:48.965 :: -3 :: C0A0:7 :: null :: 1 :: set_runstate_configurable_digit :: !SECONDARY_PROGRAM_EXCEPTION_set_runstate_configurable_digit! :: null

urrsk, 2022-02-24:
@SCerq Is the problem only when you are using the ROS driver or is it also when you use the robot with a normal UR teach pendant program?

SCerq, 2022-02-24:
@urrsk, yes. Till now, in the tests that I did, the TP screen did not froze when using the normal UR teach. 
Only when I use ROS to control the Robot.

urrsk, 2022-02-28:
@SCerq I think it makes sens to contact your dealer or try to reset the robot SD card.
You can finde a description of how to reflash the SD card on: https://www.universal-robots.com/media/1804841/install-new-image-all-types.pdf

Back in my mind this behavior could be due to space issue on flash drive, but I am not sure if that still could be the issue.

closed 2023-03-28 by github-actions[bot]