UniversalRobots/Universal_Robots_ROS_Driver #510: Connecting UR5e with ROS

Thomas-Little, 2022-02-25:
Apologies if there are similar issues, I've read some of them and they are a bit different.  

I've can't get the Universal robots ROS Driver to connect with the Teach Pendant on the UR5e. 
I've been following the instructions in the repository (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver), but when it comes to launching the ROS Driver and the External Control on the Teach Pendant, I get some errors. Been working on this for a few weeks, appreciate any advice or points in the right direction. 

 - ROS Driver version: 1.15.14, ROS Noetic
- Ubuntu Virtual Box 20.04

My aim is just to use the Driver for basic movements of the Arm.

When I try and run my ROS Driver with:
_

> roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.1.192 \
>   kinematics_config:=$(rospack find ur_calibration)/etc/ur5e_calibration.yaml

_

Then I get the following errors:
[ERROR] [1645177956.726449460]: Pipeline producer overflowed! <RTDE Data Pipeline>
(This one is repeated many times)

[ERROR] [1645178071.892217848]: This controller requires a hardware interface of type 'scaled_controllers::ScaledPositionJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.

[ERROR] [1645178071.895339068]: Initializing controller 'scaled_pos_traj_controller' failed

[ERROR] [1645178072.896927]: Failed to load scaled_pos_traj_controller

What I've done/tried.

I've installed the External Control URCaps on the Teach Pendent and tried to connect with the PC with IP = 192.168.1.192 and port = 50002.

When I try and run the External Control I get this error on the Teach Pendant:
"The connection to the remote PC at 192.168.1.192:50002 could not be established. Reason: Connection refused (Connection refused)"

I believe there is some issue in the connection between the Teach Pendant and Computer, as "Pipeline producer overflowed!" as something to do with an overflowing queue and that these errors are the same issue.

Let me know if you need anymore information, much appreciated.

fmauch, 2022-02-25:
Hi,

could you please provide some more information: What's the IP of your robot and what is the IP of your ROS machine? From your post it seems you are mixing them up.

t-little, 2022-02-28:
Hi, thanks for getting back to me.

The Teach Pendent is defiantly 192.168.1.192

When I type "ip addr show" on Ubuntu I get "127.0.0.1/8 scope host lo"

fmauch, 2022-02-28:
The question is: Under which IP  can the robot reach your ROS machine?

If you use a NAT interface for your VM, but your robot is physically connected to your host machine, you will most probably have problems connecting them easily. Best, use a "bridge" network interface for your VM. This will generate a separate virtual network interface that will get an own IP address. Depending on your local network setup you will have to specify an IP address manually. In your URCap you will have to setup that IP address of your VM in order to connect to the ROS machine.

As I haven't said it yet: Using a VM to control the robot is not necessarily a good idea. Expect it to lose the connection from time to time, as the control loop requires closely sticking to the desired control frequency. It might be ok for trying things out, but for a productive setup you should use a dedicated (physically real) Ubuntu machine connected to the robot.

t-little, 2022-03-04:
Thank you, I've given that a go.

I've swtiched the Network Adapter 1 to Bridged Adapter. 

Following up on the IP addresses used: the teach pendent is 192.168.1.192, the command prompt/ethernet connection is 192.168.1.129 and Ubuntu is 192.168.1.190.
I am able to ping each address from my Ubuntu terminal. 

Should the Bridged Adapter have a set static address different to these?

The error on the Teach Pendant has changed from "Connection Refused" to "connect time out".

fmauch, 2022-03-09:
Why are there now 3 interfaces involved? I'm afraid, I can't quite follow.

What I have got:
- Your robot has the IP address `192.168.1.192`
- You've got a bridged adapter now
- Your ROS machine (VM) has the IP `192.168.1.190` on the bridged interface
- From the ROS machine you can ping the robot with `ping 192.168.1.192`
- You can ssh to the robot (`ssh root@192.168.1.192`) and ping the ubuntu machine with `ping 192.168.1.190`
- I don't know what the "command prompt/ethernet connection" should be, so I'm ignoring this for now.

With this setup you should
- Setup your robot's installation as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_e_series.md). As "Remove host's IP" you should set `192.168.1.190`
- start the driver using `roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.1.192` (ignoring the kinematics config for not for the sake of simplicity)

With that you can verify with `ss -alpn | grep 5000` that your ubuntu machine opened the correct ports. It should give something like this:
```
ss -alpn | grep 5000 | column -t
tcp  LISTEN  0  1  0.0.0.0:50001  0.0.0.0:*  users:(("ur_robot_driver",pid=1157600,fd=18))
tcp  LISTEN  0  1  0.0.0.0:50002  0.0.0.0:*  users:(("ur_robot_driver",pid=1157600,fd=14))
tcp  LISTEN  0  1  0.0.0.0:50003  0.0.0.0:*  users:(("ur_robot_driver",pid=1157600,fd=21))
```

If everything up to this point is as expected, the next best thing would be to check whether there is a firewall involved.

As you already can ssh to the robot you can do so and **on the robot controller** execute `nc 192.168.1.190`. This should not produce any output but (visually) simply set your cursor at the beginning of a new line. If the command immediately returns and you are back in your prompt, you cannot reach the driver. This could be due to a still wrong IP setup or due to a firewall. Sometimes firewalls also will show as a timeout here.
If the `nc` command did not return immediately, type `request_program` and hit <kbd>Enter</kbd>. You should now a URScript program code on the screen. It should contain the lines

```
socket_open("192.168.1.190", 50001, "reverse_socket")
socket_open("192.168.1.190", 50003, "trajectory_socket")
```

If the IP addresses there are different, then there exist other network routes between your VM and the robot controller. In that case either 
 - make sure to deactivate all other routes (e.g. deactivating a secondary network device with NAT configuration in your VM)
 - add the [`reverse_ip`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/c3378599d5fa73a261328b326392e847f312ab6b/ur_robot_driver/launch/ur5_bringup.launch#L5) argument to your launch command where you set `reverse_ip:=192.168.1.190`

closed 2022-03-16 by t-little

ShahhhVihaan, 2023-05-01:
I am trying to calibrate the robot but I get the error message
```
[ERROR] [1682903577.756730462]: Failed to connect to robot on IP 192.168.0.6. Please check that the robot is booted and reachable on 192.168.0.6. Retrying in 10 seconds
```

So I ping my UR3 arm at 192.168.0.6 and it is successful then I try to ssh into the arm and after I enter the password it says Permission denied, please try again.

I entered the password "easybot" and tried it multiple times but it failed. So I thought maybe the password has been changed and I'm not aware of it. I plugged a keyboard into the robot and pressed ctrl+alt+F1 to open the CUI. On opening the CUI I used "root" for login and when asked for the password I typed "easybot" and it gave me access to the root. So now I'm unsure as to why when I try ```ssh root@192.168.0.6``` and enter the password it denies permission. I'd really appreciate help here.

Once I had access to the root I tried pinging my ROS machine and it failed.

For some reason I am only able to ping the robot from my ROS machine but not able to SSH into it and the connection from the robot to the ROS machine is not established.