UniversalRobots/Universal_Robots_ROS_Driver #512: How to cancel the automatic setting of the force sensor to 0 after the UR5e robot is started？

MZJR, 2022-03-10:
Every time I re-run the external control program on the teach pendant, the wrench data is automatically zeroed(), no matter what pose the end of the arm is in. As we all know, the data after zeroing does not reflect the real force, and I want to calibrate myself through [force_torque_tools](https://github.com/kth-ros-pkg/force_torque_tools)

# Versions
Ubuntu20.04 + Noetic

fmauch, 2022-03-10:
@urrsk is the wrench data zeroed on each program start or would we have to take a look into the generated script code here?

MZJR, 2022-03-11:
BTW,After zeroing, I let the end of the robotic arm rotate at a constant speed along the z-axis (the center of mass of the end fixture is on the z-axis), and obtained the force data of the xy-axis (blue is the x-axis, red is the y-axis). Is this a bug in the program acquisition or a bug in the force sensor itself?
[image]

urrsk, 2022-03-14:
> @urrsk is the wrench data zeroed on each program start or would we have to take a look into the generated script code here?

Yes, it is zeroed every time you start a program.
There exist a raw wrench channel on the RTDE protocol. I will suggest using the raw tool wrench if you want to make your own e.g. gravity compensation.
@MZJR Regarding the sinus error. This is an unwanted physical phenom in the sensor. If it is too high, you should contact your distributor/service for getting it corrected.

MZJR, 2022-03-14:
> Yes, it is zeroed every time you start a program. There exist a raw wrench channel on the RTDE protocol. I will suggest using the raw tool wrench if you want to make your own e.g. gravity [...]

Thanks for your reply.

closed 2022-03-14 by MZJR

159chenJie159, 2025-07-03:
Hello, based on my observation, the load compensation effect achieved by setting the payload and zero for UR5e and FT300 seems to be average — I'm not sure if this is reasonable.

With a 0.8 kg payload, changing the robot's posture results in an error of several Newtons.
So I tried reading the ft_raw_wrench for gravity compensation, but the values I got were very large, over 20,000. I'm not sure if this is correct or what the units are.