UniversalRobots/Universal_Robots_ROS_Driver #517: Using concurrently with ur_rtde

gycn, 2022-04-03:
Is it possible to use this driver at the same time as the [ur_rtde](https://gitlab.com/sdurobotics/ur_rtde) library?

fmauch, 2022-04-04:
Hi, this depends on what you want to do with `ur_rtde`. However, most probably there will be conflicts, as you can only have one instance controlling each writable field in the RTDE protocol...

Why would you like to use `ur_rtde` meaning which parts are you missing in this driver?

gycn, 2022-04-08:
I want to be able to use force control and change the force with low latency. Would it be possible to simply resend the program in headless mode for both ur_rtde and this library?

fmauch, 2022-04-11:
That seems not like something I am able to answer, I'm afraid, as I don't know how `ur_rtde` implents force_mode. However, my initial assumption would be that this does not work atm.

closed 2023-02-28 by github-actions[bot]