UniversalRobots/Universal_Robots_ROS_Driver #519: Robotiq Copilot with External Control (RTDE cannot initialize)

julien-audet, 2022-04-08:
## Summary
*Introduction to the issue*

I've used this driver successfully in the past with ROS Kinetic and another URCap called Robotiq Copilot, but I encountered a problem when switching to ROS Noetic. When starting the External Control program, I have the following error "Robotiq Copilot: RTDE cannot initialize. Make sure that Ethernet/IP adapter and profinet are disabled, and that no other URCaps use the RTDE interface". Both IP adapter and profinet are disabled, so the other culprit is the Robotiq Copilot. 

After removing the Copilot URCap, everything's works as expected.

So my question is, do you have any ideas with it works with ROS Kinetic, but not Noetic? Was there any changes in the External Control URCap version 1.01 to version 1.05 causing this?

## Versions (for Noetic)
 - ROS Driver version: Universal_Robots_ROS_Driver
 - ROS Distribution: Noetic
 - Universal Robots Software: URSoftware 3.11.0.82155
 - URCaps Software version(s): External Control 1.05 and Robotiq Copilot 1.17.2.9189

## Versions (for Kinetic)
Only difference is External Control 1.01 instead of 1.05.

fmauch, 2022-04-11:
There have been changes regarding RTDE resource claiming, but mostly the other way around, that we actually check whether there's another client blocking the interfaces needed. As the error comes from the Copiliot URCap there, did you use the same Copilot version? I cannot recall of any changes regarding actually claiming the resources.

julien-audet, 2022-04-27:
Sorry for my slow response. In both cases, I used the same Copilot version (1.17.2.9189). I'll maybe try newer (and older) versions of Copilot to see if I can make it work. Or revert to an older version of External Control.

closed 2023-02-27 by github-actions[bot]