UniversalRobots/Universal_Robots_ROS_Driver #520: Problem using controllers for robot motion control UR10 CB3

MiroslavKohut, 2022-04-12:
# Summary

Hi, I am trying to use UR robot driver for controlling the robot and I am not able to make the robot move through the ROS driver. I have followed the instalation manual. Communication is established I am able to run URcap via robot dashboard but I could not execute any program.

*Introduction to the issue*

# Versions
 - ROS Driver version:
 - Affected Robot Software Version(s): UR 3.15.3.106223
 - Affected Robot Hardware Version(s): UR10 CB3 series
 - Robot Serial Number: 2021301302
 - UR+ product(s) installed: I followd the manual for installationd and build of Universal robor Ros Driver 2.0.0
 - URCaps Software version(s): I tried 1.0.3, 1.0.4, 1.0.5

# Impact
High impact. I am using the robot for the real project and I am not able to control it correctly 

# Issue details
I tried all the possible options but I am blocked at the beginning I am no able to make the controller run any of it but nothing is working. I am running **ROS Noetic on WSL2** machine. through windows.

When I tried to run the robot afte start I will get following error:

```
[ERROR] [1649768175.860973500]: Robot control is currently inactive. Starting controllers that claim resources is currently not possible. Not starting controller 'scaled_vel_joint_traj_controller'
[ERROR] [1649768175.861054900]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
[ERROR] [1649768208.823321100]: Can't accept new action goals. Controller is not running.
```

The controller status afte loading the robot driver is:

```
controller:
  -
    name: "joint_state_controller"
    state: "running"
    type: "joint_state_controller/JointStateController"
    claimed_resources:
      -
        hardware_interface: "hardware_interface::JointStateInterface"
        resources: []
  -
    name: "pos_joint_traj_controller"
    state: "initialized"
    type: "position_controllers/JointTrajectoryController"
    claimed_resources:
      -
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources:
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  -
    name: "scaled_pos_joint_traj_controller"
    state: "stopped"
    type: "position_controllers/ScaledJointTrajectoryController"
    claimed_resources:
      -
        hardware_interface: "scaled_controllers::ScaledPositionJointInterface"
        resources:
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  -
    name: "joint_group_vel_controller"
    state: "initialized"
    type: "velocity_controllers/JointGroupVelocityController"
    claimed_resources:
      -
        hardware_interface: "hardware_interface::VelocityJointInterface"
        resources:
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  -
    name: "speed_scaling_state_controller"
    state: "running"
    type: "scaled_controllers/SpeedScalingStateController"
    claimed_resources:
      -
        hardware_interface: "scaled_controllers::SpeedScalingInterface"
        resources: []
  -
    name: "force_torque_sensor_controller"
    state: "running"
    type: "force_torque_sensor_controller/ForceTorqueSensorController"
    claimed_resources:
      -
        hardware_interface: "hardware_interface::ForceTorqueSensorInterface"
        resources: []
  -
    name: "scaled_vel_joint_traj_controller"
    state: "initialized"
    type: "velocity_controllers/ScaledJointTrajectoryController"
    claimed_resources:
      -
        hardware_interface: "scaled_controllers::ScaledVelocityJointInterface"
        resources:
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
```

When I tried to start any controller I will get following error:

```
[ERROR] [1649768346.616678500]: Robot control is currently inactive. Starting controllers that claim resources is currently not possible. Not starting controller 'scaled_pos_joint_traj_controller'
[ERROR] [1649768346.616792100]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
[ERROR] [1649768350.798879000]: Can't accept new action goals. Controller is not running.
```

When I tried to move robot through the demo:

I am getting the same error on every controller type:

```
rosrun ur_robot_driver test_move
[INFO] [1649768330.860863]: Available trajectory controllers:
[INFO] [1649768330.862598]: 0 (joint-based): scaled_pos_joint_traj_controller
[INFO] [1649768330.863992]: 1 (joint-based): scaled_vel_joint_traj_controller
[INFO] [1649768330.865150]: 2 (joint-based): pos_joint_traj_controller
[INFO] [1649768330.866331]: 3 (joint-based): vel_joint_traj_controller
[INFO] [1649768330.867795]: 4 (joint-based): forward_joint_traj_controller
[INFO] [1649768330.869443]: 5 (Cartesian): pose_based_cartesian_traj_controller
[INFO] [1649768330.871007]: 6 (Cartesian): joint_based_cartesian_traj_controller
[INFO] [1649768330.872289]: 7 (Cartesian): forward_cartesian_traj_controller
Please choose a controller by entering its number (Enter '0' if you are unsure / don't care): 0
[WARN] [1649768346.623394]: The robot will move to the following waypoints:
[[0, -1.57, -1.57, 0, 0, 0], [0.2, -1.57, -1.57, 0, 0, 0], [-0.5, -1.57, -1.2, 0, 0, 0]]
Please confirm that the robot path is clear of obstacles.
Keep the EM-Stop available at all times. You are executing
the motion at your own risk. Please type 'y' to proceed or 'n' to abort: y
[INFO] [1649768350.786986]: Executing trajectory using the scaled_pos_joint_traj_controller
[INFO] [1649768350.799658]: Trajectory execution finished in state -1
```

Could you please help me what could cause the issue. Thank you for your help.

fmauch, 2022-04-12:
Please note, that running on WSL2 might have some implications on the driver, also see #486.

Your output seems to be missing the "Robot ready to receive control commands", as explained [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#i-started-everything-but-i-cannot-control-the-robot). That seems to be the same as reported in #486. Therefore, I am going to close this and mark as a duplicate. If you feel that this is wrong, please comment and / or reopen.

fmauch, 2022-04-12:
Duplicate of #486

closed 2022-04-12 by fmauch

MiroslavKohut, 2022-04-12:
Hi, 

in my opinion it is not duplicate because I dont have the problem with information during connect as mentioned in 486. Connection is succesful for my setup. The issue in 486 is about port forwarding I could help fixing it. This is different problem. I am already correctly connected and when I run the robot program the terminal output after pressing play button. is
```
[ INFO] [1649769714.128080800]: Robot requested program
[ INFO] [1649769714.128161100]: Sent program to robot
```

See in 486:
I'm currently trying to control a UR10e using WSL2. I'm able to get join trajectories, and rviz displays and updates the robot correctly in real time. But when i try to start the program on the **UR10e it is unable to connect to the host.** I am able to connect and eveything looks fine.

MiroslavKohut, 2022-04-13:
Hi,

I was able to solve the problem and I have written down the solution to https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/486#issuecomment-1096776059

I have just last question if it is safe to use ur_driver_->setKeepaliveCount(5); ?

Thank tou for your time and help

fmauch, 2022-04-13:
Depends on your definition of `safe`. If you miss a command from the external (ROS) side, the robot program will extrapolate using the latest command. So, if you would set this value very high and stop the ROS driver, the robot will continue in a straight line in the last commanded direction. With a value of 5, this will happen for 10ms on an e-series robot and 40ms on a CB3 robot. Depending on the application's speed and error tolerance this may be critical or not. If you think that the deviation from the desired path that your robot can take in 40ms is tolerable, then this should be fairly safe to use IMHO.