UniversalRobots/Universal_Robots_ROS_Driver #524: where did "scaled_pos_joint_traj_controller/command" topic defined?

LEESHOHYUN, 2022-05-05:
when I saw rqt_graph, with roslaunch ur hardware driver ~ and rosrun rqt_controller ~,
I can see /scaled_pos_joint_controller/command topic,
but I cannot find where it defined. and also cannot understand, from {topic_a}/{topic b} architecture, how 'topic b' is added. 
may I ask where to find, and how to add topic after / ??

fmauch, 2022-05-05:
I am not sure whether I understand your question correctly. But if you want to understand where the subscription to the topic comes from, you'll have to dig into ros_control a bit. The scaled JTC is a modified `joint_trajectory_controller`, so you'll find its action server there: 

https://github.com/ros-controls/ros_controllers/blob/noetic-devel/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L296-L299

(or, since you asked for the `command` topic, see [this line](https://github.com/ros-controls/ros_controllers/blob/95ebe05483f4b2381b343bb76cdec6135b044007/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L290).)

Inside this repository the controller is defined in the `controllers.yaml` file:

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/config/ur5e_controllers.yaml#L32-L46

As it is named `scaled_pos_joint_traj_controller`, ros_control generates a node handle using the controller name and adds the `command` topic inside this namespace, hence the `/scaled_pos_joint_controller/command topic`.

I hope, this answers your question, if not please try to specify your question a bit more.

LEESHOHYUN, 2022-05-09:
It's a perfect answer for my question. I appreciate your help so much!