UniversalRobots/Universal_Robots_ROS_Driver #529: test_move script hangs when ask_comfirmation fuction comment out from send_joint_trajectory function

AnthonyLe93, 2022-05-18:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*

# Versions
 - ROS Driver version: noetic
 - Affected Robot Software Version(s): polyscope 3.8
 - Affected Robot Hardware Version(s): UR10
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): 1.0.5

# Impact
cannot integrate the code with other codes

# Issue details

test_move script hangs when ask_comfirmation fuction comment out from send_joint_trajectory function. No error was reported. I suspect this is something to do with wait for sevice timeout but not sure. I am relatively new to ROS so have not familiar with all the debugging procedure in ROS.

## Use Case and Setup
*What does the setup look like and what are the objectives?*

## Project status at point of discovered
*When did you first observe the issue?* when trying to reuse part of the test_move script 
 - *Before I started?*
 - *In first couple of tries?*
 - *In normal use?*
 - *etc...*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

## Workaround Suggestion
*If a workaround has been found, you are welcome to share.*

fmauch, 2022-06-02:
This is really hard to help you there, as you didn't provide us with what you were actually doing. Could you please show us the changes you actually made and verify that the driver as such is actually up and running? (see https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#i-started-everything-but-i-cannot-control-the-robot).

AnthonyLe93, 2022-06-21:
Hi Felix, thanks for getting back to me. Just to clarify, I did have the system working and can control the arm running the original code. The bug happens when i comment out the **ask_comfirmation** method call in the send_joint_trajectory method. No error were reported which leads me to suspect this is something to do with the wait time for sevice timeout but not sure. I am also not controlling the robot on a real time kernal so this could also be the potential cause.

fmauch, 2022-06-28:
Could you try again with the current master? #535 should have addressed this.

AnthonyLe93, 2022-06-28:
Yes works like a charm. Many thanks!!

closed 2022-06-28 by AnthonyLe93