UniversalRobots/Universal_Robots_ROS_Driver #539: Switching between forward_joint_traj_controller and pose_based_cartesian_traj_controller produces unexpected results in URSim

Makuh17, 2022-06-02:
# Summary
I created a small script to have a service that brings my robot to a fixed joint configuration. To do so, I first switch to `forward_joint_traj_controller`, send a trajectory goal with a single point, wait for the goal to be reached and then switch back to the previously used Cartesian controller, `pose_based_cartesian_traj_controller`. I observe that for larger joint space movements, the robot will quickly jump back to the previous pose when triggering the controller switch. See animation below.
[image]

# Versions
 - ROS Driver version: 1.0
 - Affected Robot Software Version(s): URSim 3.15.7
 - Affected Robot Hardware Version(s): Not tested
 - URCaps Software version(s): externalcontrol-1.0.5

# Impact
Using the two above mentioned controllers in sequence is not possible as potentially dangerous robot behavior can be encountered.

# Issue details
The issue seems to primarily happen when commanding larger movements in joint space. Smaller movements seem to not trigger the issue. The issue also does not seem to happen when switching between `forward_joint_traj_controller` -> `forward_cartesian_traj_controller` or  `scaled_pos_joint_traj_controller`->`pose_based_cartesian_traj_controller`. 
I have not tested the issue on real hardware, only in URSim. The workaround is also rather straight forward, so I would give the issue low priority.

## Use Case and Setup
To briefly command the robot to a certain joint pose in between Cartesian trajectories. E.g. to move to a safe position/configuration before next movement.

## Project status at point of discovered
 - *In first couple of tries*

## Steps to Reproduce
1. Launch URSim 3.15.7 (UR10 with externalcontrol)
2. Launch ROS driver (with only the passive controllers started.)
```
roslaunch ur_robot_driver ur10_bringup.launch robot_ip:=192.168.56.101 controllers:="joint_state_controller speed_scaling_state_controller force_torque_sensor_controller"
```
3. Move (e.g. manually) robot to "far away" configuration, e.g. [0, -40, -100, -60, 40, -80]
4. Run the provided node and call the service `/move_away`
[move_away.py.txt](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/files/8824690/move_away.py.txt)

## Expected Behavior
Robot moves to desired joint configuration and stays there.

## Actual Behavior
Robot moves to desired joint configuration and then rapidly moves back.

## Workaround Suggestion
Use either  `forward_joint_traj_controller` with `forward_cartesian_traj_controller` or  `scaled_pos_joint_traj_controller` with `pose_based_cartesian_traj_controller`.

fmauch, 2022-06-28:
Thanks for reporting this. We'll have to dig into that a bit...

urfeex, 2026-03-30:
I'm sorry, that this was never followed up since it was reported. Since ROS 1 is now end of life, this will most likely no longer be addressed.