UniversalRobots/Universal_Robots_ROS_Driver #542: Failed trajectory execution after set IO

shuobh, 2022-06-08:
# Summary
UR Ros Driver fails during execution.

*Introduction to the issue*
When multiple trajectories are passed to the UR Ros Driver, including setio calls through service, the Driver may be interrupted and claims "Sending data through socket failed". This is more likely to happen right after Setio calls.
Sequence to reproduce is: follow_joint_traj 1-> setio ->follow_joint_traj 2. It fails right after setio. The error msgs are

`
Jun 08 12:41:14 nuc-19-robot-14 bluehill-start[2192481]: [ERROR] [1654717274.958094026]: Sending data through socket failed.
`
`
Jun 08 12:41:15 nuc-19-robot-14 bluehill-start[2192481]: [ERROR] [1654717275.476732795]: Can't accept new action goals. Controller is not running.
`

# Versions
 - ROS Driver version: Noetic
 - Affected Robot Software Version(s): 5.11.10
 - Affected Robot Hardware Version(s): UR3e
 - Robot Serial Number: 20225380049
 - URCaps Software version(s):1.0.5

# Impact
Robot trajectory execution get interrupted consistently.

# Issue details

## Project status at point of discovered
*When did you first observe the issue?*
 - *In first couple of tries*

## Steps to Reproduce
Execute trajectories with setio in between.

closed 2022-06-08 by shuobh

gavanderhoorn, 2022-06-09:
What was the resolution?

shuobh, 2022-06-09:
> What was the resolution?

So I realized I was setting payload through UR Script instead of using the defined service call. Changing it to the proper service call solved the issue