UniversalRobots/Universal_Robots_ROS_Driver #547: Can't accept new action goals. Controller is not running

sadofatai, 2022-06-14:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

*Introduction to the issue*
Sometimes after running the robotic arm for a while, the arm controller stops to accept new action goals and then we see this error message on the universal_robots_ros_driver node ("Can't accept new action goals. Controller is not running"). We resolve this problem simply by killing and relaunching the universal_robots_ros_driver node and everything works fine after that. What is really the problem here? Why does it stop accepting new action goals after some time of running the the controller, the ros driver, and the arm? Can someone help solve this problem without having to relaunch the ros driver script?

# Versions
  Ubuntu 18.04, ROS melodic
  
 - Affected Robot Software Version(s):  CB3 and e-Series  
 - Affected Robot Hardware Version(s):  UR5e
 - Robot Serial Number:
 - UR+ product(s) installed: 
 - URCaps Software version(s):  externalcontrol-1.0.5.urcap, rs485-1.0.urcap

fmauch, 2022-06-28:
Do you see any other output in the shell regarding a dropped connection to the reverse interface? Are you running a realtime kernel?

sadofatai, 2022-07-01:
Hi Felix,

Yes it comes sometimes with a dropped connection output which obviously
means that the program was not running, but it also happens when the
connection is active.
To the best of my knowledge, we are not running a realtime kernel. Please,
explain more to me about the realtime kernel and the relationship to our
error message.

Thank you and kind regards

Sado

fmauch, 2022-07-11:
If you aren't using  a real-time kernel and a dedicated network device for communicating with the robot, it can happen, that the connection to the program running on the robot gets stopped.

When this happens, restarting the program on the teach panel should be sufficient. In the shell running the driver, you should see the "Robot ready to receive control commands" again.

sadofatai, 2022-10-11:
Hi Felix,
We are using ROS as our communication interface. Is ROS not sufficient to
keep it stable? What real-time kernel can you recommend?
Our network is a dedicated one but we are using multiple PCs in the
network. We also run a BT action server on one of the connected PCs. Can
this cause a conflict?
Please advise what you think.

Kind regards

Sado

closed 2023-02-24 by github-actions[bot]