UniversalRobots/Universal_Robots_ROS_Driver #565: Cannot locate external_control.script when lauching the driver.

urmahp, 2022-08-09:
When using the all source build and building the client library from source, it is not possible to locate the external_control.urscript from the client library, when launching the driver. 

See the log output below: 

```bash
roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=127.0.0.1
... logging to /root/.ros/log/b1cd7ffc-17d5-11ed-8ba4-482ae3ba050a/roslaunch-ur-li-95qgtn2.unirobotts.local-2313.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Resource not found: The following package was not found in <arg name="urscript_file" default="$(find ur_client_library)/resources/external_control.urscript" doc="Path to URScript that will be sent to the robot and that forms the main control program."/>: ur_client_library
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration
ROS path [2]=/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver
ROS path [3]=/catkin_ws/src/fmauch_universal_robot/ur_kinematics
ROS path [4]=/catkin_ws/src/fmauch_universal_robot/ur_gazebo
ROS path [5]=/catkin_ws/src/fmauch_universal_robot/ur5e_moveit_config
ROS path [6]=/catkin_ws/src/fmauch_universal_robot/ur5_moveit_config
ROS path [7]=/catkin_ws/src/fmauch_universal_robot/ur3e_moveit_config
ROS path [8]=/catkin_ws/src/fmauch_universal_robot/ur3_moveit_config
ROS path [9]=/catkin_ws/src/fmauch_universal_robot/ur16e_moveit_config
ROS path [10]=/catkin_ws/src/fmauch_universal_robot/ur10e_moveit_config
ROS path [11]=/catkin_ws/src/fmauch_universal_robot/ur10_moveit_config
ROS path [12]=/catkin_ws/src/fmauch_universal_robot/ur_description
ROS path [13]=/catkin_ws/src/Universal_Robots_ROS_Driver/ur_dashboard_msgs
ROS path [14]=/catkin_ws/src/fmauch_universal_robot/universal_robots
ROS path [15]=/catkin_ws/src/Universal_Robots_ROS_Driver/controller_stopper
ROS path [16]=/opt/ros/noetic/share
The traceback for the exception was written to the log file
``` 

To reproduce follow the steps to build the driver from an all source build. Once the driver has been build, launch the driver and you should see an output similar to the one above.

fmauch, 2022-08-12:
That's definitively interesting! Though I am a bit puzzled as I use an all-src build for development all the time. Which version of the client library did you use?

urmahp, 2022-08-12:
I realized when running the ROS driver in a docker container. I cloned the newest version of the client library from the master branch. I don't know if it has something to do with running the driver in a docker container. I have added the Dockerfile below. 
```Dockerfile
ARG ROS_DISTRO=noetic

FROM ros:$ROS_DISTRO

RUN echo "hello world"

# Install build tools
RUN apt-get update && apt-get install -y \
    python3-catkin-tools \
    git \
	vim \
	nano \
    && rm -rf /var/lib/apt/lists/*

# Setup workspace
RUN mkdir -p catkin_ws/src
WORKDIR /catkin_ws

# Clone the driver
RUN git clone https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git src/Universal_Robots_ROS_Driver

# Clone the calibration devel
RUN git clone -b calibration_devel https://github.com/fmauch/universal_robot.git src/fmauch_universal_robot

# Clone the client library
RUN git clone https://github.com/UniversalRobots/Universal_Robots_Client_Library.git src/Universal_Robots_Client_Library

# Install Dependencies
RUN sudo apt-get update -qq
RUN rosdep update
RUN rosdep install --from-paths src --ignore-src -y

# catkin_make
RUN /bin/bash -c ". /opt/ros/${ROS_DISTRO}/setup.bash; catkin_make_isolated -j1 -l1"

RUN echo "source /catkin_ws/devel_isolated/setup.bash" >> /etc/bash.bashrc

ENTRYPOINT /bin/bash
```

fmauch, 2024-07-03:
I'm closing this, since this never has been confirmed and didn't show up anywhere else in the meantime.

closed 2024-07-03 by fmauch